espressif/esp-idf #7312: Migrating a project from mbedTLS to WolfSSL

baldhead69, 2021-07-23:
Hi,

For the little i researched the WolfSSL library is faster and use less processor resources (ram memory at least), the flash memory is a little bigger, which wouldn't be a problem for me.

Why does not exist a component wolfssl in esp-idf ?

Is it possible to put wolfssl at the place of mbedtls ?
If yes,  how ?

Application: WebSocket Secure Server(WSS).

esp-idf version: v4.2.1

Thank's.

ESP-YJM, 2021-07-23:
@baldhead69 The wolfssl license has expired. If you want to use it, please refer to https://github.com/espressif/esp-wolfssl.

baldhead69, 2021-07-24:
Hi,

In "menuconfig -> Component Config -> ESP-TLS -> choose SSL Library", only appear mbedTLS.

NOTE These options are valid for esp-tls only if wolfSSL is selected as its SSL/TLS Library.
Where i select ?

How can i integrate wolfssl at the place of mbedTLS ???

Thank's.

[image]
[image]

baldhead69, 2021-07-24:
Hi,

I follow this link:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_tls.html

But now, at compile the, the link error message appear:
.../esp-wolfssl/wolfssl/src/internal.c:30779: undefined reference to `LowResTimer'

mahavirj, 2021-07-26:
@baldhead69 

Can you please confirm that wolfssl submodule pointer is correct on your side?

```
$ cd esp-wolfssl
$ git submodule 
3f13b49fa318fbd3216d7da36d942e7c276d3413 ../../wolfssl (v4.3.0-stable)
```

baldhead69, 2021-07-26:
[image]

baldhead69, 2021-07-26:
@mahavirj,

https://github.com/wolfSSL/wolfssl/issues/4241

baldhead69, 2021-07-26:
Hi,

How to enable wolfssl debug on monitor ?

Thank's.

baldhead69, 2021-07-27:
How to configure wolfssl lib to use spiram memory ?

mahavirj, 2021-07-27:
@baldhead69 

- Seems like you have not cloned `esp-wolfssl` repository recursively, see instructions at https://github.com/espressif/esp-wolfssl#getting-started
- Currently supported `wolfssl` version is v4.3.0 (as I had highlighted in my earlier comment), so request to use same on your side as well.

> How to enable wolfssl debug on monitor ?

You may try to enable debug options through configuration file at https://github.com/espressif/esp-wolfssl/blob/master/port/user_settings.h#L86

> How to configure wolfssl lib to use spiram memory ?

This is not supported as part of wolfssl port yet.

mahavirj, 2021-08-18:
Closing as relevant PR https://github.com/espressif/esp-wolfssl/pull/12 is now merged in.

closed 2021-08-18 by mahavirj