espressif/esp-idf #7315: Intermittent http server error: httpd_txrx: httpd_sock_err: error in recv : 104

mbratch, 2021-07-23:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      ESP32-PICO-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4.1
- Module or chip used:  ESP32-PICO-D4
- IDF version (run ``git describe --tags`` to find it): 4.3
- Build System:        idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 8.4.0
- Operating System:     WIndows 10
- (Windows only) environment type: PowerShell
- Using an IDE?: Yes (Eclipse)
- Power Supply:        USB

## Problem Description

I am running an application on my ESP32 Pico kit which runs an http REST server which also serves files. I have a number of request handlers for REST paths, and I have one handler for a built-in web page which is modeled after the example code provided by Espressif for a REST server. It serves web pages from a spiffs partition.

I run the server continually and have the monitor on. Approximately once per day I will get a couple of messages on the monitor as follows:

```
W (923632) httpd_txrx: httpd_sock_err: error in recv : 104
W (923642) httpd_txrx: httpd_sock_err: error in recv : 104
```
These will occur even though my server is IDLE: I have no requests for web pages or REST commands being sent. The application otherwise appears to be working normally, and I have not noticed an error in behavior other than to see these warnings appear. If fact, if I execute a wide variety of REST API calls and make requests to the web server, I do not get these messages. They seem to appear independent of my use of the server.

I've done quite a bit of research on the 104 (connection reset by peer) error but have not been able to come to a conclusion on what the cause could be, especially since they occur when I am not making any requests to the server.

### Expected Behavior
During operation of http server application, I should not see 104 (connection reset by peer) errors

### Actual Behavior
During operation of http server application, I occasionally see 104 (connection reset by peer) errors, even if I am not making any requests to the server

### Steps to reproduce

1. Start application with console monitor connected and on
2. Start http server and verify correct initialization and operation of the REST API and http server without errors or warnings
3. Stop sending commands/requests to the server and let it run idle for up to 1 day
4. Observe 104 error appear on the monitor

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
[... 101 lines cut ...]
I (1412) wifi:security: WPA2-PSK, phy: bgn, rssi: -49
I (1412) wifi:pm start, type: 1

I (1412) wifi:AP's beacon interval = 102400 us, DTIM period = 3
I (1422) BD-PTB: wifiEvent: wifi connected
I (3152) esp_netif_handlers: sta ip: 192.168.1.32, mask: 255.255.255.0, gw: 192.168.1.1
I (3152) BD-PTB: wifiEvent: wifi obtained IP
I (3152) BD-PTB: Wifi connected with prior configuration
I (3162) BD-PTB: SSID = Antioch
I (3162) BD-PTB: MAC address = D8:A0:1D:5D:CB:64
I (3172) BD-PTB: IP address = 192.168.1.32
I (3172) BD-PTB: Gateway = 192.168.1.1
I (3182) BD-PTB: Netmask = 255.255.255.0
I (3182) BD-PTB: Setting up mDNS
I (3192) BD-PTB: mDNS host name = BD-PTB-5DCB64
I (3202) BD-PTB: Starting web server
I (3202) BD-PTB: Starting HTTPD server
I (3222) BD-PTB: Partition /www size: total: 180971, used: 58232
I (3472) BD-PTB: Initialization complete
W (923632) httpd_txrx: httpd_sock_err: error in recv : 104
W (923642) httpd_txrx: httpd_sock_err: error in recv : 104

```
## Other items if possible
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/6868370/sdkconfig.txt)

negativekelvin, 2021-07-23:
Is there something port scanning on your network?  Have you captured the network traffic to coincide with these events?

mbratch, 2021-07-23:
> Is there something port scanning on your network? Have you captured the network traffic to coincide with these events?

Thanks for the reply!

I do not, to my knowledge, have port scanning turned on on my network. I have not done any general network traffic capture. I am logging any network or wifi events in the device, but none are occurring.

ESP-YJM, 2021-07-27:
@mbratch Could you please help open debug log(idf.py menuconfig->component config->Log output->Default log verbosity (Debug)  and component config->LWIP->[*] Enable LWIP Debug->[*] Enable TCP debug messages). If possible, help capture the packet between server and client.  Server recveive `104` indicates client send `RST`. Maybe client close socket with `RST` instead of  normal close `FIN`.

mbratch, 2021-07-27:
> @mbratch Could you please help open debug log(idf.py menuconfig->component config->Log output->Default log verbosity (Debug) and component config->LWIP->[*] Enable LWIP Debug->[*] Enable TCP debug [...]

Yes, I will try that. Thank you for the suggestion. I'll provide relevant log snippets when I get them.

ESP-YJM, 2021-07-29:
@mbratch I mean `Enable LWIP Debug` and `TCP debug` is this https://github.com/espressif/esp-idf/blob/v4.3/components/lwip/Kconfig#L851 and https://github.com/espressif/esp-idf/blob/v4.3/components/lwip/Kconfig#L910.  I need know the way peer close the socket(`FIN` or `RST`). From errno `104`, it maybe peer send `RST`. So we will print `104`. I think it's normal issue due to peer close socket in innormal way. For new errno `128`, it maybe peer send `FIN`. I think this not affect the function, We need to figure out the original.

mbratch, 2021-07-29:
@ESP-YJM thank you for the reply. I will keep trying to get more information. I agree that 104 does not cause much problem. I do not see any problems in functionality of my product when this error appears, so I am not very concerned.

I have reported two other issues which I am more concerned about, especially #7286 (IDFGH-5563) in which I get sudden wifi disconnection and it takes up to 60 seconds before it is able to automatically reconnect.  Also, #7328 (IDFGH-5608) in which the device sometimes will pause for up to 30 seconds before responding to HTTP request. But #7286 is the biggest concern.

ESP-YJM, 2021-09-13:
@mbratch I think you can foucs on when occur the 104 error, do the device can work normal.

mbratch, 2021-09-13:
@ESP-YJM thank you for the comment. The 104 error does not appear to negatively affect the operation of the device. My most serious problem is #7286 (which is still open) followed by #7328.

ESP-YJM, 2021-09-27:
@mbratch Could this issue be closed? For other issues, i think you can track them in corresponding issue.

mbratch, 2021-09-27:
@ESP-YJM it can be closed on the basis that it doesn't seem to cause any harm.

Alvin1Zhang, 2021-10-08:
@mbratch Thanks for reporting and sharing the updates, feel free to reopen.

closed 2021-10-08 by Alvin1Zhang

mbratch, 2022-08-14:
I know we closed this, but not because it's not there. :)
I'm still getting this issue and it sometimes causes a disruption in the wifi communication: for a few seconds, a client is unable to connect to the ESP32.

ESP-YJM, 2022-08-15:
Hi @mbratch , thanks for your reply. ESP32 client can't connect to ESP32, do you mean wifi connection or http connection? Do you have any log that can share with me.

mbratch, 2022-08-16:
Hi @ESP-YJM , the client I am using is not another ESP32. I am using a client on Android or Windows to connect to the ESP32. It is an http connection over wifi.

ESP-YJM, 2022-08-16:
@mbratch I think you can use PC to connect to the ESP32 with http and open wireshark to capture the packet.

CharlesScoville, 2023-04-11:
I, too, have this error.... I am just now noticing.

I can mostly confirm that it is just the client (Firefox in my case) sending RST to close TCP connection. That's what it says in Wireshark anyway. . .

`296	2.881250	192.168.1.9	192.168.1.14	TCP	54	49888 → 80 [RST, ACK] Seq=601 Ack=203511 Win=0 Len=0`

The best way to reproduce so far has been leaving Firefox open on the graph page of the RESTful server example, then reloading the page after a good wait. However, this is still pretty hit/miss. I haven't quite narrowed down the exact mechanism that causes Firefox to do this... but that is a Firefox issue anyway.

Either way, a more graceful way for the http daemon to handle RST would be cool.

caco3, 2023-04-14:
Hi @CharlesScoville 

Thanks for your investigations and explanations.
I agree it is not critical, but it can confuse users (we had that quite often). And since we as device developers can not fix it, I think it would be great if this warning could get disabled or catched or changed to debug log.