espressif/esp-idf #7317: cJSON_CreateIntArray(i_numbers, 0) causes a panic

nopnop2002, 2021-07-24:
## Environment
```
Core is ESP32@160Mhz
freeRTOS version:V10.2.1
NEWLIB version:3.3.0
lwIP version:2-1-3-0
ESP-IDF version:v4.4-dev-2184-g166c30e7b
package is ESP32D0WDQ6
cJSON_Version:1.7.14
```

## Problem Description
cJSON_CreateIntArray(i_numbers, 1) don't causes a panic.
But cJSON_CreateIntArray(i_numbers, 0) causes a panic.

### Expected Behavior
cJSON_CreateIntArray(i_numbers, 0) don't causes panic.

### Actual Behavior
cJSON_CreateIntArray(i_numbers, 0) causes a panic.

```
I (0) cpu_start: Starting scheduler on APP CPU.
I (330) JSON: Serialize.....
I (340) JSON: my_json_string
{
        "intArray":     [1, 11, 111]
}
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400e738d  PS      : 0x00060130  A0      : 0x800d2562  A1      : 0x3ffb5900
0x400e738d: cJSON_CreateIntArray at /home/nop/esp-idf/components/json/cJSON/cJSON.c:2565

A2      : 0x3ffb7018  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x3ffb5920  A7      : 0x3ffb70c8  A8      : 0x00000000  A9      : 0x00000001
A10     : 0x3ffb7018  A11     : 0x00000000  A12     : 0x3f401c7c  A13     : 0x3ffb5900
A14     : 0x3ffb58e0  A15     : 0x0000000c  SAR     : 0x00000010  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000004  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

Backtrace:0x400e738a:0x3ffb59000x400d255f:0x3ffb5920 0x400ebb3c:0x3ffb5950 0x400874f1:0x3ffb5970
0x400e738a: cJSON_CreateIntArray at /home/nop/esp-idf/components/json/cJSON/cJSON.c:2547 (discriminator 1)

0x400d255f: app_main at /home/nop/rtos/esp-idf-json/json-create-array-test/main/main.c:37

0x400ebb3c: main_task at /home/nop/esp-idf/components/freertos/port/port_common.c:142 (discriminator 2)

0x400874f1: vPortTaskWrapper at /home/nop/esp-idf/components/freertos/port/xtensa/port.c:168

ELF file SHA256: f85e81c078cc2294
```

## My code
```
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "cJSON.h"

static const char *TAG = "JSON";

void app_main()
{
        ESP_LOGI(TAG, "Serialize.....");
        cJSON *root;
        root = cJSON_CreateObject();

        int i_numbers[3];
        i_numbers[0] = 1;
        i_numbers[1] = 11;
        i_numbers[2] = 111;
        cJSON *intArray;
        intArray = cJSON_CreateIntArray(i_numbers, 3);
        cJSON_AddItemToObject(root, "intArray", intArray);

        char *my_json_string = cJSON_Print(root);
        ESP_LOGI(TAG, "my_json_string\n%s",my_json_string);
        cJSON_Delete(root);

        // Buffers returned by cJSON_Print must be freed by the caller.
        // Please use the proper API (cJSON_free) rather than directly calling stdlib free.
        cJSON_free(my_json_string);

        cJSON *root2;
        root2 = cJSON_CreateObject();
        intArray = cJSON_CreateIntArray(i_numbers, 0);
        cJSON_AddItemToObject(root2, "intArray", intArray);

        char *my_json_string2 = cJSON_Print(root2);
        ESP_LOGI(TAG, "my_json_string\n%s",my_json_string2);
        cJSON_Delete(root2);

        // Buffers returned by cJSON_Print must be freed by the caller.
        // Please use the proper API (cJSON_free) rather than directly calling stdlib free.
        cJSON_free(my_json_string2);
}
```

## Workaround
I can't think of it.

negativekelvin, 2021-07-24:
I think this is fixed in upstream cjson

AxelLin, 2021-07-24:
FYI:
The esp-idf master is using cJSON-v1.7.14 release, however there are  NULL pointer dereference bug fixes after v1.7.14 release.
https://github.com/DaveGamble/cJSON/commits/master

nopnop2002, 2021-07-24:
Do I need to update the cJSON library myself?

AxelLin, 2021-07-26:
It's fixed by https://github.com/DaveGamble/cJSON/commit/7b6645794d67eebb20678d8d30e8e8ba5357184a , the esp-idf needs update.

nopnop2002, 2021-07-26:
@AxelLin 

Thank you.

I hope the libraries included in esp-idf master will be updated with new ones.

mahavirj, 2021-07-27:
@AxelLin @nopnop2002

Thanks for notifying on this. But it appears that cJSON never had release after `1.7.14`, and we usually prefer stable upstream releases. We will check and create required fix for this case.

AxelLin, 2021-08-25:
@mahavirj 
FYI, https://github.com/DaveGamble/cJSON/releases/tag/v1.7.15

mahavirj, 2021-08-25:
@AxelLin Thanks for notifying. We will merge this soon.

closed 2021-08-29 by espressif-bot