espressif/esp-idf #7320: New link errors with master branch

surfskidude, 2021-07-24:
I have two projects, one create a library (libmylib.a) and the other project uses this library.

I just checked out the latest master branch (full checkout including compiler) and I now get the following link errors from libmylib.a. Note, that I have rebuilt everything.

__wrap_longjmp
xTimerCreate
xTimerGenericCommand

Alvin1Zhang, 2021-07-27:
@surfskidude Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

surfskidude, 2021-07-28:
esp-idf (tested head and V4.3), ESP32-Wrover-Kit, CMake, idf.py, xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0, Linux.

Steps to reproduce: try to compile and link this ready to use CMake project, which includes a pre-compiled library: https://github.com/RealTimeLogic/LspAppMgr-ESP32

surfskidude, 2021-10-31:
I found a solution. Moving from V4.1.2 to newer releases requires a change to the add_prebuilt_library command.

I can link when I change:
add_prebuilt_library(prebuilt "libbas.a")
to:
add_prebuilt_library(prebuilt "libbas.a" PRIV_REQUIRES xtensa)

closed 2021-10-31 by surfskidude