espressif/esp-idf #7322: c语言标准 the standard of C language

PickaxeHit, 2021-07-25:
esp-idf 用的是c语言的标准？c99还是c11？

Which standard of the C language used in esp-idf? c99 or c11?

nopnop2002, 2021-07-26:
I think it's C11 because these headers can be used.

```
#include "float.h"
#include "stdalign.h"
#include "complex.h"
```

closed 2021-09-29 by Dazza0

tingjhenjiang, 2024-08-19:
how do I set the C standard for the "main" (entire project)? 
I've tried:

rootfolder/CMakeLists.txt
```
cmake_minimum_required(VERSION 3.5)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(my_project)

# Set the C standard to C11
target_compile_options(${CMAKE_PROJECT_NAME} PRIVATE -std=gnu11)
```
OR
```
rootfolder/main/CMakeLists.txt
target_compile_options("." PRIVATE -std=gnu11)
idf_component_register(SRCS "main.c"
                    INCLUDE_DIRS "")
```
both failed.
@Dazza0 Thanks!

igrr, 2024-08-19:
@tingjhenjiang In main/CMakeLists.txt, you can use this:

```c++
idf_component_register(SRCS main.c INCLUDE_DIRS ".")
target_compile_options(${COMPONENT_LIB} PRIVATE -std=gnu11)
```

However, please note that this is probably unnecessary... C language standard in all currently maintained IDF versions (v5.x) is already `gnu17`.

tingjhenjiang, 2024-08-19:
> @tingjhenjiang In main/CMakeLists.txt, you can use this:
>
> ```c++
> idf_component_register(SRCS main.c INCLUDE_DIRS ".")
> target_compile_options(${COMPONENT_LIB} PRIVATE -std=gnu11)
> ```
>
> However, please [...]

@igrr It succeeds. Thanks!
Btw, the "gnu11" is simply an example for this post. Anyway Thanks!