espressif/esp-idf #7323: Connecting HomeSpan keeps requesting password

Adajaken, 2021-07-25:
I have 3 questions,

1. For those esp32 module which has been connected, even if I factory reset it, it keeps requesting ‘enter password’ when re-connecting to hotspot HomeSpan. What’s the password?

2. After circuit board is made (welded esp32 module is burn with program and tested well), if not connecting to tx, rx Pin, then PCB’s status LED will keep flickering; and if connecting to rx, tx Pin, then it will connect to WiFi automatically (based on printed information, it’s not always able to connect to WiFi successfully, sometimes, it stays at scanning WiFi process), how to solve it?

3.How to solve below error?
[image]

Pls help and thanks in advance.

negativekelvin, 2021-07-25:
Homespan is not an espressif project you should ask here https://github.com/HomeSpan/HomeSpan

Your hardware question you should ask here and post schematic and photo https://esp32.com/viewforum.php?f=12

Alvin1Zhang, 2021-07-27:
@negativekelvin Thanks for the notes.

closed 2021-07-27 by Alvin1Zhang