espressif/esp-idf #7324: PHY RF calibration data API/info/defaults

boarchuz, 2021-07-25:
Sorry to bring this up again but I've hit a different snag with RF calibration data so I'm revisiting this hoping for a solution.

**Problem:**

- RF calibration draws significantly more current than any other typical activity on an ESP32
- It's very difficult to alter if/when/how RF calibration occurs because of a limited API
- Calibration data is opaque and not documented, so it's impossible to determine if/when/why it is necessary

In practice, this means that hardware designs must accommodate >500mA peak current draw, and will likely recalibrate often which can affect reliability (brownouts), power consumption, and performance.

Currently, calibration is handled automatically, but there are use cases where setting data from application-managed memory (eg. custom partition, DROM or RTC memory) or customising the calibration mode (eg. prefer lowest power when running on battery, but allow calibration when adapter is attached) would be preferable.

**Describe the solution you'd like**

- API to set calibration data manually, or override default load function
- Calibration data struct documented *and/or* a macro to define sane defaults for struct initialisation *and/or* guidance on when and why calibration is necessary (eg. could I calibrate an individual device, then embed the resulting data for all devices that share that design, valid for their lifetime?)

eg.

```
#define ESP_PHY_CAL_DATA_DEFAULT() {...}
esp_err_t esp_phy_set_cal_data(esp_phy_calibration_data_t *cal_data, esp_phy_calibration_mode_t mode);

// eg. usage 1
esp_phy_calibration_data_t cal_data = ESP_PHY_CAL_DATA_DEFAULT();
ESP_ERROR_CHECK(esp_phy_set_cal_data(&cal_data, PHY_RF_CAL_NONE));

// eg. usage 2
RTC_DATA_ATTR esp_phy_calibration_data_t cal_data = ESP_PHY_CAL_DATA_DEFAULT();
esp_phy_set_cal_data(&cal_data, PHY_RF_CAL_PARTIAL);
esp_deep_sleep_start();
// ...
esp_phy_set_cal_data(&cal_data, PHY_RF_CAL_NONE);
```
or
```
typedef esp_err_t (*esp_phy_cal_data_get_fn_t)(esp_phy_calibration_data_t **cal_data_ptr, esp_phy_calibration_mode_t *max_mode); // allocate memory, load/ initialise, etc
typedef void (*esp_phy_cal_data_release_fn_t)(esp_phy_calibration_data_t *cal_data); // save data, free memory, etc

void esp_phy_set_cal_data_fns(esp_phy_cal_data_get_fn_t get_fn, esp_phy_cal_data_release_fn_t release_fn);
```

glitch911, 2021-09-23:
@boarchuz do you already have a workaround to avoid RF calibration? I read with interest many thread from you but I can't get it to work :/
I'm using ESP32-C3 (it's OK on ESP32: https://www.esp32.com/viewtopic.php?t=9955) but it's not working for ESP32-C3: https://www.esp32.com/viewtopic.php?f=13&t=23411). Most likely root cause to me is that calibration data is not stored correctly (but in changelog of 4.3.1 "Support for saving phy calibration data to NVS for ESP32-S2, ESP32-C3 & ESP32-S3" so this should be OK) maybe a default parameter changed or something like this...
Any help would be appreciated :)

@Espressif: IoT market is using a lot Wi-Fi in RX only (for geolocation) and is very sensitive to current pulse and energy drain so please give use descent firmware so that we can use in an optimized way your great hardware :)

Regards, 
Thomas

boarchuz, 2021-09-23:
@glitch911 You can customize the phy calibration in IDF to use some constant data if you want to try it out (dump the calibrated output to get the data you need). Without espressif guidance you're on your own though, which is a shame as much smaller/cheaper/reliable power regulation designs could be made possible.

boarchuz, 2021-09-23:
@glitch911 I have whipped up a quick and dirty example here: https://github.com/boarchuz/esp32_example_cal_partition

Peak current (3.3V):
Default: 340mA
With changes: 120mA

This loads and writes to the partition, which is not ideal, but it's good enough to demo the concept of a 'factory calibration data partition' which would allow the application to avoid expensive calibration where necessary.

jack0c, 2022-05-20:
@boarchuz It is not likely to provide what you want. And calibration is used to avoid fit the different scenarios. But we can try to shorten the calibration time. Can you provide the time and current value that can work on your case?

jack0c, 2022-05-20:
> @espressif: IoT market is using a lot Wi-Fi in RX only (for geolocation) and is very sensitive to current pulse and energy drain so please give use descent firmware so that we can use in an optimized way your great hardware :)
@glitch911 I don't know what's the mean of `RX only` here? For instance, if a device send a packet to ESP32, ESP32 may need send 80211 ACK to the device to tell the device the packet is received. Do you mean 80211 ACK is not necessary in your case?

If so, you are require a small calibration which only for RX, right?

glitch911, 2022-05-20:
@jack0c Yes, your understanding is good: we only snif surrounding access point mac adresses, we never connect to anything so we'd love to avoid unnecessary and battery killing current pulse by skipping RF calibration.

It is already possible to skip on ESP32 (https://www.esp32.com/viewtopic.php?t=9955) and same tips should have worked as well for ESP32-C3 but it's not available right now. Good news is: it will be in the future:
https://github.com/espressif/esp-idf/issues/7596#issuecomment-1103722878

boarchuz, 2022-05-20:
> It is not likely to provide what you want. And calibration is used to avoid fit the different scenarios.

@jack0c I recognise that this is not a common concern. Would it be possible to simply confirm the following, so that I can implement something myself?

- A device's PHY calibration data is not expected to change _significantly_ over its lifetime
- Calibration data from multiple devices in identical designs will be similar/identical
- Poor PHY calibration data will not damage the device

> But we can try to shorten the calibration time. Can you provide the time and current value that can work on your case?

More control over any part of this process would be fantastic. The closer to 0 for time and current (while maintaining decent RF performance), the better.

jack0c, 2022-05-24:
> * A device's PHY calibration data is not expected to change _significantly_ over its lifetime
That's true, we have full calibration and partial calibration now. In partial calibration, the calibrated part vary with hardware and environments.
 Whether we can calibrate less data is still need research.
> * Calibration data from multiple devices in identical designs will be similar/identical
Yes, they are similar somehow.
> * Poor PHY calibration data will not damage the device
Yes, I think so. The result may be the devices can't TX/RX until a reset happen.

We will continue to shorten the calibration time anyway.

JesseRiemens, 2022-07-05:
@jack0c Great work! Could this also be done for the base esp32?

MaxwellAlan, 2022-07-13:
Hi @JesseRiemens 
Yes, esp32 also can be optimized, but it can not be optimized as well as esp32c3/s3. It will take some time to do this since it is not a high priority issue.
Thanks.

elmaghba, 2022-09-22:
@boarchuz Did you find anyway to confirm that: 
- A device's PHY calibration data is not expected to change significantly over its lifetime.
- Calibration data from multiple devices in identical designs will be similar/identical.
- Poor PHY calibration data will not damage the device.
or is there any way to access the PHY calibration data for the esp32?

For the RF fingerprinting community, it is assumed that devices with identical designs have different radio impairments, so I do not think the second one holds true. 

It will be great if these data are accessible! This will advance the research in the hardware security.

elmaghba, 2022-09-22:
@MaxwellAlan 
Is there a way to access the radio imperfection values that the built-in RF calibration routines are using to do the calibration/cancel the radio imperfection?

MaxwellAlan, 2022-10-19:
Hi @elmaghba 

> Is there a way to access the radio imperfection values that the built-in RF calibration routines are using to do the calibration/cancel the radio imperfection?

No. There is no related APIs in phy lib.

MaxwellAlan, 2022-11-18:
Hi @elmaghba 

Sorry for this feature request. The RF calibration section is confidential, so we can not release related APIs.

Thanks.

Batov, 2022-12-29:
> @jack0c Yes, your understanding is good: we only snif surrounding access point mac adresses, we never connect to anything so we'd love to avoid unnecessary and battery killing current pulse by [...]

Could you please explain the workaround?

Seems like that this hack works only after first FULL_CALIBRATION because NVS hasn't contained calibration data yet. [Deepsleep hack](https://github.com/espressif/esp-idf/blob/7f5ecbe533b2666df6b10658a023b5f637320696/components/esp_phy/src/phy_init.c#L717)  will be overwritten by [empty NVS](https://github.com/espressif/esp-idf/blob/7f5ecbe533b2666df6b10658a023b5f637320696/components/esp_phy/src/phy_init.c#L722). As I understand we coulndn't pre-generate calibration data, it's only proprietary PHY library buisness. So I think that the workaround works only to avoid partitional (not first) calibrations.

boarchuz, 2023-02-18:
> Sorry for this feature request. The RF calibration section is confidential, so we can not release related APIs.

Thankyou for the improvements and information. With this confirmation, I think this can be put to rest now.

closed 2023-02-18 by boarchuz