espressif/esp-idf #7327: Floating point operation error

zenghi, 2021-07-26:
#include<stdio.h>

int main()
{

    int temp = 23001;
    printf("\n zenghi %f: %s: %u:\n",temp / 100.00f,__FILE__, __LINE__);

}

get wrong data 230.009995

igrr, 2021-07-26:
@zenghi Exactly the same output is produced on Linux: https://ideone.com/svsH6W, so it doesn't seem to be an issue with ESP-IDF.

The reason is that 32-bit `float` gives only about 7 decimal digits of precision, and you are printing more digits. If you format it as "%.4f", then 230.009995 will get rounded and you will get "correct" output (https://ideone.com/ddlu1i). 
(Regarding the reason why `printf` can't pick the right precision for you, see this answer: https://stackoverflow.com/a/20049243.)

zenghi, 2021-07-27:
@igrr I can get the correct value by compiling and testing in the window environment. Unexpectedly, I can't get the correct value in the Linux environment. Thank you very much for your explanation!

closed 2021-07-27 by zenghi