espressif/esp-idf #7328: HTTP server will intermittently take 30 or more seconds to respond to an HTTP request

mbratch, 2021-07-26:
## Environment

- Development Kit: ESP32-PICO-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4.1
- Module or chip used: ESP32-PICO-D4
- IDF version (run git describe --tags to find it): 4.3
- Build System: idf.py
- Compiler version (run xtensa-esp32-elf-gcc --version to find it): 8.4.0
- Operating System: WIndows 10
- (Windows only) environment type: PowerShell
- Using an IDE?: Yes (Eclipse)
- Power Supply: USB

## Problem Description

I have an HTTP REST server modeled after the rest server example provided by the ESP-IDF framework. When a REST API request is made, typically the request handler (registered with `httpd_register_uri_handler`) is called nearly immediately. However, there are many periods of time when the request handler is not called for 30 or more seconds after the HTTP request is sent over the network to the ESP32. The only other activity occurring in the ESP32 is a 500ms LED flash timer, and a 2 second LED status update timer, each of which takes very little time (as timer tasks should).

### Expected Behavior

When an HTTP REST API request is sent to the ESP32 REST server, the ESP32 library code that receives the request and dispatches the handler should take a lot less than 30 seconds.

### Actual Behavior

Sometimes the dispatching of the HTTP request handler takes in excess of 30 seconds.

### Steps to reproduce

1. Run ESP32 REST API code that is idle and awaiting REST API requests
2. Send regular REST API requests to the ESP32 and observe prompt response (within a second or two) to the requests
3. Observe occasional delay of up to 30 or more seconds in response to some REST API requests

## Debug Logs

No messages appear on the monitor during the operation of this test.

negativekelvin, 2021-07-26:
Suggest you turn on debug level logging for relevant components and get a Wireshark capture

mbratch, 2021-07-27:
> Suggest you turn on debug level logging for relevant components and get a Wireshark capture

I'll be working on capturing some useful logs.

mbratch, 2021-08-02:
I'm in the process of doing some testing. I believe this is an issue either with my phone client or the network, but not the ESP32. I'm doing further testing to confirm this hypothesis. It takes some time because the scenario is very intermittent.

AxelLin, 2021-08-02:
https://www.esp32.com/viewtopic.php?f=13&t=22478

mbratch, 2021-08-02:
> https://www.esp32.com/viewtopic.php?f=13&t=22478

That's very interesting. I definitely need more testing as I am not convinced the testing I've done thus far is very conclusive.

MarcelKr, 2021-08-03:
I'm experiencing the same issue. I'm answering a get request with a ~300 byte JSON. Sometimes it takes around 100 ms, more commonly it takes 2 to sometimes more than 60 seconds to respond.

This is a wireshark capture, where it took 20 seconds: [20s_to_respond.zip](https://github.com/espressif/esp-idf/files/6923137/20s_to_respond.zip)

The URI handler looks like this:
```c
static esp_err_t channels_get(httpd_req_t *req)
{
    // Prepare response data struct for user callback function
    response_data_t response_data = {
        .content = NULL,
        .content_len = 0,
        .has_error = true,
        .error_code = HTTPD_501_METHOD_NOT_IMPLEMENTED,
    };

    channels_get_user(&response_data);

    if (response_data.has_error) {
        httpd_resp_set_status(req, get_error_status(response_data.error_code));
    }

    httpd_resp_set_type(req, HTTPD_TYPE_JSON);
    httpd_resp_sendstr(req, response_data.content);

    free(response_data.content);

    return ESP_OK;
}

static const httpd_uri_t uri_channels_get = {
    .uri      = "/channels",
    .method   = HTTP_GET,
    .handler  = channels_get,
    .user_ctx = NULL
};
```
I used `esp_timer_get_time()` to measure how long the handler takes and it's always around 8 ms. (not shown in this code).

From introducing debug prints into the handler and observing the timing of the prints, it seems that the handler is called quite early after sending the get request. The time after the handler is executed and the actual response seems to be the part where the issue is.

Here's a piece of the log:
```
D (169981) httpd: httpd_server: processing socket 57
D (169981) httpd_sess: httpd_sess_process: httpd_req_new
D (169981) httpd_txrx: httpd_recv_with_opt: requested length = 128
D (169981) httpd_txrx: httpd_recv_with_opt: received length = 92
D (169991) httpd_parse: read_block: received HTTP request block size = 92
D (170001) httpd_parse: cb_url: message begin
D (170001) httpd_parse: cb_url: processing url = /channels
D (170011) httpd_parse: verify_url: received URI = /channels
D (170011) httpd_parse: cb_header_field: headers begin
D (170021) httpd_txrx: httpd_unrecv: length = 68
D (170021) httpd_parse: pause_parsing: paused
D (170031) httpd_parse: cb_header_field: processing field = Host
D (170031) httpd_txrx: httpd_recv_with_opt: requested length = 128
D (170041) httpd_txrx: httpd_recv_with_opt: pending length = 68
D (170051) httpd_parse: read_block: received HTTP request block size = 68
D (170051) httpd_parse: continue_parsing: skip pre-parsed data of size = 5
D (170061) httpd_parse: continue_parsing: un-paused
D (170061) httpd_parse: cb_header_value: processing value = 192.168.178.54
D (170071) httpd_parse: cb_header_field: processing field = User-Agent
D (170081) httpd_parse: cb_header_value: processing value = insomnia/2021.4.1
D (170091) httpd_parse: cb_header_field: processing field = Accept
D (170091) httpd_parse: cb_header_value: processing value = */*
D (170101) httpd_parse: cb_headers_complete: bytes read     = 92
D (170101) httpd_parse: cb_headers_complete: content length = 0
D (170111) httpd_parse: pause_parsing: paused
D (170111) httpd_parse: cb_no_body: message complete
D (170121) httpd_parse: httpd_parse_req: parsing complete
D (170121) httpd_uri: httpd_uri: request for /channels with type 1
D (170131) httpd_uri: httpd_find_uri_handler: [0] = /api/version
D (170141) httpd_uri: httpd_find_uri_handler: [1] = /status
D (170141) httpd_uri: httpd_find_uri_handler: [2] = /settings
D (170151) httpd_uri: httpd_find_uri_handler: [3] = /settings
D (170151) httpd_uri: httpd_find_uri_handler: [4] = /dimming/setlevel
D (170161) httpd_uri: httpd_find_uri_handler: [5] = /dimming/mode
D (170171) httpd_uri: httpd_find_uri_handler: [6] = /dimming/mode
D (170171) httpd_uri: httpd_find_uri_handler: [7] = /sensors
D (170181) httpd_uri: httpd_find_uri_handler: [8] = /sensors/temperature
D (170191) httpd_uri: httpd_find_uri_handler: [9] = /sensors/temperature
D (170191) httpd_uri: httpd_find_uri_handler: [10] = /sensors/ph/calibration
D (170201) httpd_uri: httpd_find_uri_handler: [11] = /sensors/ph/calibration
D (170211) httpd_uri: httpd_find_uri_handler: [12] = /channels
D (170221) httpd_txrx: httpd_send_all: sent = 70
D (170221) httpd_txrx: httpd_send_all: sent = 2
D (170221) httpd_txrx: httpd_send_all: sent = 373
I (170231) API_USER: Handler fcn took 8419 us
D (170241) httpd_sess: httpd_sess_process: httpd_req_delete
D (170241) httpd_sess: httpd_sess_process: success
D (170251) httpd: httpd_server: doing select maxfd+1 = 58
```

At the end of the log, see API_USER where I print the time the URI handler took. This is the full log of one request. This is printed quite soon after I send the request. While waiting or when the response is received after several seconds, there's not more log output generated.

* HTTPD is initialized with default settings except for a increased number of handlers. 
* ESP-IDF: v4.3-194-g5b5e46971
* Chip: ESP32-D0WDQ6

mbratch, 2021-08-03:
@MarcelKr in my case, I think I do not see any debug outputs between issuing the request from the client and the end of the delay I'm getting. It's as if there's a delay before the ESP32 even gets the request. But I need to do more testing to confirm, as I'm not sure yet that I trust my observation (it wasn't a good setup). This issue happens infrequently for me (maybe once a day).

For your scenario, you indicated that:

> it seems that the handler is called quite early after sending the get request. The time after the handler is executed and the actual response seems to be the part where the issue is.

The log fragment you are showing spans less than 300ms. Can you show a section of the log that shows the delay? I would like to compare it to the log output I get.

shubhamkulkarni97, 2021-08-03:
Hi @mbratch @MarcelKr, have you enabled `lru_purge_enable` in httpd_config_t?

This helps to purge least recently used connection if no more sockets are available to accept new connection.

MarcelKr, 2021-08-03:
> I do not see any debug outputs between issuing the request from the client and the end of the delay

@mbratch Sorry, maybe I didn't describe that clear enough. I'm having the same hehaviour here. No log output when the client receives the answer. The sequence is as follows:

* Send GET request
* Wait a few ms, at most around 1 s
* Log output as seen in my prev. comment. In my case the URI handler takes ~8 ms.
* Wait for 200 ms to several seconds.
* Answer is received without any further log output

> have you enabled `lru_purge_enable`

Hey @shubhamkulkarni97, enabling this switch doesn't help in my case.

MarcelKr, 2021-08-03:
I just flashed the example `esp-idf/examples/protocols/http_server/simple/` and it also shows this behavior. Also the ping requests take long or are not answered reliably:

```
[marcel@marcel-desktop ~]$ ping 192.168.178.54
PING 192.168.178.54 (192.168.178.54) 56(84) bytes of data.
64 Bytes from 192.168.178.54: icmp_seq=7 ttl=255 Time=2961 ms
64 Bytes from 192.168.178.54: icmp_seq=14 ttl=255 Time=3443 ms
From 192.168.178.55 icmp_seq=21 Host not reachable
From 192.168.178.55 icmp_seq=25 Host not reachable
From 192.168.178.55 icmp_seq=26 Host not reachable
64 Bytes from 192.168.178.54: icmp_seq=44 ttl=255 Time=1059 ms
64 Bytes from 192.168.178.54: icmp_seq=46 ttl=255 Time=1166 ms
64 Bytes from 192.168.178.54: icmp_seq=47 ttl=255 Time=141 ms
From 192.168.178.55 icmp_seq=55 Host not reachable
From 192.168.178.55 icmp_seq=56 Host not reachable
From 192.168.178.55 icmp_seq=57 Host not reachable
From 192.168.178.55 icmp_seq=58 Host not reachable
From 192.168.178.55 icmp_seq=59 Host not reachable
From 192.168.178.55 icmp_seq=60 Host not reachable
From 192.168.178.55 icmp_seq=61 Host not reachable
From 192.168.178.55 icmp_seq=62 Host not reachable
From 192.168.178.55 icmp_seq=63 Host not reachable
From 192.168.178.55 icmp_seq=64 Host not reachable
64 Bytes from 192.168.178.54: icmp_seq=70 ttl=255 Time=1755 ms
64 Bytes from 192.168.178.54: icmp_seq=72 ttl=255 Time=2228 ms
64 Bytes from 192.168.178.54: icmp_seq=74 ttl=255 Time=2857 ms
64 Bytes from 192.168.178.54: icmp_seq=81 ttl=255 Time=4320 ms
64 Bytes from 192.168.178.54: icmp_seq=83 ttl=255 Time=3790 ms
64 Bytes from 192.168.178.54: icmp_seq=84 ttl=255 Time=3114 ms
64 Bytes from 192.168.178.54: icmp_seq=85 ttl=255 Time=2338 ms
64 Bytes from 192.168.178.54: icmp_seq=88 ttl=255 Time=2256 ms
64 Bytes from 192.168.178.54: icmp_seq=91 ttl=255 Time=1113 ms
From 192.168.178.55 icmp_seq=102 Host not reachable
From 192.168.178.55 icmp_seq=103 Host not reachable
From 192.168.178.55 icmp_seq=104 Host not reachable
64 Bytes from 192.168.178.54: icmp_seq=107 ttl=255 Time=1077 ms
```

mbratch, 2021-08-03:
> @mbratch Sorry, maybe I didn't describe that clear enough. I'm having the same hehaviour here. No log output when the client receives the answer. The sequence is as follows:
>
> * Send GET request
> * [...]

No worries. I think mine is different, but as I mentioned, I don't claim that the test was 100% reliable (did I look over at the console in time to confirm the timing of log messages?). This is the failure case for me (I think):

* Send the GET request
* Wait up to 30 seconds
* Suddenly get a bunch of log messages showing that the url handler got the request and processed it, all within a few hundread ms

MarcelKr, 2021-08-03:
Oh dear, the problem is just gone and the responses are reliably received within at most 200 ms. What the hell? I don't know exactly what I changed. It works using the hello-world server from the examples and my own code.

> I actually think it is the same problem as yours, ultimately.

It would be interesting to know if it's the same thing and obviously to find the root cause. At least one more guy (the one creating the forum thread) seems to have a similar issue.

negativekelvin, 2021-08-03:
Do you have WiFi power save disabled?

mbratch, 2021-08-03:
> Do you have WiFi power save disabled?

In my case, I did not specifically enable or disable WiFi power save. I just went looking for the option but couldn't find it. Can you say where it is in the menuconfig?

mbratch, 2021-08-03:
> Hi @mbratch @MarcelKr, have you enabled `lru_purge_enable` in httpd_config_t?
> 
> This helps to purge least recently used connection if no more sockets are available to accept new connection.

I have not tried this option. I will try it after the next time the problem occurs (to make sure my current setup still exhibits the issue).

shubhamkulkarni97, 2021-08-16:
@MarcelKr @mbratch Any updates on the issue?

mbratch, 2021-08-16:
@shubhamkulkarni97 I do not have any updates. I've been planning on using some of the suggested settings, but have been focusing on a more serious problem at the moment (#7286). For the HTTP response delay, it's difficult to catch "live". I will need to write a script so I can check timings and the effect of some of the suggested settings. Not too difficult, I just have to get to it.

MarcelKr, 2021-08-19:
@shubhamkulkarni97 I didn't experience the issue again. I tinkered around with WiFi reception (different angles of the antenna or PCB, inhibiting reception a bit with hands/fingers). Depending on the signal strength, the "round trip" time increased. But I couldn't reproduce these persistent high timings.

mbratch, 2021-08-19:
@MarcelKr my experience seems fundamentally different. Running my tests, I typically get a < 1s response time, sometimes 1-2s. But once in awhile (average a couple of times a day maybe) it will be 30s or more. I'm running a script now and monitoring it versus some of the suggested settings to see what happens.

MarcelKr, 2021-08-19:
@mbratch OK that's interesting. Having a script for tests an monitoring is a great approach. I hope the issue will be found this way.

mbratch, 2021-08-24:
@shubhamkulkarni97 I've been collecting data over many days. I have two ESP32 devices each running the same firmware except one of them has the wifi power saving turned off via `esp_wifi_set_ps(WIFI_PS_NONE);`. I have two scripts running on a Linux server, each sending an HTTP request every 2-3 seconds to its respective ESP32 device and I record the response time if it's greater than 1 second. In the case where the power savings is default, I get numerous response delays per day that are > 30s and some > 60s. For the unit that has the power savings turned off, I have gotten several > 1s or maybe 2s, and one that was 16s. I'm puzzled as to why wifi power saving would affect this since I'm regularly interacting with the devices over wifi, so they shouldn't be going into any kind of power save mode I wouldn't think.

mbratch, 2021-08-29:
I've been running my comparison tests for some time now. In this test I have 3 independent scripts on a server sending an HTTP REST command to its respective ESP32.

There were two different settings recommended:

- `esp_wifi_set_ps(WIFI_PS_NONE)`
- `lru_purge_enable = true`

I have one ESP32 setup with neither of these set. One with just the power savings turned off, and another with both of these set. (I did an earlier test with just the second set and didn't really notice a difference on average.)

What I've found is that without these settings, I get frequent delays (several per day) which are between 16 seconds and as high as 60 seconds in some cases. If I turn off power savings, but leave `lru_purge_enable` set to `false`, I will get an occasional delay of ~15 seconds or more (maybe once per day or a bit less). If I utilize both these settings, I don't see any delay more than about 2 seconds or so.

shubhamkulkarni97, 2021-08-30:
> @shubhamkulkarni97 with modem sleep the average response time will be higher but DTIM is probably 300ms so 30s is really bad

@negativekelvin The response time might also be dependent on the implementation of AP. However, I am not the right person to comment about high response time when WiFi sleep mode is enabled. You can create a new WiFi specific issue for this.

> What I've found is that without these settings, I get frequent delays (several per day) which are between 16 seconds and as high as 60 seconds in some cases. If I turn off power savings, but leave [...]

@mbratch  Good to know that you are not observing this issue after applying both the settings

mbratch, 2021-09-03:
@shubhamkulkarni97 Ironically, over the last day or two, it's been the reverse: the unit without those settings has been performing with consistency low delay, whilst the other two with the settings have had several occurrences of high delay. Very strange.

shubhamkulkarni97, 2021-09-28:
@mbratch Are you still observing this issue?

mbratch, 2021-09-28:
@shubhamkulkarni97 using the two settings recommended above, the issue has been greatly reduced.