espressif/esp-idf #7329: Users should be able to specify the name of their Kconfig.projbuild files

daltonv, 2021-07-26:
## Problem Environment
* esp-idf v4.2
* Building in espressif/idf docker image on a windows 10 host
* Building with idf.py

## Problem Description
The lack of this feature poses a unique issue when trying to build projects the specify their own Kconfig.projbuild files from a docker image with the project mounted from a windows file system. This is because the windows file system is not case sensitive but esp-idf is searching specifically for config files called "Kconfig.projbuild" files with and upper case "K". Since the docker image is running under linux but the project is mounted on a windows file system esp-idf is unable to locate my local "kconfig.projbuild" file. 

## Current Problem Solutions
* Developers could have their project in the wsl2 file system which is a linux file system. This is not ideal in some circumstances as some windows programs have a hard time accessing this file system.
* Funnily enough you can rename files in windows with a capital letter and it was reflected in the docker mount, but this doesn't play nice with version control systems
* The solution I used was to override the `__kconfig_component_init` function in my local CMakeList.txt file for my project. This can be seen below:
```cmake
#
# Special code for Docker. This overides the function of the same name in
# the $IDF_PATH/tools/cmake/kconfig.cmake. The only difference is that we
# also search for kconfig.projbuild files with a lower case 'k'. This is
# necessarry when we are building from the docker image against a mount on
# a windows filesystem as windows isn't case sensitive.
#
function(__kconfig_component_init component_target)
    __component_get_property(component_dir ${component_target} COMPONENT_DIR)
    file(GLOB kconfig "${component_dir}/Kconfig")
    __component_set_property(${component_target} KCONFIG "${kconfig}")
    file(GLOB kconfig "${component_dir}/Kconfig.projbuild"
                      "${component_dir}/kconfig.projbuild"
        )
    __component_set_property(${component_target} KCONFIG_PROJBUILD "${kconfig}")
    file(GLOB sdkconfig_rename "${component_dir}/sdkconfig.rename")
    __component_set_property(${component_target} SDKCONFIG_RENAME "${sdkconfig_rename}")
endfunction()
```

## Proposed Problem Solution
Allow users to pass in a custom name for their local Kconfig.projbuild file in the project. It could work like how users can already specify a name for the sdkconfig.defaults file by setting and env var in their CMakeLists.txt file. Something like shown below:
```cmake
set(KCONFIG_NAME kconfig.projbuild)
```

igrr, 2021-07-26:
Hi @daltonv, could you please clarify whether you it is the renaming of `kconfig.projbuild` to `Kconfig.projbuild` that is a problem for you? If you had a way to rename the file to `Kconfig.projbuild` in your Windows/WSL setup, would the issue be solved?

> Funnily enough you can rename files in windows with a capital letter and it was reflected in the docker mount, but this doesn't play nice with version control systems

Which version control system are you having a problem with when changing the file name case? With `git` it is possible by doing `git mv kconfig.projbuild Kconfig.projbuild`, with optional `-f` flag if you have modified the file.

daltonv, 2021-07-26:
@igrr, I am using Surround SCM right now for versioning, but I do plan to move to git soon. 

Renaming the file works for my local clone of my project, but this becomes an issue with the version control systems. I am aware git it case sensitive, but when cloning onto windows I see that even if the file is tracked as `Kconfig.projbuild` by git I still get a local `kconfig.projbuild`. 

I also don't really understand why the rename works in the first place. It appears to only be a visual rename in windows, because if I add another `kconfig.projbuild` file in a dir with a `Kconfig.projbuild` windows asks me if I want to replace the file. Somehow the docker mount treats this visual rename as an actual rename in the docer container, but I don't know if that would always work. Especially since I can't use the linux `mv` command to rename it inside the container.

In the end it would be ideal if my project could use filenames that are OS cross compatible, as even if I could get git to clone the files in the right case it would be impossible to use to have to files that only differ by case. Windows couldn't differentiate them.

igrr, 2021-07-26:
I'm afraid that if the SCM converts file names to lowercase when checking out the files, then Kconfig.projbuild will not be the only problem that we will have to solve. There aren't a lot of files with capital letters in their names in IDF, but at least a few are present in various 3rd party libraries, such as Segger's SystemView library.
There should probably be a way to configure your SCM to keep the file names the way they are supposed to be, when checking out the files. With git you can do this using `git config core.ignorecase false`. I'm not familiar with Surround SCM to say if there's a way to do the same there, I'm sorry.

> as even if I could get git to clone the files in the right case it would be impossible to use to have to files that only differ by case

Is there an actual use case you see for having files with names differing only in their case in the same directory? I.e. if you can add `Kconfig.projbuild` to the repository with the correct case, why would you need to also add `kconfig.projbuild`?

daltonv, 2021-07-27:
Ah, my apologizes for my git ignorance. I tested this again with `git config core.ignorecase false` and I am able to preserve the case now. 

This does seem like the easiest solution for everyone.

igrr, 2021-07-27:
Good to know that it did the trick!

closed 2021-07-27 by igrr