espressif/esp-idf #7330: esp-idf https_server not working on any ports but 80 or 443

schaffster, 2021-07-26:
I was developing using esp-idf 3.1 and coded an and implemented an HTTP and HTTPS server using the socket and mbedtls libraries (I was able to use any port in either HTTP or HTTPS mode).  I switched to esp-idf 4.3 and started using the HTTPS Server component.  Using the HTTPS server I can only get ports 443 when conf.transport.mode=HTTPD_SSL_TRANSPORT_SECURE and 80 when conf.transport.mode=HTTPD_SSL_TRANSPORT_INSECURE to work.  No other ports ever work. My code is below followed by error messages I get when I change either server to a non standard port and attempt a connection from a browser.
 
My HTTP Server Code:

#include <stdio.h>
#include <string.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "freertos/event_groups.h"
#include "driver/gpio.h"
#include "driver/adc.h"
#include "esp_system.h"
#include "esp_event.h"
#include "esp_adc_cal.h"
#include "esp_log.h"
#include "esp_err.h"
#include "esp_netif.h"
#include <esp_https_server.h>
#include <nvs_flash.h>
#include "uSwitchStructure.h"

static httpd_handle_t server=NULL;
extern esp_err_t HTTPGet(httpd_req_t *req);
	// Persistent sockets, context maintained across multiple requests on that socket.
static const char *TAG="HTTP";

httpd_uri_t root_get			={.uri="/",					.method=HTTP_GET,.handler=HTTPGet,.user_ctx=""};
httpd_uri_t userLogin_get		={.uri="/userLogin",		.method=HTTP_GET,.handler=HTTPGet,.user_ctx="userLogin"};
httpd_uri_t adminLogin_get		={.uri="/adminLogin",		.method=HTTP_GET,.handler=HTTPGet,.user_ctx="adminLogin"};
httpd_uri_t control_get			={.uri="/controlPage",		.method=HTTP_GET,.handler=HTTPGet,.user_ctx="controlPage"};

httpd_handle_t stop_HTTPServer(httpd_handle_t server,bool bHTTPS) {
	if (server!=NULL) {
		httpd_ssl_stop(server);
	}
	return(NULL);
}
httpd_handle_t start_HTTPServer(bool bHTTPS){
	httpd_handle_t server;
    httpd_ssl_config_t conf=HTTPD_SSL_CONFIG_DEFAULT();
    conf.httpd.stack_size=4096*3;
    conf.httpd.max_uri_handlers=40;
    conf.port_secure=uSwitch.Net.iHTTPSPort;		// 443
    conf.port_insecure=uSwitch.Net.iHTTPPort;		// 80
	conf.transport_mode=bHTTPS?HTTPD_SSL_TRANSPORT_SECURE:HTTPD_SSL_TRANSPORT_INSECURE;
    if (bHTTPS) {
        extern const unsigned char cacert_pem_start[] asm("_binary_cacert_pem_start");
		extern const unsigned char cacert_pem_end[]   asm("_binary_cacert_pem_end");
		conf.cacert_pem=cacert_pem_start;
		conf.cacert_len=cacert_pem_end-cacert_pem_start;

		extern const unsigned char prvtkey_pem_start[] asm("_binary_prvtkey_pem_start");
		extern const unsigned char prvtkey_pem_end[]   asm("_binary_prvtkey_pem_end");
		conf.prvtkey_pem=prvtkey_pem_start;
		conf.prvtkey_len=prvtkey_pem_end-prvtkey_pem_start;
	}
    if (httpd_ssl_start(&server,&conf)==ESP_OK) {
        httpd_register_uri_handler(server,&adminLogin_get);
        httpd_register_uri_handler(server,&userLogin_get);
        httpd_register_uri_handler(server,&root_get);
        httpd_register_uri_handler(server,&control_get);
    } else {
    	server=stop_HTTPServer(server,bHTTPS);
    	ESP_LOGI(TAG,"Error starting HTTP server!");
    }
    return(server);
}

If I'm setting an HTTP server (No SSL) and specify any port other than port 80 the console gives the following errors (no problems on port 80):

W (9275) httpd_txrx: httpd_sock_err: error in send : 104
E (9275) esp32: processHTTPWriteError(HTTPD_SOCK_ERR_FAIL)
W (9285) httpd_txrx: httpd_sock_err: error in send : 104
E (9285) esp32: processHTTPWriteError(HTTPD_SOCK_ERR_FAIL)
W (9295) httpd_txrx: httpd_sock_err: error in send : 128
E (9295) esp32: processHTTPWriteError(HTTPD_SOCK_ERR_FAIL)
W (9305) httpd_txrx: httpd_sock_err: error in send : 128
E (9315) esp32: processHTTPWriteError(HTTPD_SOCK_ERR_FAIL)
W (9315) httpd_txrx: httpd_sock_err: error in send : 128
E (9325) esp32: processHTTPWriteError(HTTPD_SOCK_ERR_FAIL)
W (9325) httpd_txrx: httpd_sock_err: error in send : 128

If I'm setting an HTTPS server and specify any port other than port 443 the console gives the following errors (no problems on port 443):

W (50574) esp-tls-mbedtls: write error : -80:
E (50574) esp32: processHTTPWriteErrror(HTTPD_SOCK_UNKNOWN)
W (50574) esp-tls-mbedtls: write error : -78:
E (50584) esp32: processHTTPWriteErrror(HTTPD_SOCK_UNKNOWN)
W (50584) esp-tls-mbedtls: write error : -78:
E (50594) esp32: processHTTPWriteErrror(HTTPD_SOCK_UNKNOWN)
...
E (61214) esp32: processHTTPWriteError(HTTPD_SOCK_UNKNOWN)
E (61214) Dynamic Impl: mbedtls_ssl_fetch_input error=-0x50
E (61224) esp-tls-mbedtls: read error :-80:
I (90424) esp_https_server: performing session handshake
E (91374) esp-tls-mbedtls: mbedtls_ssl_handshake returned -30592
E (91374) esp_https_server: esp_tls_create_server_session failed
W (91384) httpd: httpd_accept_conn: session creation failed
W (91384) httpd: httpd_server: error accepting new connection
I (91394) esp_https_server: performing session handshake
E (92464) esp-tls-mbedtls: mbedtls_ssl_handshake returned -30592
E (92464) esp_https_server: esp_tls_create_server_session failed
W (92464) httpd: httpd_accept_conn: session creation failed
W (92474) httpd: httpd_server: error accepting new connection
I (92474) esp_https_server: performing session handshake
I (94164) esp32: Got Browser Command GET / HTTP/1.1

ESP-YJM, 2021-07-27:
@schaffster I use tag v4.3 version https_server example to test and it is ok. Please check if your network firewall disables the http protocol other ports except 80 and 443.

Alvin1Zhang, 2021-08-09:
@schaffster c

closed 2021-08-09 by Alvin1Zhang