espressif/esp-idf #7331: Can't connect to WiFi with space in ssid

hoatenhu, 2021-07-27:
Hi everyone, can you help me with this problem:
- I can't connect with ssid when it has space
- I try another version but it doesn't work
- But It can connect with ssid have no space
**1. When ssis has space:**
[image]
[image]

https://user-images.githubusercontent.com/84147184/127125709-ddeffc70-b258-4161-b3ba-c2cc923a9cd0.mp4
[image]

**2. When no space**
[image]
[image]

ESP-YJM, 2021-07-28:
@hoatenhu Could you please offer your IDF version you used. I can't reproduce your issue.

hoatenhu, 2021-07-28:
Hi,
I tried IDF 4.1, 4.2, 4.3, also checkout remote/origin/release 4.1(4.2, 4.3 ) in git, and it doesn't work.
Now I'm using 4.2
[image]

hoatenhu, 2021-07-28:
I am using esp32 Wrover, does this have any effect on this error?

ESP-YJM, 2021-07-28:
I think you can run wifi sta example to test it.

hoatenhu, 2021-07-28:
This is when I run wifi sta example:
[image]

hoatenhu, 2021-07-28:
And when without space:
[image]

ESP-YJM, 2021-07-28:
@hoatenhu Is ssid `te` and `Y Nhi` used the same router AP？

hoatenhu, 2021-07-28:
yepp, I also try with ssid: `te_1` and it has the same error with ssid: `Y Nhi`

ESP-YJM, 2021-07-28:
@hoatenhu I think this ssid is ok to connect. Please check your AP wifi channel, need in 1-13 and only support 2.4G.

hoatenhu, 2021-07-28:
It still error

[image]

ESP-YJM, 2021-07-28:
@hoatenhu It's ok for me with same ssid and password. Or can you erase your flash and re-flash fw to test? Or capture wifi packet.
[image]

hoatenhu, 2021-07-28:
It works with passwords of fewer than 13 characters, what if I want the password to have more characters?
[image]

AxelLin, 2021-07-29:
> It works with passwords of fewer than 13 characters, what if I want the password to have more characters?

I don't get it.
This does not match https://github.com/espressif/esp-idf/issues/7331#issuecomment-888232386 which shows faiure with passoword:10031998 (obviously less than 13 characters)
So what was changed to make it work?

hoatenhu, 2021-07-29:
I checked out v4.3, therefore https://github.com/espressif/esp-idf/issues/7331#issuecomment-888232386  work with less 13 characters

ESP-YJM, 2021-07-29:
@hoatenhu I also use tag v4.2 version. And it's ok for space ssid and more than 13 characters password.

hoatenhu, 2021-07-29:
So why I can connect with password `123456789012`, and not connect with password `1234567890123`, I can't understand, I use v4.3 now
[image]

ESP-YJM, 2021-07-29:
@hoatenhu Do you have another AP router or mobile hotspot? You can change it and test.

hoatenhu, 2021-07-29:
but why I can't connect with this password `thanhtuyennhuy`, I use v4.3 now
[image]

AxelLin, 2021-07-29:
> but why I can't connect with this password `thanhtuyennhuy`, I use v4.3 now

Please modify the example code to print disconnect reason when connect the AP fail.

hoatenhu, 2021-07-29:
Can you help me how to modify it? I don't know where
[image]

hoatenhu, 2021-07-29:
with pass `thanhtuyen` it works, If I add 1 more character, it won't work, ex `thanhtuyenn`, `thanhtuyennh`, ..
[image]

ESP-YJM, 2021-07-29:
@hoatenhu  You can follow this to print wifi connect error reason. https://github.com/espressif/esp-idf/blob/master/components/wifi_provisioning/src/manager.c#L864 . Besides, do you try my suggestion https://github.com/espressif/esp-idf/issues/7331#issuecomment-888762203 to test?

hoatenhu, 2021-07-29:
[image]
[image]

hoatenhu, 2021-07-29:
here is if I add `6` in password

[image]

hoatenhu, 2021-07-29:
[image]
[image]

AxelLin, 2021-07-29:
@hoatenhu 

Can you share the brand / model of this AP?
Also share the AP setting screenshot if possible.

hoatenhu, 2021-07-29:
I use mobile hotspot of my Window laptop for esp now
and this is wifi I want esp to connect to( my home wifi)
[image]

ESP-YJM, 2021-07-29:
@hoatenhu WIFI connection error reason 2 is auth fail, maybe password is mismatch. You can use another device to connect AP.

hoatenhu, 2021-07-29:
and what about this error
[image]

ESP-YJM, 2021-07-29:
@hoatenhu You can find reason in this file. https://github.com/espressif/esp-idf/blob/master/components/esp_wifi/include/esp_wifi_types.h#L99

hoatenhu, 2021-07-29:
ok, thank you so much for helping me, I think the problem is my AP router

ESP-YJM, 2021-07-29:
Ok, please help close this issue if no other problem.

hoatenhu, 2021-07-29:
ok, bye

closed 2021-07-29 by hoatenhu