espressif/esp-idf #7332: AES CTR DECRYPTION gives wrong output when CONFIG_MBEDTLS_HARDWARE_AES is disabled.

Oquirella, 2021-07-27:
**Output when CONFIG_MBEDTLS_HARDWARE_AES is enabled:**
I (470354) ISC: ORIGINAL

I (470358) ISC: 0x00000000: 00 01 00 00 00 00 00 00     .  . . . . . .

I (470365) ISC: 0x00000008: 00 00                       . .            

I (470372) ISC: ENCRYPTED

I (470376) ISC: 0x00000000: 21 28 d0 67 58 eb 1a 11        g X   

I (470383) ISC: 0x00000008: 10 c0                            
        
I (470390) ISC: DECRYPTED
I (470358) ISC: 0x00000000: 00 01 00 00 00 00 00 00     .  . . . . . .

I (470365) ISC: 0x00000008: 00 00                       . .            

**Output when CONFIG_MBEDTLS_HARDWARE_AES is disabled:**
`
I (470354) ISC: ORIGINAL

I (470358) ISC: 0x00000000: 00 01 00 00 00 00 00 00     .  . . . . . .

I (470365) ISC: 0x00000008: 00 00                       . .            

I (470372) ISC: ENCRYPTED

I (470376) ISC: 0x00000000: 21 28 d0 67 58 eb 1a 11        g X   

I (470383) ISC: 0x00000008: 10 c0                            
        
I (470390) ISC: DECRYPTED

I (470393) ISC: 0x00000000: 8b 3e 9a 62 98 5b d9 d1        b    

I (470401) ISC: 0x00000008: 9b b9
`

Alvin1Zhang, 2021-07-27:
@Oquirella Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

AxelLin, 2021-07-27:
I observed similiar issue with mbedtls_aes_crypt_cbc() decryption.
When CONFIG_MBEDTLS_HARDWARE_AES=y, the decrypted data matches the original data.
When CONFIG_MBEDTLS_HARDWARE_AES=n, the decrypted data does not match the original data.

I test on ESP32-WROOM-32E, v4.3-242-g1f7172dbf968.

ESP-Marius, 2021-07-28:
@Oquirella Thanks for the report.

Can you post a simple example showing how you are using the API? Probably there is no problem with the mbedtls API itself, but if the behaviour is different than the HW implementation then that's something we definitely need to look into.

AxelLin, 2021-07-28:
@ESP-Marius 

Below is the code snip to show the issue for mbedtls_aes_crypt_cbc:
https://gist.github.com/AxelLin/41451f2e82da78df2a394155a5b7aa9d

You just need to call aes_128_cbc_test().
When When CONFIG_MBEDTLS_HARDWARE_AES=y, it prints "MATCH"
When When CONFIG_MBEDTLS_HARDWARE_AES=n, it prints "MISMATCH"

ESP-Marius, 2021-07-28:
@AxelLin Thanks for a super quick feedback (as always!)

I took a quick look and it seems like the issue in that example is that `mbedtls_aes_setkey_dec` is not called before decryption.

For the HW acceleration these two functions both map to the same setkey function so when HW acceleration is enabled missing this call doesnt matter. But when using the Mbedtls software implementation they seem to be mandatory.

@Oquirella Please help check if this also solves your issue.

AxelLin, 2021-07-28:
> @AxelLin Thanks for a super quick feedback (as always!)
> 
> I took a quick look and it seems like the issue in that example is that `mbedtls_aes_setkey_dec` is not called before decryption.
> 
@ESP-Marius 
Thanks, I confirmed adding mbedtls_aes_setkey_dec() then I got MATCH.

mahavirj, 2021-08-04:
@Oquirella Closing this issue, however if you are still facing problem please feel free to reopen with test code and details per github issue template

closed 2021-08-04 by mahavirj