espressif/esp-idf #7334: Chunked response in chunks

silardg, 2021-07-27:
## Environment

- Module or chip used: ESP32-WROOM-32D
- Using an IDE?:Vscode Platformio V4.2 IDF

I'm expecting (getting this result with cURL)
```
4d
{"id":581,"config":{"weekdays":[1,2,3,4,5,6,7],"start":45240,"duration":600}}
```
Getting:
```
I (4176) HTTP REQUEST: CHUNKED, len=72
E (4180) DEBUG: {"id":581,"config":{"weekdays":[1,2,3,4,5,6,7],"start":45240,"duration":
I (4190) HTTP REQUEST: CHUNKED, len=5
E (4193) DEBUG: 600}}
```

This is my http event:
```
case HTTP_EVENT_ON_DATA:                                                                                // even on data receive
            if (!esp_http_client_is_chunked_response(evt->client))                                              // if the response is not chunked (in more packets)
            {
                info(TAG, "HTTP_EVENT_ON_DATA, len=%d", evt->data_len);
                // Write out data
                receivedBuffer.append((char*)evt->data,evt->data_len);                                            // append the data in the config buffer
            } else {
                
                info(TAG, "CHUNKED, len=%d", evt->data_len);

                std::string holder((char*)evt->data,evt->data_len);
                debug("%s", holder.c_str());
```

Buffer:
```
this->httpConfig.buffer_size                = DEFAULT_HTTP_BUF_SIZE * 4;   
    this->httpConfig.buffer_size_tx             = DEFAULT_HTTP_BUF_SIZE * 4; 
```

ESP-YJM, 2021-07-28:
@silardg Maybe LWIP receive window is full and only store part data, then read only read the part data,then left data comes to second read. You can enlarge the `CONFIG_LWIP_TCP_WND_DEFAULT` and test. But i think it is ok. Because we will append the data in `HTTP_EVENT_ON_DATA` event and be sure that all data received in `HTTP_EVENT_ON_FINISH` event.

silardg, 2021-07-28:
> @silardg Maybe LWIP receive window is full and only store part data, then read only read the part data,then left data comes to second read. You can enlarge the `CONFIG_LWIP_TCP_WND_DEFAULT` and [...]

This was my guess too, but wanted to make sure. Checking and reporting back.

silardg, 2021-07-29:
Yes, the fix is to increase that buffer size and also have a bit more heap available

closed 2021-07-29 by silardg