espressif/esp-idf #7337: Unable to burn flash encryption key

tsctrl, 2021-07-28:
idf 4.3, esp devkit v1 with universal serial bus connection and windows

1. no modification on any secure configuration (all disabled as default)
2. run this command to generate key: espsecure.py generate_flash_encryption_key abc/key.bin
3. burn the key: espefuse.py --port port burn_key flash_encryption abc/key.bin

error:
```
Connecting........_
Detecting chip type... ESP32
espefuse.py v3.1-dev
Burn keys to blocks:
 - BLOCK1 -> [8e . .. . . .. 5f]
        Reversing the byte order
A fatal error occurred:         BLOCK1 is read-protected. The written value can not be read, the efuse/block looks as all 0.
        Burn in this case may damage an already written value.(use '--force-write-always' option to ignore it)
```
why? saw others done this command with flying colors...

thanks!

negativekelvin, 2021-07-28:
What is the output of espefuse.py --port port summary

tsctrl, 2021-07-28:
Detecting chip type... ESP32
espefuse.py v3.1-dev
EFUSE_NAME (Block)                       Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Calibration fuses:
BLK3_PART_RESERVE (BLOCK0):              BLOCK3 partially served for ADC calibration data   = False R/W (0b0)
ADC_VREF (BLOCK0):                       Voltage reference calibration                      = 1072 R/W (0b10100)

Config fuses:
XPD_SDIO_FORCE (BLOCK0):                 Ignore MTDI pin (GPIO12) for VDD_SDIO on reset     = False R/W (0b0)
XPD_SDIO_REG (BLOCK0):                   If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset    = False R/W (0b0)
XPD_SDIO_TIEH (BLOCK0):                  If XPD_SDIO_FORCE & XPD_SDIO_REG                   = 1.8V R/W (0b0)
CLK8M_FREQ (BLOCK0):                     8MHz clock freq override                           = 55 R/W (0x37)
SPI_PAD_CONFIG_CLK (BLOCK0):             Override SD_CLK pad (GPIO6/SPICLK)                 = 0 R/W (0b00000)
SPI_PAD_CONFIG_Q (BLOCK0):               Override SD_DATA_0 pad (GPIO7/SPIQ)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_D (BLOCK0):               Override SD_DATA_1 pad (GPIO8/SPID)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_HD (BLOCK0):              Override SD_DATA_2 pad (GPIO9/SPIHD)               = 0 R/W (0b00000)
SPI_PAD_CONFIG_CS0 (BLOCK0):             Override SD_CMD pad (GPIO11/SPICS0)                = 0 R/W (0b00000)
DISABLE_SDIO_HOST (BLOCK0):              Disable SDIO host                                  = False R/W (0b0)

Efuse fuses:
WR_DIS (BLOCK0):                         Efuse write disable mask                           = 128 R/W (0x0080)
RD_DIS (BLOCK0):                         Efuse read disable mask                            = 1 R/W (0x1)
CODING_SCHEME (BLOCK0):                  Efuse variable block length scheme
   = NONE (BLK1-3 len=256 bits) R/W (0b00)
KEY_STATUS (BLOCK0):                     Usage of efuse block 3 (reserved)                  = False R/W (0b0)

Identity fuses:
MAC (BLOCK0):                            Factory MAC Address
   = ..... R/W
MAC_CRC (BLOCK0):                        CRC8 for factory MAC address                       = 48 R/W (0x30)
CHIP_VER_REV1 (BLOCK0):                  Silicon Revision 1                                 = True R/W (0b1)
CHIP_VER_REV2 (BLOCK0):                  Silicon Revision 2                                 = False R/W (0b0)
CHIP_VERSION (BLOCK0):                   Reserved for future chip versions                  = 2 R/W (0b10)
CHIP_PACKAGE (BLOCK0):                   Chip package identifier                            = 0 R/W (0b000)
MAC_VERSION (BLOCK3):                    Version of the MAC field                           = 0 R/W (0x00)

Security fuses:
FLASH_CRYPT_CNT (BLOCK0):                Flash encryption mode counter                      = 0 R/W (0b0000000)
UART_DOWNLOAD_DIS (BLOCK0):              Disable UART download mode (ESP32 rev3 only)       = False R/W (0b0)
FLASH_CRYPT_CONFIG (BLOCK0):             Flash encryption config (key tweak bits)           = 0 R/W (0x0)
CONSOLE_DEBUG_DISABLE (BLOCK0):          Disable ROM BASIC interpreter fallback             = True R/W (0b1)
ABS_DONE_0 (BLOCK0):                     Secure boot V1 is enabled for bootloader image     = False R/W (0b0)
ABS_DONE_1 (BLOCK0):                     Secure boot V2 is enabled for bootloader image     = False R/W (0b0)
JTAG_DISABLE (BLOCK0):                   Disable JTAG                                       = False R/W (0b0)
DISABLE_DL_ENCRYPT (BLOCK0):             Disable flash encryption in UART bootloader        = False R/W (0b0)
DISABLE_DL_DECRYPT (BLOCK0):             Disable flash decryption in UART bootloader        = False R/W (0b0)
DISABLE_DL_CACHE (BLOCK0):               Disable flash cache in UART bootloader             = False R/W (0b0)
BLOCK1 (BLOCK1):                         Flash encryption key
   = ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? -/-
BLOCK2 (BLOCK2):                         Secure boot key
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLOCK3 (BLOCK3):                         Variable Block 3
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Flash voltage (VDD_SDIO) determined by GPIO12 on reset (High for 1.8V, Low/NC for 3.3V).

negativekelvin, 2021-07-29:
It looks like you already successfully burned the key however you would have to enable encryption to verify whether it matches the key you wanted to burn.

tsctrl, 2021-07-29:
ya, look like. but i never get successful burn message before.

KonstantinKondrashov, 2021-07-29:
@tsctrl I see that it was like this: the first run `espefuse.py --port port burn_key flash_encryption abc/key.bin` sets the key and protection bits; the second run the same command gives the error as you show.

if you have the log from the first attempt please share it.

tsctrl, 2021-07-29:
cant recall if i have done it before. i guess i exidentally burn it somehow. is it possible to override the key with new key? any impact?

KonstantinKondrashov, 2021-07-29:
if the key is write-protected then it is not possible to burn a new key https://github.com/espressif/esptool/wiki/espefuse#read--and-write--protecting-efuses, the tool will show the error like above. (hardware blocks this operation anyway).

tsctrl, 2021-07-29:
ok, tried the same with my other kits and it burn successfully, 
i guess, i got too excited, burn the keys and forgotten doing it, lesson learned.

closing this, thanks!

closed 2021-07-29 by tsctrl