espressif/esp-idf #7338: suspicious  of memory leak in  socket scene

hiproz, 2021-07-29:
chip: esp32c3
sdk: ESP-IDF v4.4-dev-1849-g8e3e65a-dirty

app process:
1. init ap+sta
2. init uart
3.  loop: send data to udp multicast from uart
4. connect some pc to ap , fresh the chrome page some times

there is NOMEM  error in running:
 > IPV4 sendto failed. errno: 12

i suspect that there is some memory leak in lwip lib:

the begins heapsize log is   90-90K:
> Minimum free heap size: 87468 bytes

the last  heapsize log is  about 8K:
>  Minimum free heap size: 8608 bytes

at last ,the syetem is panic:
···
W (153263610) Anch-wifi: Minimum free heap size: 8608 bytes

W (153266610) Anch-wifi: Minimum free heap size: 8608 bytes

W (153269610) Anch-wifi: Minimum free heap size: 8608 bytes

W (153272610) Anch-wifi: Minimum free heap size: 8608 bytes

Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x4003fc94  RA      : 0x4003fc86  SP      : 0x3fca2f80  GP      : 0x3fc91800
TP      : 0x3fc7258c  T0      : 0x6e52626d  T1      : 0x4000133c  T2      : 0x7673337a
S0/FP   : 0x3fccaf68  S1      : 0x3fce0000  A0      : 0x00450008  A1      : 0x3fca2f60
A2      : 0x00000000  A3      : 0x3fc9221c  A4      : 0x067f0040  A5      : 0x3fca0b80
A6      : 0x02000000  A7      : 0xaf5283ae  S2      : 0x00000002  S3      : 0x00000001
S4      : 0x3ff1f000  S5      : 0x00000028  S6      : 0x00000004  S7      : 0x3c12c000
S8      : 0x00000002  S9      : 0x3fce0000  S10     : 0x3fcdf934  S11     : 0x00000000
T3      : 0x3fc969f0  T4      : 0xbfffffff  T5      : 0x80000000  T6      : 0x346d4f47
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x067f0040
MHARTID : 0x00000000

Stack memory:
3fca2f80: 0x3c12ba90 0x3fce0000 0x3fcdf284 0x4003dc0e 0x00000001 0x00000000 0x3ff1f000 0x00000001
3fca2fa0: 0x3c12ba90 0x00000000 0x00000002 0x420ba13e 0x3fce0000 0x3fce0000 0x00000000 0x00e30000
3fca2fc0: 0x3fcdf934 0x3fce0000 0x3ff1b000 0x3fce0000 0x3fce0000 0x3fce0000 0x3fce0000 0x3ff1b5a8
3fca2fe0: 0x00000000 0x3fcdf918 0x3fce0000 0x400408b8 0x00000000 0x00000000 0x00000017 0x00000000
3fca3000: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca3020: 0x00000000 0x00000000 0x00000000 0x4038c1d2 0x00000000 0x00000000 0x00000000 0x00000000
3fca3040: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x27146414 0x01010107 0x00000001 0x000001ff
3fca3060: 0x62746c65 0x000000ff 0x7f7fffff 0x00000000 0x3fe00000 0x00000000 0x3ff00000 0x00000000
3fca3080: 0x40000000 0xffc00000 0x41dfffff 0x94a03595 0x3fdfffff 0x35afe535 0x3fe00000 0x94a03595
3fca30a0: 0x3fcfffff 0xffffffff 0x7fefffff 0x00000000 0x3fc00000 0x00000000 0x40300000 0x00000000
3fca30c0: 0x3ff80000 0x636f4361 0x3fd287a7 0x8b60c8b3 0x3fc68a28 0x509f79fb 0x3fd34413 0x00000000
3fca30e0: 0x40240000 0x00000000 0x401c0000 0x00000000 0x40140000 0x00000000 0x43500000 0x00000000
3fca3100: 0x7ff80000 0x7fc00000 0x3fc91000 0x000033f0 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x0000015c
3fca3120: 0x3fca2f10 0x3fc98a74 0x3fc94e14 0x3fc94e14 0x3fca3120 0x3fc94e0c 0x00000002 0x3fca1048
3fca3140: 0x3fca1048 0x3fca3120 0x00000000 0x00000017 0x3fca171c 0x69666977 0xc7adc200 0xeab7d2b0
3fca3160: 0x00a358ca 0x00000000 0x3fca3110 0x00000007 0xdd01731b 0x00000017 0x00000000 0x00000000
3fca3180: 0x00000000 0x00000000 0x3fc9928c 0x3fc992f4 0x3fc9935c 0x00000000 0x00000000 0x00000001
3fca31a0: 0x00000000 0x00000000 0x00000000 0x420dad04 0x00000000 0x00000000 0x00000000 0x00000000
3fca31c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca31e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca3200: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca3220: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca3240: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca3260: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x3fca0000 0x0000005c
3fca3280: 0x3fca3280 0x3fca3280 0x3fca3280 0x3fca3280 0x00000000 0x3fca3298 0xffffffff 0x3fca3298
3fca32a0: 0x3fca3298 0x00000000 0x3fca32ac 0xffffffff 0x3fca32ac 0x3fca32ac 0x00000000 0x00000001
3fca32c0: 0x00000000 0xce00ffff 0x00000000 0x541207da 0x6fef9e02 0x78e948f7 0x3fca3278 0x000007ec
3fca32e0: 0x6f6d706f 0x00006564 0x2e617473 0x64697373 0x00000000 0x2e617473 0x68747561 0x65646f6d
3fca3300: 0x00010000 0x00000000 0x00000003 0x00000002 0x3fc9633c 0x2e617473 0x64697373 0x00000000
3fca3320: 0x2e617473 0x68747561 0x65646f6d 0x00000000 0x2e617473 0x00240701 0x00000000 0x00000000
3fca3340: 0x00000000 0x3fc96340 0x2e617473 0x68747561 0x65646f6d 0x00000000 0x2e617473 0x64777370
3fca3360: 0x00000000 0x2e617473 0x00010002 0x00000001 0x00000007 0x00000001 0x3fc9636a 0x2e617473
···

hiproz, 2021-07-29:
i will do some test around sendto invokes refto :https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/heap_debug.html#how-to-diagnose-memory-leaks

hiproz, 2021-07-31:
i have nothing useful info of HEAP_TRACE_LEAKS process.  

when i open the  netif  and pbuf  debug info  of lwip , i got some related information :
this is the change point:
[image]

let we see what else happened of this change, i found some new  error:

[image]

```
[2021/7/31 3:09:25] pbuf_free(0x3fccc498)
[2021/7/31 3:09:25] pbuf_alloc(length=1506)
[2021/7/31 3:09:25] pbuf_alloc(length=1506) == 0x3fcd456c
[2021/7/31 3:09:25] pbuf_alloc(length=1506)
[2021/7/31 3:09:25] pbuf_alloc(length=1506) == 0x3fcd8754
[2021/7/31 3:09:25] pbuf_alloc(length=1506)
[2021/7/31 3:09:25] pbuf_alloc(length=1506) == 0x3fcd2ed4
[2021/7/31 3:09:25] pbuf_alloc(length=1506)
[2021/7/31 3:09:25] pbuf_alloc(length=1506) == 0x3fcd4b6c
[2021/7/31 3:09:25] pbuf_alloc(length=1506)
[2021/7/31 3:09:25] pbuf_alloc(length=1506) == 0x3fcd516c
[2021/7/31 3:09:25] pbuf_alloc(length=1506)
[2021/7/31 3:09:25] pbuf_alloc(length=1506) == 0x3fcd576c
[2021/7/31 3:09:25] pbuf_alloc(length=1506)
[2021/7/31 3:09:25] pbuf_alloc(length=1506) == 0x3fcd5d6c
[2021/7/31 3:09:25] pbuf_alloc(length=1506)
[2021/7/31 3:09:25] pbuf_alloc(length=1506) == 0x3fcd636c
[2021/7/31 3:09:25] pbuf_alloc(length=1506)
[2021/7/31 3:09:25] pbuf_alloc(length=1506) == 0x3fcd696c
[2021/7/31 3:09:25] pbuf_alloc(length=1506)
[2021/7/31 3:09:25] pbuf_alloc(length=1506) == 0x3fcd6f6c
[2021/7/31 3:09:25] pbuf_alloc(length=1506)
[2021/7/31 3:09:25] pbuf_alloc(length=1506) == 0x3fcdc294
[2021/7/31 3:09:25] pbuf_alloc(length=1506)
[2021/7/31 3:09:25] pbuf_alloc(length=1506) == 0x3fcd93fc
[2021/7/31 3:09:25] pbuf_alloc(length=1506)
[2021/7/31 3:09:25] pbuf_alloc(length=1506) == 0x3fcd7c14
[2021/7/31 3:09:25] pbuf_alloc(length=1506)
[2021/7/31 3:09:25] pbuf_alloc(length=1506) == 0x3fcd756c
[2021/7/31 3:09:25] ethernetif_input: IP input error
[2021/7/31 3:09:25] pbuf_free(0x3fcd756c)
[2021/7/31 3:09:25] pbuf_free: deallocating 0x3fcd756c
[2021/7/31 3:09:25] pbuf_alloc(length=1506)
[2021/7/31 3:09:25] pbuf_alloc(length=1506) == 0x3fcd756c
[2021/7/31 3:09:25] ethernetif_input: IP input error
```
i think maybe this error is the memory reduce reason.

hiproz, 2021-08-03:
After a long time of testing, I found that the memory reduction is and panic inevitable , but it is often maintained in a stable low stack memory state if the scene is stable. So I think this is a design risk of lwip, not a coding defect.

Does anyone have any suggestions? If not, this topic will be closed.

ESP-YJM, 2021-08-05:
@hiproz Thanks for sharing the very detailed problem location. It seems the upper layer deal with data to slow that the lower layer data full of mbox. This data will occupy the memory until it de dealed with. So i think it is very normal problem. You need to check the upper layer logic to deal with this data faster or reduce the mbox size(LWIP_TCPIP_RECVMBOX_SIZE) and reduce this may reduce some throughput, but will not occupy to much memory data.

hiproz, 2021-08-05:
1. i think the first point is that the udp multicast make the wifi  nat channel became slow ,but i can do nothing ,bcz  the udp speed is influenced  by the uart data flow. 
2. LWIP_TCPIP_RECVMBOX_SIZE =32 , LWIP_TCP_RECVMBOX_SIZE=6, LWIP_UDP_RECVMBOX_SIZE=6, what should i set ?

ESP-YJM, 2021-08-05:
You can reduce `LWIP_TCPIP_RECVMBOX_SIZE = 32` to smaller.

hiproz, 2021-08-05:
> You can reduce `LWIP_TCPIP_RECVMBOX_SIZE = 32` to smaller.

@ESP-YJM, i will try, thanks!

Alvin1Zhang, 2021-08-18:
@hiproz Thanks for reporting, would you please help share if any updates for the issue? Thanks.

hiproz, 2021-08-18:
@Alvin1Zhang I am still following up on this issue, the current phenomenon is still there, i need more test of it. if there is a new progress, I will post out, thank you！

Alvin1Zhang, 2021-08-18:
@hiproz Thanks for sharing the updates.

hiproz, 2021-08-19:
I can accept the fact that the memory is reduced. I continue to pay attention to this issue because it has been reducing and will not be stable, and finally caused the stack memory to run out, and then restart

closed 2021-09-03 by hiproz