espressif/esp-idf #7339: timing for I2S DMA queue events via esp_timer_get_time

sdrsdr, 2021-07-29:
I'm in a project where ADC via I2S DMA is used to track an analog signal. The signal must be correlated to other events. As you an imagine the timing is important and with tight limits. Using the I2S driver`s capability to send events via queue is almost perfect with the only downside is that the events might get variable time to get delivered. This can be easily solved by punting a timestamp from a call to esp_timer_get_time()  in the queue event. The only problem I see is keeping the code compatible with older users of the event queue.  To keep it clean this can be made selectable from the configuration system with default to non-timestamped events.
   I can create a pull request with such feature if it is OK by the maintainers.  Any suggestions/guides are welcome :)

ginkgm, 2021-07-30:
Hi @sdrsdr ,

I'm afraid due to the random SW delay caused by higher level interrupts, any way of adding SW timestamp will not be as acurate as the HW solution - calculate the time by the point number read from the queue.

Seems that you have tried to add a timestamp in the ISR? Have you tested your project with Wi-Fi on?

Michael

sdrsdr, 2021-08-02:
There seems to be a lot of "problems" in the I2S DMA code   like for some reason I2S_IN_DONE_INT is not monitored but just IN_SUC_EOF and some handled descriptors are lost plus data consumers don't have any chance handling the "ready" data untill all buffers are filled by DMA

ginkgm, 2021-08-03:
Hi @sdrsdr ,

Yes, that's why I think we cannot solve the issue from the SW side. The HW will not immediately store the data, and the driver use a less-frequenty interrupt to handle the finished data. Even if you put the code in to the ISR, I can still see there are some unexpected reason for the SW timestamp to be in accurate.

Michael

sdrsdr, 2021-08-03:
Hi @ginkgm, 
SW timestams accuracy in ISR is a matter of IRQ priorities. I don't know any HW way to take a timestamp.  lldesc_t sctruct is all the DMA care for and I don't see any field about timestams

ginkgm, 2021-08-04:
@sdrsdr 
If you can ensure the I2S queue is never full, then each data point should be sampled at a fixed interval, this is guaranteed by the timer.

This can give you a good relative time among samples.

sdrsdr, 2021-08-06:
Data handling from DMA is actually OK every DMA descriptor/buffer is marked as EOF so you get interrupt for every ready buffer. A buffer clock is missing and the way the read is implemented is quite inefficient. Also the initialization is practically missing and the hardware is initialized "from zero" every time "start" is issued using some global variables .. anyway I've started a huge rework of the current code as separate component centered at what I care: ADC sampling via DMA so I'm closing this issue now

closed 2021-08-06 by sdrsdr