espressif/esp-idf #7340: w5500-mac: no mem for receive buffer

Deepblue1996, 2021-07-30:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v2
- Module or chip used:  ESP32-WROVER-I
- IDF version (run ``git describe --tags`` to find it): v4.4
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: No
- Power Supply: USB

## Problem Description

Network cables Connect to the W5500 module，The IP address is obtained，The Tcp connection to the server succeeded. Procedure，This is what happens when you collect 200 or 300 pieces of data continuously，There seems to be a message out of memory，How to solve the memory problem，Follow this example address：esp-idf\examples\ethernet\basic

## Debug Logs

[0;31mE (82223) w5500-mac: no mem for receive buffer[0m
[0;31mE (82383) w5500-mac: no mem for receive buffer[0m
[0;31mE (83223) w5500-mac: no mem for receive buffer[0m
[0;31mE (83223) w5500-mac: no mem for receive buffer[0m
[0;31mE (83383) w5500-mac: no mem for receive buffer[0m
[0;31mE (84173) w5500-mac: no mem for receive buffer[0m

Hairy-Lee, 2021-08-12:
I have a similar issue running the eth2ap example...

- Development Kit: ESP32-S2-DevKitM-1
- Kit version (for WroverKit/PicoKit/DevKitC): 
- Module or chip used: ESP32-MINI-1
- IDF version (run git describe --tags to find it): v4.3
- Build System: idf.py
- Compiler version (run xtensa-esp32-elf-gcc --version to find it): 8.4.0
- Operating System: Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: Visual Studio Code
- Power Supply: USB

I can run the basic ethernet example successfully, but when I come to run the eth2ap example I am constantly faced with errors. My Wi-Fi station (e.g. mobile phone) connects to the ESP32's Wi-Fi

```
I (20521) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (20521) wifi:station: 8a:ad:60:d7:c7:10 join, AID=1, bgn, 20
I (20541) eth_example: Wi-Fi AP got a station connected
```

but immediately afterwards I am faced with `send flow control message failed or timeout`. So as per the troubleshooting guide associated with this example I have increased the value of `FLOW_CONTROL_QUEUE_LENGTH` (to 150) but this then changes to `E (23650) emac_esp32: no mem for receive buffer` I have tried tweaking some of the more obvious configuration settings (number of stations, number of receive buffers, etc.) but this has made no discernible difference.

Any advice as to what might be causing this and how I can resolve it would be gratefully appreciated.

Thanks,

HL

david-cermak, 2021-08-17:
Please follow these guidelines as a general suggestion on reducing RAM usage:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/ram-usage.html

There's no universal solution and very much depends on the application.
@Deepblue1996 I think you can try to use PSRAM or enable this option to prefer allocation from PSRAM for lwip and WiFi. Since it's an SPI device, you can also try to adjust the SPI frequency to somehow *reduce* the interface throughput and thus the need for buffering.

@Hairy-Lee it seems like the `eth2ap` example doesn't provide sufficient buffering to balance between the WiFi and ethernet bandwidth. You can try to set the `LWIP_TCP_HIGH_SPEED_RETRANSMISSION=n` so the LWIP won't retransmit that frequently.

Hairy-Lee, 2021-08-17:
@david-cermak thank you, I've just given that a go and got some partial success however it reverted back to `emac_esp32: no mem for receive buffer` after a few seconds of having a web page open on my phone. 

I've been playing about with a number of things including IDF version so I'm not entirely sure what state I'm now in. I'll give a fresh instal ago at somepoint although I was able to prove out this functionality using this [eth2wifi](https://github.com/espressif/esp-iot-solution/tree/release/v1.1/examples/eth2wifi) example so I'm satisfied that the functionality works which is enough for now. Cheers.

david-cermak, 2021-08-30:
@Hairy-Lee Thanks for the feedback.

@Deepblue1996 Can we close this?

Deepblue1996, 2021-08-31:
> @Hairy-Lee Thanks for the feedback.
> 
> @Deepblue1996 Can we close this?

Probably due to insufficient memory, the program has been optimized, and this problem has not occurred for the time being. Thank you very much for your reply

closed 2021-08-31 by Deepblue1996

Alvin1Zhang, 2021-08-31:
@Deepblue1996 Thanks for sharing updates.