espressif/esp-idf #7343: Wrap idf.py with sh script

ufbycd, 2021-07-30:
We should wrap `idf.py` with sh scrip like this:

File: $IDF_PATH/tools/idf
```bash
#!/usr/bin/env sh
idf.py $@
```
then we can call the `idf` command like this: 
```
idf build
```

igrr, 2021-07-31:
Hi @ufbycd, you can also add `alias idf="idf.py"` to one of your profile scripts, if you like the shorter version of the script name.

We had considered renaming the `idf.py` script to `idf` but in the end decide not to do this, mainly to keep the documentation more readable. When the documentation refers to "idf.py", it is clear that it talks about a command line tool. However if instead it referenced "idf", it wouldn't be clear if that was the whole SDK, or some specific program.

ufbycd, 2021-08-01:
Yes, `alias` is another way. 
I think most linuxer would perfer `idf` rather than `idf.py`. Just like `pip` script in python. And we use `which` or `where` all the time, so the `SDK` thing is not the case.

igrr, 2021-08-01:
Good point about `which`. I'm not sure of the exact use case to run `which idf.py`, though — idf.py is added to PATH at the same time IDF_PATH environment variable is set, so you might just as well use $IDF_PATH to reference some location in IDF. I do use `idf` alias myself and don't recall it being a problem, except when I'm at someone else's computer which doesn't have such an alias :)

I appreciate your suggestion on how ESP-IDF can be improved, but in this case I'm afraid we won't implement your proposal for the reason I stated before. If the documentation says, in some context, to "run idf.py flash", it is clear that the instruction refers to the command line tool. However "run idf flash" may be unclear or ambiguous.

The issue could be avoided if the tool had a completely different name (like Zephyr's `west`). We will likely consider this approach if idf.py is re-implemented in some other language one day.

closed 2021-08-01 by igrr