espressif/esp-idf #7344: Ringbuffer timeouts in certain cases

bugadani, 2021-07-30:
I was experiencing some weird issues with regards to the Ringbuffer api, so I wrote a few tests to reproduce my problem. Both of the following tests fail on the "Timeout" assert:

```

TEST_CASE("ringbuf xRingbufferSend problem case", "[ringbuf]") {
    RingbufHandle_t ringbuffer = xRingbufferCreate(200, RINGBUF_TYPE_NOSPLIT);
    uint8_t data[9] = {3, 4, 5, 6, 7, 8, 9, 10, 11};

    for (size_t size = 8; size <= sizeof(data); size++) {
        for (size_t i = 0; i < 40; i++) {
            // Make space by removing records
            while (xRingbufferGetCurFreeSize(ringbuffer) < size) {
                void* item = xRingbufferReceive(ringbuffer, NULL, 0);
                if (item == NULL) {
                    TEST_ASSERT(xRingbufferGetCurFreeSize(ringbuffer) ==
                                xRingbufferGetMaxItemSize(ringbuffer));
                    break;
                }
                // Removed record has the expected data
                TEST_ASSERT_EQUAL_MEMORY(item, data, size);
                vRingbufferReturnItem(ringbuffer, item);
            }
            TEST_ASSERT(xRingbufferGetCurFreeSize(ringbuffer) >= size);

            if (xRingbufferSend(ringbuffer, data, size, 0) != pdTRUE) {
                printf("Failed to acquire memory in event buffer. Required: %u Max: %u Free: %u\n",
                       size, xRingbufferGetMaxItemSize(ringbuffer),
                       xRingbufferGetCurFreeSize(ringbuffer));
                // If there is enough space, xRingbufferSend timed out.
                TEST_ASSERT_MESSAGE(xRingbufferGetCurFreeSize(ringbuffer) < size,
                                    "xRingbufferSend Timeout");
            }
        }

        // Flush buffer
        void* item;
        do {
            item = xRingbufferReceive(ringbuffer, NULL, 0);
            if (item) {
                // Record has expected data
                TEST_ASSERT_EQUAL_MEMORY(item, data, size);
                vRingbufferReturnItem(ringbuffer, item);
            }
        } while (item != NULL);
    }

    vRingbufferDelete(ringbuffer);
}

TEST_CASE("ringbuf xRingbufferSendAcquire problem case", "[ringbuf]") {
    RingbufHandle_t ringbuffer = xRingbufferCreate(200, RINGBUF_TYPE_NOSPLIT);
    uint8_t data[9] = {3, 4, 5, 6, 7, 8, 9, 10, 11};

    for (size_t size = 8; size <= sizeof(data); size++) {
        for (size_t i = 0; i < 40; i++) {
            // Make space by removing records
            while (xRingbufferGetCurFreeSize(ringbuffer) < size) {
                void* item = xRingbufferReceive(ringbuffer, NULL, 0);
                if (item == NULL) {
                    TEST_ASSERT(xRingbufferGetCurFreeSize(ringbuffer) ==
                                xRingbufferGetMaxItemSize(ringbuffer));
                    break;
                }
                // Removed record has the expected data
                TEST_ASSERT_EQUAL_MEMORY(item, data, size);
                vRingbufferReturnItem(ringbuffer, item);
            }
            TEST_ASSERT(xRingbufferGetCurFreeSize(ringbuffer) >= size);

            void* pItem;
            if (xRingbufferSendAcquire(ringbuffer, &pItem, size, 0) == pdTRUE) {
                memcpy(pItem, data, size);

                xRingbufferSendComplete(ringbuffer, pItem);
            } else {
                printf("Failed to acquire memory in event buffer. Required: %u Max: %u Free: %u\n",
                       size, xRingbufferGetMaxItemSize(ringbuffer),
                       xRingbufferGetCurFreeSize(ringbuffer));
                // If there is enough space, xRingbufferSendAcquire timed out.
                TEST_ASSERT_MESSAGE(xRingbufferGetCurFreeSize(ringbuffer) < size,
                                    "xRingbufferSendAcquire Timeout");
            }
        }

        // Flush buffer
        void* item;
        do {
            item = xRingbufferReceive(ringbuffer, NULL, 0);
            if (item) {
                // Record has expected data
                TEST_ASSERT_EQUAL_MEMORY(item, data, size);
                vRingbufferReturnItem(ringbuffer, item);
            }
        } while (item != NULL);
    }

    vRingbufferDelete(ringbuffer);
}
```

AxelLin, 2021-07-30:
What is your esp-idf version? (git describe --tags)

bugadani, 2021-07-30:
It's dirty due to some accidental saves, but `v4.2`

AxelLin, 2021-08-06:
```
TEST_ASSERT(xRingbufferSend(ringbuffer, data, 10, 0));
```
You have uint8_t data[9], but pass 10 in above code.
Isn't this overflow?

bugadani, 2021-08-06:
It's a mistake on my part but doesn't matter. I'll fix the comment

Dazza0, 2021-09-28:
@bugadani I've confirmed that this indeed a bug. The source of the issue is in `xRingbufferReceive()`. The internal implementation (i.e., `prvGetItemDefault()`) forgets to wrap the read pointer around a dummy data block. This results in the free pointer also being corrupted when the item is returned.

I'll push a fix for this issue shortly. For now, to work around this issue you can replace the following lines of `prvGetItemDefault()` in `ringbuf.c` as such:

From...

```c
     //Check if pucRead requires wrap around
     if ((pxRingbuffer->pucTail - pxRingbuffer->pucRead) < rbHEADER_SIZE) {
         pxRingbuffer->pucRead = pxRingbuffer->pucHead;
     }
```

To 

```c
     //Check if pucRead requires wrap around
     if (((ItemHeader_t *)pxRingbuffer->pucRead)->uxItemFlags & rbITEM_DUMMY_DATA_FLAG ||    //Next item is a dummy
         ((pxRingbuffer->pucTail - pxRingbuffer->pucRead) < rbHEADER_SIZE)) {                //Remaining space cannot fit a header
             pxRingbuffer->pucRead = pxRingbuffer->pucHead;  //Wrap pucREAD around
         }
```

bugadani, 2021-09-28:
Good catch, thanks!

AxelLin, 2021-09-29:
> ```c
>      //Check if pucRead requires wrap around
>      if (((ItemHeader_t *)pxRingbuffer->pucRead)->uxItemFlags & rbITEM_DUMMY_DATA_FLAG ||    //Next item is a dummy
>          ((pxRingbuffer->pucTail [...]

@Dazza0 

Test on v4.4-dev-3235-g3e370c429624
If I made above code change, I got below assert:
assert failed: prvReturnItemDefault ringbuf.c:604 ((pxCurHeader->uxItemFlags & ( ( UBaseType_t ) 1 )) == 0)

AxelLin, 2021-10-04:
@Dazza0 
Do you have fix for above assert issue?

AxelLin, 2021-10-19:
@Dazza0 

I test it using pppos client, the uart driver uses the ringbuffer and I *always* hit below assertion if I change the code as you suggested.
**assert failed: prvReturnItemDefault ringbuf.c:604 ((pxCurHeader->uxItemFlags & ( ( UBaseType_t ) 1 )) == 0)**

ps. It works if I don't change the code as you suggested.

0xjakob, 2021-10-20:
Hi @AxelLin, thanks for the feedback. We are still working on this issue at the moment and trying to figure out what the problem with the workaround is.

sudeep-mohanty, 2021-10-21:
Hi @AxelLin , @bugadani ,
Thanks for reporting this issue and for your feedback.
We've narrowed down the bug to an incorrect use of the comparison operation used in the function `prvGetCurMaxSizeNoSplit()` to get the free size available in a no-spilt buffer. The comparison happens between a signed (`xFreeSize`) and an unsigned (`pxRingbuffer->xMaxItemSize`) operand. When the free size available is less than the no-spilt item header size, `xFreeSize` is a negative value. But the comparison operation treats it as a large unsigned value as is the nature of binary operators, therefore resulting in a free size equal to `xMaxItemSize` when in reality it should be 0.

I'll make a fix for this. In the interim, you could test your code by making the following changes in `prvGetCurMaxSizeNoSplit()` in `components/esp_ringbuf/ringbuf.c` file:

From:

```
    //No-split ring buffer items need space for a header
    xFreeSize -= rbHEADER_SIZE;
    //Limit free size to be within bounds
    if (xFreeSize > pxRingbuffer->xMaxItemSize) {
        xFreeSize = pxRingbuffer->xMaxItemSize;
    } else if (xFreeSize < 0) {
        //Occurs when free space is less than header size
        xFreeSize = 0;
    }
```

To

```
    //No-split ring buffer items need space for a header
    xFreeSize -= rbHEADER_SIZE;
    if (xFreeSize < 0) {
        //Occurs when free space is less than header size
        xFreeSize = 0;
    } else if (xFreeSize > pxRingbuffer->xMaxItemSize) {
        //Limit free size to be within bounds
        xFreeSize = pxRingbuffer->xMaxItemSize;
    }
```

sudeep-mohanty, 2021-10-21:
Hi @AxelLin,
My bad. I missed mentioning that you do not need the below code change -

```
     //Check if pucRead requires wrap around
-    if ((pxRingbuffer->pucTail - pxRingbuffer->pucRead) < rbHEADER_SIZE) {
+    if (((ItemHeader_t *)pxRingbuffer->pucRead)->uxItemFlags & rbITEM_DUMMY_DATA_FLAG ||    //Next item is a dummy
+        ((pxRingbuffer->pucTail - pxRingbuffer->pucRead) < rbHEADER_SIZE)) {                //Remaining space cannot fit a header
```

Please keep the `pucRead` update as before and only update the `prvGetCurMaxSizeNoSplit()` function as mentioned in an earlier comment.
Would you help try this update and let us know if it works for you?

AxelLin, 2021-10-21:
Hi @sudeep-mohanty 

It works if only update the prvGetCurMaxSizeNoSplit() function.

I have some suggestions:
1. The change is not trivial, please add a comment to the code.
2. I use pppos_client and the uart driver uses esp_ringbuf heavily.
   However, it seems I didn't hit the timeout issue before changing any code.
   So I cannot verify if the fix actually works or not (because it was working in my use case).
   I suggest add a TEST_CASE for this issue to encure it's correct.

closed 2021-11-03 by espressif-bot

AxelLin, 2021-11-08:
@sudeep-mohanty 
Need to backport this fix to stable branches.

AxelLin, 2021-12-24:
Just remind that v4.3 branch still does not include this fix.

Alvin1Zhang, 2021-12-27:
@AxelLin Thanks for the attention, we have back ported this fix to 4.3, but still under internal reviewing, will sync up here once the fix is available on GitHub.