espressif/esp-idf #7345: Documented log level options missing from config

mbratch, 2021-07-31:
## Environment

- Development Kit: ESP32-PICO-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4.1
- Module or chip used: ESP32-PICO-D4
- IDF version (run git describe --tags to find it): v4.3
- Build System: [Make|CMake|idf.py]
- Compiler version (run xtensa-esp32-elf-gcc --version to find it): 8.4.0
- Operating System: Windows
- (Windows only) environment type: PowerShell
- Using an IDE?: Yes (Eclipse with ESP-IDF plugin)
- Power Supply: USB, or External with DC-DC regulator to 5V
- Wifi Router: Apple Airport model A1408 (~10 years old)

## Problem Description

According to the documentation (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/log.html), there should be two menuconfig parameters for managing log level:

- At compile time: in menuconfig, set the verbosity level using the option **CONFIG_LOG_DEFAULT_LEVEL**.

- Optionally, also in menuconfig, set the maximum verbosity level using the option **CONFIG_LOG_MAXIMUM_LEVEL**. By default this is the same as the default level, but it can be set higher in order to compile more optional logs into the firmware.

According to this document (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html), both of these CONFIG_LOG options should be available in menuconfig under: `Component config > Log output`.

However, `Component config > Log output` only shows the "Default log verbosity". It does not show maximum log level. Further, the "Default log verbosity" appears to act as both options simultaneously: it sets the default level _and_ it defines the maximum level that the program can set, not allowing the program to set it any higher using `esp_set_log_level`.

[image]

### Expected Behavior

It was expected that both the default and maximum log level settings would be available per the documentation so that each of these can be individually controlled/configured.

### Actual Behavior

Only one of the two documented options appears to be implemented in menuconfig, and the one implemented does not completely behave as described in the documentation.

### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

ESP-Marius, 2021-07-31:
Hi,

It looks like you are using IDF v4.2. The feature you are talking about was added recently and doesnt exist in v4.2. On the latest master branch this option is available in menuconfig:

[image]

If you are using an earlier version of IDF we recommend using the docs that corresponds to that version, for release/v4.2 this would be https://docs.espressif.com/projects/esp-idf/en/release-v4.2/esp32/

mbratch, 2021-07-31:
@ESP-Marius thank you very much.

When I execute `idf.py --version` I get `ESP-IDF v4.3-dirty`. How can I make sure I have a complete 4.3 version?

AxelLin, 2021-08-01:
> Hi,
> 
> It looks like you are using IDF v4.2. The feature you are talking about was added recently and doesnt exist in v4.2. On the latest master branch this option is available in menuconfig:

**v4.3** does not support  CONFIG_LOG_MAXIMUM_LEVEL, it's only available in master (v4.4).

mbratch, 2021-08-01:
> 
> **v4.3** does not support CONFIG_LOG_MAXIMUM_LEVEL, it's only available in master (v4.4).

@AxelLin  Thank you! This was my mistake as I thought I was using the latest ESP-IDF.

ESP-Marius, 2021-08-02:
@mbratch That's good! I'm closing the issue then.

closed 2021-08-02 by ESP-Marius