espressif/esp-idf #7346: Classic BT Peering algorithm when setting io cap to display only

mohannadmaklad, 2021-07-31:
## Environment

- Development Kit:      ESP32-Wrover
- Kit version 1.1
- Module or chip used:  ESP32-WROOM
- IDF version 4.3
- Build System:         idf.py
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0

- Operating System:     Windows
- Using an IDE?: Eclipse
- Power Supply:        USB

## Problem Description

I'm writing an application for classic bluetooth to demo pairing algorithms.  I will document both scenarios (Laptop &  Smartphone).

**In both cases I set io cap to display only.**

### Expected Behavior
Passkey display on both devices
### Actual Behavior
Laptop : Displays a pin to compare to but ESP doesn't display anything.
Android phone : Connects directly (Just Works)

### Steps to reproduce

1. set  esp_bt_io_cap_t iocap = ESP_BT_IO_CAP_OUT; //Display only
2. Connect to esp from a laptop/android phone

[image]

mohannadmaklad, 2021-08-03:
@BetterJincheng  Hi, Peter
Thanks for your reply :) 

Actually the screenshot is taken from 4.2 Core specs, this is page 323  on the specs : 
[image]

I already demonstrated it with other configuration. I'm interested though if the other device have io cap set as "Display Keyboard". Unfortunately, I don't have another ESP board in hand to test this scenario, Do you suggest anything?

Also, I Think laptop has the setting already as "Display Keyboard" because it's already asking for a pin number. So, I highly doubt that this is a bug.

Another thing that might be irrelevant, What if the other device sets its configuration on purpose to force Just Works pairing algorithm, I can see any event that let's me decide if I accept of the reject the connection. Is there any other option?

BetterJincheng, 2021-08-04:
Hi, @mohannadmaklad!

The table of  IO capability I mentioned is on page 323 in Vol 3, Part 3. It demonstrated all the settings and corresponding behaviors.

In ESP-IDF v4.3, we only provided four options listed as `ESP_BT_IO_CAP_OUT (DisplayOnly)`, `ESP_BT_IO_CAP_IO (DisplayYesNo)`, `ESP_BT_IO_CAP_IN (KeyboardOnly)` and `ESP_BT_IO_CAP_NONE (NoInputNoOutput)`. We just follow the Core Spec.

BetterJincheng, 2021-08-04:
Hi, @mohannadmaklad 

The table of IO capability I mentioned is on page 323 in Vol 3, Part 3.

It does look a little weird. I will do some checking and reply to you as soon as possible.

mohannadmaklad, 2021-08-04:
This is the correct one : 

[image]

BR/EDR v4.2 Doesn't have display + Keyboard input?

Anyway, Do you know how can I force authentication process?

BetterJincheng, 2021-08-04:
As the IO cap of Initiator (laptop or android phone) is unknown, you can just set the Responder (esp32) with `esp_bt_io_cap_t iocap = ESP_BT_IO_CAP_IN;  //Keyboard Only` to force the authentication process. Then the PIN will show on your phone or laptop, input it into the terminal to establish the connection. (To do this, an input method is needed. r.f. `/esp-idf/examples/bluetooth/bluedroid/classic_bt/bt_spp_initiator/`)