espressif/esp-idf #7347: 802.11 LR can be used in Japan?

uecken, 2021-08-01:
In below page, 802.11 LR is subscribed. In Japan 802.11 LR can be used?
https://github.com/espressif/esp-idf/blob/58022f8/docs/en/api-guides/wifi.rst

ESP32 has Japanese certification "Gijyutukijyun Tekigou Syoumeisyo" about 802.11 a/b/g/n,
so 802.11 a/b/g/n can be used in Japan.
However the 802.11 LR can be used in Japan ? 
Also ESP-Now can be used in Japan ?

I thine the target point are whether 
･ PHY layer is changed or not
･ Carrier sense is executed or not

I would be grateful if you could give me some advice.

MaxwellAlan, 2021-08-02:
hi @uecken 
ESP-NOW not change anything about PHY layer, which can be used. 
While LR mode mainly change about the PHY Data Rate, which support 250Kbps & 500Kbps, others such as channel, tx power do not change, not quite sure about the data rate can be used in Japan.

sagb2015, 2021-08-02:
@uecken  Can you elaborate on the the certification requirements?   As @MaxwellAlan noted, ESPNOW is completely covered in the upper layer and no PHY layer modifications are needed. LR mode is a proprietary mode and does not increase the Tx power of the device. The modulation is different for LR mode and is not compatible with any other standard WiFi device.

uecken, 2021-08-06:
Thank you for your reply. LR mode changes only the PHY Data Rate?
In nay case, this LR mode is not standard WiFi. 
So, if we use LR mode in Japan, need o be tested and meet requirements.

I have not known certification requirements for 2.4GHz. So, I investigated it.
Reference to https://www.tele.soumu.go.jp/j/sys/equ/tech/type/,
The 2.4GHz system needs to obey "2.4GHz帯高度化小電力データ通信システム(2,400～2,483.5MHz)"
https://www.tele.soumu.go.jp/resource/j/equ/tech/betu/43.pdf

Of course power, obw, spurious is needed, however Carrier sense is not needed in case the OBW is less than 26MHz.
If the LR mode uses less than 26MHz and other RF characteristic is same with 802.11a/b/g/n, it's not difficult to take Japanese certification.

By the way, LR mode uses like LoRa protocol?   Receiver sensitivity is +4dB than WiFi?

MaxwellAlan, 2021-11-23:
hi @uecken ,

- LR mode changes only the PHY Data Rate?
  - Yes.
- LR mode uses like LoRa protocol? Receiver sensitivity is +4dB than WiFi?
  - Yes.
- What is the OBW?
  - OBW is Occupied Bandwidth ([OBW](https://rfmw.em.keysight.com/rfcomms/refdocs/tdscdma/Occupied_Bandwidth_(OBW)_Measurement.htm)), which is the same as 11b.
- If PHY is same with 802.11, use Carrier Sense?
  - Yes.

uecken, 2021-11-23:
@MaxwellAlan @sagb2015 
Sorry for my late reply.
I understand your reply.
LR Phy is same with other WiFi. So i t its not needed additional certification.

BTW, Can I use LR mode from PlatformIO?

I am developping below system.
https://protopedia.net/prototype/2356
I want to cover 100 meter without repeater

MaxwellAlan, 2021-11-23:
hi @uecken ,
Sorry for that I'm not familiar with PlatformIO IDE.
Btw, I think the LR mode only depends on which IDF verion you are working on, since IDF release/v3.3 and later, all these IDF versions/branches are support LR mode.
Thanks.

uecken, 2021-11-23:
@MaxwellAlan 
Thanks.
It seems that platform-espressif32 uses ESP-IDF latest 4.3.1. 
https://github.com/platformio/platform-espressif32/releases

Thank you! I will try 801.11LR.
OK, I want to ask how to use 11LR.
Which? 
https://github.com/platformio/platform-espressif32
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiAP.cpp

Additionally, I also want to try 11b on PlatformIO IDE.

Regards.

MaxwellAlan, 2021-11-23:
hi @uecken ,
May be you can try esp-idf/examples/wifi/getting_started , both in SoftAP and Station example, after wifi init, you can call [esp_err_t esp_wifi_set_protocol(wifi_interface_t ifx, uint8_t protocol_bitmap)](https://github.com/espressif/esp-idf/blob/a18cb7090511d235ebe3c8b833180630f4f8bb33/components/esp_wifi/include/esp_wifi.h#L512) to set operating protocol, eg: esp_wifi_set_protocol(0, WIFI_PROTOCOL_LR); and sorry for lack of some docs about LR mode.

uecken, 2021-11-23:
thanks @MaxwellAlan !
Now I am installing ESP-IDF with PlatformIO.

Get started with ESP-IDF and ESP32-DevKitC: debugging, unit testing, project analysis¶ 
https://docs.platformio.org/en/stable/tutorials/espressif32/espidf_debugging_unit_testing_analysis.html

I will try!

uecken, 2021-11-24:
I found the Arduino source code to use 802.11 LR.
First I will try this.
https://github.com/jnogues/ESP32-Long-Range-WiFi/blob/master/master.ino

BTW, 11b and 11LR is very good foy 2.4GHz IoT communication.
I have an idea to improve this WiFi more.

# not execute carrier sense.
 Under OBW 26MHz, carrier sense is not needed.
 [image]
 If not doing carrier sense, many device can send the data simultaniously. 

# implement power control
If power control are implemented like 3G system that uses DSSS. 
simultaneous transmission will be success, I think.

Regard.

closed 2022-03-04 by Alvin1Zhang