espressif/esp-idf #7348: esp_wifi_connect() fails at boot

DiegoActionItem, 2021-08-01:
We have a few systems in the field that are randomly failing `esp_wifi_connect()` (which is insidehttps://github.com/espressif/esp32-wifi-lib/tree/a037ea80b03b3253a17fb59ec63350720ee72ce8 so I cannot debug any more).

Unfrotunately, I don't have a way to reproduce and I don't have the exact return error (only "non zero").  All I know for now is that `esp_wifi_connect()` is not returning `ESP_OK`.

This happens on version 4.2.0. I will try and send the customer a version based on 4.3.

How should I aproach in debugging this issue? the devices connected to the network before (ehm... they got OTA'ed and then fail). 

Sorry about the lack of details. I am also asking - what more details can I provide?

AxelLin, 2021-08-01:
FYI, the example codes removed checking return value of  esp_wifi_connect()
https://github.com/espressif/esp-idf/commit/c260c223e979c2604806cb585064e4fcfe80fdba#diff-4de2b17b2c681cc1921bb3ebf820735f4fdbbde4da35790b99c5db9948bcf38f

espxiehang, 2021-08-02:
Hi @DiegoActionItem , Can you provide some logs？

DiegoActionItem, 2021-08-02:
We are investigating other causes for this fault (a new internal version, with v4.4). I will re-open if this issue re-accurs.

closed 2021-08-02 by DiegoActionItem