espressif/esp-idf #7350: nvs_flash_erase_partition_ptr reporting ESP_ERR_INVALID_ARG on valid partition

CSHaze, 2021-08-02:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version DevKitC: v4
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):
    // v4.3
- Build System:  idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:  Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: Yes VSCode with Espressif IDF extension
- Power Supply:  USB

## Problem Description

both nvs_flash_erase_partition and nvs_flash_erase_partition_ptr are reporting ESP_ERR_INVALID_ARG

I am outputting the result from esp_partition_find_first and am getting a valid pointer with my partition.

### Expected Behavior

The partition should erase correctly.

### Actual Behavior

The function nvs_flash_erase_partition_ptr throws an error

### Steps to reproduce
1. Create new partition and add values.

2. Attempt to erase the partition
```
esp_partition_t *part = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_NVS, CERT_NVS);   ESP_LOGI(TAG, "partition name %s", part->label);
ESP_ERROR_CHECK(nvs_flash_erase_partition_ptr(part));
```

### Code to reproduce this issue

```
ESP_ERROR_CHECK(nvs_flash_init_partition(CERT_NVS));
ESP_ERROR_CHECK(nvs_open_from_partition(CERT_NVS, CERT_STORE, NVS_READWRITE, &handle));

esp_partition_t *part = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_NVS, CERT_NVS);   ESP_LOGI(TAG, "partition name %s", part->label);
ESP_ERROR_CHECK(nvs_flash_erase_partition_ptr(part));
```

## Debug Logs

```
I (28336) WIFI_CONNECTION: partition name cert_nvs
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x4008f0c8
0x4008f0c8: _esp_error_check_failed at C:/Users/Chris/esp/esp-idf/components/esp_common/src/esp_err.c:41

file: "../components/wifi/wifi.c" line 167
func: reset_device
expression: nvs_flash_erase_partition_ptr(part)

abort() was called at PC 0x4008f0cb on core 0
0x4008f0cb: _esp_error_check_failed at C:/Users/Chris/esp/esp-idf/components/esp_common/src/esp_err.c:42

Backtrace:0x4009005b:0x3ffc5970 0x4009082d:0x3ffc5990 0x40097766:0x3ffc59b0 0x4008f0cb:0x3ffc5a20 0x400e1266:0x3ffc5a40 0x400e12c0:0x3ffc5a60 0x40093ce5:0x3ffc5a80
0x4009005b: panic_abort at C:/Users/Chris/esp/esp-idf/components/esp_system/panic.c:356

0x4009082d: esp_system_abort at C:/Users/Chris/esp/esp-idf/components/esp_system/system_api.c:112

0x40097766: abort at C:/Users/Chris/esp/esp-idf/components/newlib/abort.c:46

0x4008f0cb: _esp_error_check_failed at C:/Users/Chris/esp/esp-idf/components/esp_common/src/esp_err.c:42

0x400e1266: reset_device at C:\Users\Chris\Documents\VSCode\esp32_plantbox\build/../components/wifi/wifi.c:167 (discriminator 1)

0x400e12c0: watch_device_reset at C:\Users\Chris\Documents\VSCode\esp32_plantbox\build/../components/wifi/wifi.c:455

0x40093ce5: vPortTaskWrapper at C:/Users/Chris/esp/esp-idf/components/freertos/port/xtensa/port.c:168
```

CSHaze, 2021-08-02:
shows up on latest development branch too

igrr, 2021-08-02:
Can you please attach your partition table, or your sdkconfig file if you are using one of the default partition tables?

CSHaze, 2021-08-02:
Hi @igrr , I am sorry I didn't attach it initially. I was thinking the problem may have something to do with that and have been playing around with the values. Though I don't get any errors initing or adding things to the  partition, just deleting it.

`
# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
nvs,      data, nvs,     0x11000,  0x6000,
phy_init, data, phy,     ,  0x1000,
# factory,  app,  factory, , 0x19F0A0,
factory, app, factory, , 0x1E8480,
cert_nvs, data, nvs,     , 0x70000,
`

igrr, 2021-08-02:
@CSHaze I think the issue is that your factory partition size is not divisible by flash sector size (4kB), so the NVS partition start address is not aligned to sector start. This causes a failure during the erase operation, since erase operations can only by done for aligned sectors.

Please change this to
```
factory, app, factory, , 0x1E9000
```
and try again.

igrr, 2021-08-02:
Linking https://github.com/espressif/esp-idf/issues/7295 which is another issue where the partition table generation tool doesn't produce an error for an unaligned partition offset/size, and the error only happens at run time.

CSHaze, 2021-08-03:
Hey @igrr, I appreciate the help. I am now seeing a different nvs related error

```
 nvs: nvs_open_from_partition cert_store 1
I (4336) WIFI_CONNECTION: Starting provisioning
D (4336) nvs: nvs_get opmode 1
D (4346) ADC: Wi-Fi takes adc2 lock.
I (4346) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
D (4346) phy_init: loading PHY init data from application binary
D (4356) nvs: nvs_open_from_partition phy 0
D (4366) phy_init: esp_phy_load_cal_data_from_nvs: failed to open NVS namespace (0x1102)
```
Do I need to change the size of cert_nvs now as well?

igrr, 2021-08-04:
This might be related to how you are initializing the two NVS partitions found in the partition table, and whether your flashing process involves erasing the main NVS partition. Would you mind opening a new issue for this, describing the code flow of your app (or better, attaching some sample code) and telling more about the relation between the occurrence of this issue and running idf.py monitor?

closed 2021-08-24 by espressif-bot