espressif/esp-idf #7354: 4-bit SDIO data transfer issues with example

QSchulz, 2021-08-02:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32UE
- IDF version (run ``git describe --tags`` to find it):
    // Release v4.2 (tested on v4.3 too)
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     Linux
- (Windows only) environment type: N/A.
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

We already (and succesfuly) use an ESP32-WROOM-32UE over SPI in our current design. We want to switch over to SDIO (4-bit mode) for performance reasons. We've had issues communicating with the ESP32_DevKitC_V4 over SDIO with a Linux kernel running on a Rockchip PX30 with most data coming from the ESP32 being corrupted.

I've made a setup in which I can reproduce the issue with example code from esp-idf. Two ESP32_DevKitC_V4 devkits, one acting as an SDIO host and the other as an SDIO slave. Note that we've put 10K Ohm Pull-ups (3.3V) on all lanes except clock as recommended in the documentation (https://docs.espressif.com/projects/esp-idf/en/v4.3-rc/esp32/api-reference/peripherals/sd_pullup_requirements.html#no-pull-ups) and burnt the fuse for the bootstrapping pin GPIO12 to be usable (effectively forcing the internal regulator for the SPI flash to 3.3V) (https://docs.espressif.com/projects/esp-idf/en/v4.3-rc/esp32/api-reference/peripherals/sd_pullup_requirements.html#conflicts-between-bootstrap-and-sdio-on-dat2).

The code running on the devkits are https://github.com/espressif/esp-idf/blob/master/examples/peripherals/sdio/slave/main/app_main.c for the slave and https://github.com/espressif/esp-idf/blob/master/examples/peripherals/sdio/host/main/app_main.c for the host. On the host, CONFIG_EXAMPLE_SDIO_4BIT and SLAVE_PWR_GPIO (to GPIO_B1) are defined to use SDIO in 4-bit mode and use 4 data lanes (D0-D3) and enable power reset of the SDIO slave at boot.

Data transfer (using FIFOs and CMD53) is corrupted. I've attached a screenshot of a capture of the data lanes done in the PX30 setup with the clock running at 400KHz. Each data transfer is 128 bytes long with each byte being +1 of the previous byte, starting at 0. Therefore, expected data is 0x00, 0x01, 0x02, ... 0x18, 0x19, ...
In the attached capture, you can see that between 0x07 and what is supposed to be 0x08, the 0x0 nibble of 0x08 is missing. On a 128 bytes payload, at least 3 nibbles are missing. The position of the missing nibbles isn't consistent.

In the ESP32-only setup, running at 40MHz, most if not all messages are incorrect or not received, either timing out or erroring out because of CRC mismatch (due to missing nibble I assume, my logic analyzer cannot capture at that speed).

What is surprising is that in the PX30 setup, data from the PX30 (host) to the ESP32 (slave) is always correct (though sometimes timing out) but data from ESP32 (slave) to PX30 (host) is never (always missing at least a nibble).

### Expected Behavior

Valid data for transfers on SDIO in 4-bit mode.

### Actual Behavior

Missing nibble or timeout when transferring data corrupting data blocks.

### Steps to reproduce

1. Flash one ESP32_DevKitC_V4 with SDIO host example (with CONFIG_EXAMPLE_SDIO_4BIT defined and SLAVE_PWR_GPIO constant set to GPIO_B1),
2. Flash one ESP32_DevKitC_V4 with SDIO slave example,
3. Connect SCLK, CMD, DAT0-DAT3 between the two modules, with 10K Ohm pull-ups on CMD, DAT0-DAT3,
4. Connect GPIO21 from devkit running the SDIO host example to EN pin on devkit running the SDIO slave example,
5. Supply power to each devkit via micro USB cable,

### Code to reproduce this issue

- https://github.com/espressif/esp-idf/blob/master/examples/peripherals/sdio/slave/main/app_main.c
- https://github.com/espressif/esp-idf/blob/master/examples/peripherals/sdio/host/main/app_main.c + CONFIG_EXAMPLE_SDIO_4BIT defined and SLAVE_PWR_GPIO constant set to GPIO_B1

## Debug Logs

Host:
```
I (2402) example_host: send reset to slave...
I (2912) example_host: slave io ready
I (2912) example_host: ========JOB: write slave reg========
I (3012) example_host: read registers:
I (3012) example_host: 0x3ffb6880   0a 0d 10 13 16 19 1c 1f  22 25 28 2b 2e 31 34 37  |........"%(+.147|
I (3012) example_host: 0x3ffb6890   3a 3d 40 43 46 49 4c 4f  52 55 58 5b 6a 6d 70 73  |:=@CFILORUX[jmps|
I (3022) example_host: 0x3ffb68a0   76 79 7c 7f 82 85 88 8b  8e 91 94 97 9a 9d a0 a3  |vy|.............|
I (3032) example_host: 0x3ffb68b0   a6 a9 ac af b2 b5 b8 bb  be c1 c4 c7 01 00 00 00  |................|
I (3042) example_host: ========JOB: get interrupts from slave========
I (3072) example_host: host int: 0
I (3572) example_host: host int: 1
I (4072) example_host: host int: 2
I (4572) example_host: host int: 3
I (5072) example_host: host int: 4
I (5572) example_host: host int: 5
I (6072) example_host: host int: 6
I (6572) example_host: host int: 7
I (8062) example_host: ========JOB: get interrupts from slave========
I (8072) example_host: host int: 0
I (8572) example_host: host int: 1
I (9072) example_host: host int: 2
I (9572) example_host: host int: 3
I (10072) example_host: host int: 4
I (10572) example_host: host int: 5
I (11072) example_host: host int: 6
I (11572) example_host: host int: 7
I (13072) example_host: ========JOB: send fifos========
E (13072) sdmmc_io: sdmmc_io_rw_extended: sdmmc_send_cmd returned 0x109
ESP_ERROR_CHECK failed: esp_err_t 0x109 (ESP_ERR_INVALID_CRC) at 0x40083a68
file: "../main/app_main.c" line 420
func: job_fifo
expression: ret
```
Slave:
```
I (308) example_slave: ================ slave ready ================
I (1098) example_slave: ================ JOB_RESET ================
I (1588) example_slave: ================ JOB_WRITE_REG ================
I (1588) example_slave: write regs:
I (1588) example_slave: 0x3ffb6670   0a 0d 10 13 16 19 1c 1f  22 25 28 2b 2e 31 34 37  |........"%(+.147|
I (1598) example_slave: 0x3ffb6680   3a 3d 40 43 46 49 4c 4f  52 55 58 5b 00 00 00 00  |:=@CFILORUX[....|
I (1608) example_slave: 0x3ffb6690   6a 6d 70 73 76 79 7c 7f  82 85 88 8b 8e 91 94 97  |jmpsvy|.........|
I (1618) example_slave: 0x3ffb66a0   9a 9d a0 a3 a6 a9 ac af  b2 b5 b8 bb be c1 c4 c7  |................|
I (1738) example_slave: ================ JOB_SEND_INT ================
```

## Other items if possible
Logic analyzer capture screenshot:
[image]
Full host log: [host.log](https://github.com/espressif/esp-idf/files/6916317/host.log)
Full slave log: [slave.log](https://github.com/espressif/esp-idf/files/6916328/slave.log)

Alvin1Zhang, 2021-08-03:
Thanks for the very detailed issue report, we will look into.

ginkgm, 2021-08-03:
Hi @QSchulz ,
    Seen from your description, it's very likely to be an SI issue, which is very common when using 4-bit mode, and not solvable unless the is properly design.

    Basically this is because your grounding is not good enough, when there are more than 1 data line is toggling, the inrush current will cause the floating of ground between the host and device, resulting in the glitch in the clock signal (seen from the slave).
    There are two ways to confirm this. Firstly, you can use an oscilloscope to test DIRECTLY on the slave's clock pin and slave's ground. You can see when the issue happens, the clock signal is very bad.
    The second way is to add more grounding wires between host & slave. I guess you are using DuPont  wires? Adding more will make it a little better. But this is not the final solution, sometimes we find it still not very good even we use 6+ grounding lines.

     If you confirmed this issue (you see glitch on clock, or the communication can be improved by HW approach), you can redesign your PCB to solve the issues.

    Please contact our support team sales@espressif.com to get detailed suggestions on designing the HW, and maybe some evaluation boards for your use case.

Best Regards,
Michael

ginkgm, 2021-08-03:
btw, another clue is that, 1-bit mode is working, but 4-bit mode is not, and you have confirmed the pullups are correect.

according to my experience, using a lower host clock doesn't help to solve the issue in 4-bit mode.

QSchulz, 2021-08-06:
Hi @ginkgm,

We eventually removed the dupont wires for the SDIO lanes and soldered an ESP32-WROOM-32U module as close as possible to the pins on the PX30 board. We did need to use thicker cables for ground and we now have three connected.

This made 4b mode SDIO work just fine. Thanks for the support!

We used a scope on the clock line before but couldn't detect any glitch.

I used a lower host clock just because my logic analyzer was not able to capture at the default 25MHz. I didn't expect it to change much :)

Closing this issue since it was a HW issue that is now fixed thanks to your support.

closed 2021-08-06 by QSchulz