espressif/esp-idf #7357: Alternating MQTT disconnection and reconnection

Wasabinary, 2021-08-02:
Hi, 

I'm currently working with multiple ESP32-S2 connected securely to a distant MQTT broker. My code is very similar to the MQTT examples given by Espressif : I basically connect to the MQTT broker using a certificate, and send a message approximately every second.

I managed to successfully reconnect to Wi-Fi and to the broker in case the connection is lost... I mean, most of the time, because sometimes, randomly, one of many devices starts alternating between disconnecting and reconnecting states very quickly, sending a few messages once reconnecting and then disconnecting again.

There's nothing obvious in my code that could explain this behavior. It can appear after running for a few minutes, and sometimes it takes hours before I reach this problem. Does anyone know this issue?

My ESP-IDF version is release/v4.3

Thanks for your help

ESP-YJM, 2021-08-05:
@Wasabinary Could you please offer some detailed log that can describe your issue.

Wasabinary, 2021-08-05:
@ESP-YJM thanks for answering

I don't really have error logs so it's even stranger.

Basically, I initialize my secure MQTT client and use a free RTOS task to send a 3kB payload to a secure broker every second. I use QoS0, and simply log a message every time I send a message from my task. I also have an event handler, which is almost the same as the one given in the MQTT examples:

```
void CLOUD_MQTT_event_handler(void *handler_args, esp_event_base_t base, int32_t event_id, void *event_data)
{
    ESP_LOGD(TAG, "Event dispatched from event loop base=%s, event_id=%d", base, event_id);
    esp_mqtt_event_handle_t event = event_data;
    esp_mqtt_client_handle_t client = event->client;
    switch ((esp_mqtt_event_id_t)event_id) {
    case MQTT_EVENT_CONNECTED:
        ESP_LOGI(TAG, "MQTT_EVENT_CONNECTED");
        break;
    case MQTT_EVENT_DISCONNECTED:
        ESP_LOGI(TAG, "MQTT_EVENT_DISCONNECTED");
        break;
    case MQTT_EVENT_SUBSCRIBED:
        ESP_LOGI(TAG, "MQTT_EVENT_SUBSCRIBED, msg_id=%d", event->msg_id);
        break;
    case MQTT_EVENT_UNSUBSCRIBED:
        ESP_LOGI(TAG, "MQTT_EVENT_UNSUBSCRIBED, msg_id=%d", event->msg_id);
        break;
    case MQTT_EVENT_PUBLISHED:
        ESP_LOGI(TAG, "MQTT_EVENT_PUBLISHED, msg_id=%d", event->msg_id);
        break;
    case MQTT_EVENT_DATA:
        ESP_LOGI(TAG, "MQTT_EVENT_DATA");
        break;
    case MQTT_EVENT_ERROR:
        ESP_LOGI(TAG, "MQTT_EVENT_ERROR");
        break;
    case MQTT_EVENT_BEFORE_CONNECT:
        ESP_LOGI(TAG, "MQTT_EVENT_BEFORE_CONNECT");
        break;
    default:
        ESP_LOGD(TAG, "Other event id:%d", event->event_id);
        break;
    }
}
```
After a running the device for a while I get the MQTT disconnected event triggered, and I also get a message when publishing telling me that my client is disconnected:
>MQTT_EVENT_DISCONNECTED
>Publishing skipped: client is not connected
>Publishing skipped: client is not connected
>Publishing skipped: client is not connected

Nothing too surprising for now. The thing is, after a 30-40 seconds, the client reconnects and starts sending message again:
>MQTT_EVENT_BEFORE_CONNECT
>MQTT_EVENT_CONNECTED
>message sent (this is the message I log every time I publish a message)
>message sent
>message sent

... and then disconnects again:
>MQTT_EVENT_DISCONNECTED
>Publishing skipped: client is not connected

So it starts alternating between connected and disconnected state for 5 to 10 times, and everything goes back to normal. Any idea?

ESP-YJM, 2021-08-05:
@Wasabinary Could you share your complete log file? I think before print `MQTT_EVENT_DISCONNECTED`, maybe some details indicate the reason.

Wasabinary, 2021-08-05:
Ok I'll enable debug log and try to re-run the program and hopefully reach the issue (it's quite random) and then I'll share it you