espressif/esp-idf #7358: DFS negatively impacts LEDC

JonRobo, 2021-08-02:
## Environment

- Module or chip used:  ESP32-WROOM-32
- IDF version v3.3.4
- Build System: idf.py
- Operating System:  Windows

## Problem Description

When DFS is set in the power management section of menu config I do have ~40% power savings which is great, however the 4 LEDs I have in my product skip/glitch in their fading animations. When I disable DFS the performance returns to normal.

### Expected Behavior
I would really be blown away if DFS can save me the power I know it can while maintaining smooth LEDC animations 

### Actual Behavior
DFS saves power, but impacts LEDC performance

Please let me know if this can be solved, or perhaps is already solved in a later version of IDF

Thank-you

songruo, 2021-08-04:
Hi @JonRobo ,

New feature has been added to allow the users to configure ledc source clock in ledc timer configuration since IDF version v4.0. For ESP32, both `RTC8M_CLK` and `REF_TICK` clock sources will not be affected by DFS. You can specify your source clock to be either one of them depending on your requirement. Please check [https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/ledc.html#_CPPv4N19ledc_timer_config_t7clk_cfgE](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/ledc.html#_CPPv4N19ledc_timer_config_t7clk_cfgE) to know how to specify the source clock in `ledc_timer_config_t`. 

If you are not really willing to upgrade IDF version to v4.0 or later, there is a walkround to keep the ledc clock freq unchanged by holding a `ESP_PM_APB_FREQ_MAX` lock while fading is going on. 
Example code:
`#include "esp_pm.h"`

`static esp_pm_lock_handle_t ledc_apb_sleep_lock;`
`esp_pm_lock_create(ESP_PM_APB_FREQ_MAX, 0, "ledc_apb", &ledc_apb_sleep_lock);`

`esp_pm_lock_acquire(ledc_apb_sleep_lock);`
`ledc_set_fade_with_time(LEDC_MODE, LEDC_CHANNEL, 4000, 5000);`
`ledc_fade_start(LEDC_MODE, LEDC_CHANNEL, LEDC_FADE_WAIT_DONE);`
`esp_pm_lock_release(ledc_apb_sleep_lock);`

JonRobo, 2021-08-04:
Hello, the lock solution worked perfectly for me on V3.3.4

Thank you so much, this is exactly what I needed!

closed 2021-08-04 by JonRobo