espressif/esp-idf #7361: Compiler error on esp-idf v4.4-dev-2359-g58022f859

baldhead69, 2021-08-03:
Hi,

I installed de esp-idf version: v4.4-dev-2359-g58022f859, and the following compiler errors appear:

* C:/esp-idf/components/esp_http_client/esp_http_client.c:1055:25: error: implicit declaration of function 'esp_tls_get_and_clear_last_error'
* C:/esp-idf/components/esp_http_client/esp_http_client.c:1055:124: error: 'ESP_ERR_ESP_TLS_TCP_CLOSED_FIN' undeclared (first use in this function);

I am not using "esp_http_client" component.
I am using "esp_https_server" component.

Some suggestion ???

baldhead69, 2021-08-03:
No one ?

ESP-Marius, 2021-08-04:
Did you by any chance disable `ESP_HTTP_CLIENT_ENABLE_HTTPS`. Seems like the component fails to build with this disabled on master.

baldhead69, 2021-08-04:
Where disable this ?
On idf.py menuconfig ?

I think i didn't set up this.

baldhead69, 2021-08-04:
@ESP-Marius 

[image]

baldhead69, 2021-08-04:
@ESP-Marius 

How can i upgrade my esp-idf using git command ?

I am using windows.

ESP-Marius, 2021-08-04:
@baldhead69 That means it's disabled, try enabling "Enable https", and I think you should be able to compile successfully again.

(And we'll fix the issue as well, so in the future you shouldnt need to enable this)

baldhead69, 2021-08-04:
@ESP-Marius 

Can you help me with this issues ?

https://github.com/espressif/esp-idf/issues/5602
https://github.com/espressif/esp-idf/issues/7027

Thank's

baldhead69, 2021-08-04:
> @baldhead69 That means it's disabled, try enabling "Enable https", and I think you should be able to compile successfully again.
> 
> (And we'll fix the issue as well, so in the future you shouldnt need to enable this)

Worked. is compiling.

Before this i had done a workaround to compiling.
I had commented the code that was bugging the component/file "esp_htto_client.c" and so it also compiled.

closed 2021-08-24 by espressif-bot