espressif/esp-idf #7364: Comprehensive heap memory debugging (slow poisoning) not as comprehensive as of 4.3

ftab, 2021-08-04:
## Problem Description

multi_heap_internal_check_block_poisoning is no longer called as of https://github.com/espressif/esp-idf/commit/bd9b921713d80f4b19931f4d94f38e87881f04cd

Expand the diff for components/heap/multi_heap.c - multi_heap_check() changed significantly when the allocator was changed to TLSF, so each call to heap_caps_check_integrity() or heap_caps_check_integrity_all() now apparently no longer checks poisoning, only performs TLSF's internal checking.

### Expected Behavior

Heap poison should be checked if MULTI_HEAP_POISONING is set. The behavior in v4.2.2 and prior is that a more comprehensive check is performed at each call to heap_caps_check_integrity and it catches an overwritten poison area

### Actual Behavior

Only tlsf structure checks are performed, apparently no poison checks according to heap_tlsf.c

This allowed a corruption early on in my program to go unnoticed until something seemingly irrelevant was changed and finally touched one of the structures that the TLSF code checks

I am still trying to narrow down when the corruption happens in my project, so I don't have a minimal reproducible example yet. (I know, I know. A thousand hexes upon me and all my descendants for failing to produce this. but figured I'd go ahead and open up the issue to see if I need to bother.)

If this is something we have to give up in return for a better allocator, then I can accept that and either stick with 4.2, or try some deeper debugging tools to chase the wild goose. If instead this is an unintended regression, I am happy to assist however I can.

On that note...Any recommendations for static memory overrun analysis? Static stack usage profile? I've done what I can with cppcheck, clang-tidy, etc. but I'd give my right arm for a [stack usage view](https://software-dl.ti.com/ccs/esd/documents/ccs_stack_usage_view.html) like that from Texas Instruments Code Composer Studio, or the [command line equivalent](http://web.archive.org/web/20210114142101/https://processors.wiki.ti.com/index.php/Stack_issues#Finding_out_static_stack_usage) from before that was added

ftab, 2021-12-06:
This is still a thing. Is there a way to enable more comprehensive heap corruption detection while using the new allocator?

ftab, 2021-12-08:
If we revert to the heap component from IDF 4.2 then the comprehensive debugging works

SoucheSouche, 2022-08-16:
Hi @ftab,
Sorry for letting this issue hang for so long.
The issue is finally being addressed and will be fixed shortly in our upstream master. It will be backported to 5.0 and 4.4 as well.

Alvin1Zhang, 2022-11-16:
* Fix on release/5.0 is available at https://github.com/espressif/esp-idf/commit/aad557d70b8b5d2f329c012c8c5a0c229fc17578
* Fix on release/4.4 is at https://github.com/espressif/esp-idf/commit/a29f770c7a4bb643144f76ccb055269c33667f81

Thanks for reporting, feel free to reopen.

closed 2022-11-16 by Alvin1Zhang