espressif/esp-idf #7367: How to disable/enable all interrupts in ESP32 for handling critical section

vntsn, 2021-08-04:
## Environment

- Development Kit:  ESP32-Mini-1
- Kit version : 
- Module or chip used: ESP32
- IDF version:v4.2.1
- Build System: Make|CMake|idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: Eclipse
- Power Supply: USB

## Problem Description

I want to disable/enable all interrupts for handling critical section, for this I have identified below APIs 
portDISABLE_INTERRUPTS()
portENABLE_INTERRUPTS()
These are defined in file portmacro.h, which is located at  <project_dir>\esp-idf\components\freertos\xtensa\include\freertos\
These APIs are also used in below esp-idf components
[image]

Test code for validation of above APIs:
For validation I have used timer interrupt, using example code : \esp-idf\examples\peripherals\timer_group\main\
In this example code , I added the interrupt counter and printing the interrupt counter value before and after enabling/disabling the interrupt, if these APIs are working fine then interrupt counter should be same or count should be less when all interrupts are disabled. Timer interval is reduced to 0.0001 sec. 
Highlighting the updates in example code,
[image]
[image]
[image]

I have tried using below APIs also, but no luck
portENTER_CRITICAL( &spinlock );
portEXIT_CRITICAL( &spinlock );

Documentation referred:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/intr_alloc.html?highlight=interrupt#overview
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/freertos-smp.html

### Expected Behavior
Interrupt should not be serviced during critical section when all interrupts are disabled using portDISABLE_INTERRUPTS(), hence interrupt count in above test code should be zero interrupts are disabled.

### Actual Behavior
Interrupt count is same in both cases given below
1. portDISABLE_INTERRUPTS() and portENABLE_INTERRUPTS() called to handle critical section.
2. Interrupts are not disabled.

### Steps to reproduce
Explained in the description with test code.

### Code to reproduce this issue

```cpp
#include <stdio.h>
#include "esp_types.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "driver/periph_ctrl.h"
#include "driver/timer.h"
#include "freertos/portmacro.h"

#define TIMER_DIVIDER         16  //  Hardware timer clock divider
#define TIMER_SCALE           (TIMER_BASE_CLK / TIMER_DIVIDER)  // convert counter value to seconds
#define TIMER_INTERVAL0_SEC   (0.0001) // sample test interval for the first timer
#define TIMER_INTERVAL1_SEC   (0.0001)   // sample test interval for the second timer
#define TEST_WITHOUT_RELOAD   0        // testing will be done without auto reload
#define TEST_WITH_RELOAD      1        // testing will be done with auto reload

/*
 * A sample structure to pass events
 * from the timer interrupt handler to the main program.
 */
typedef struct {
    int type;  // the type of timer's event
    int timer_group;
    int timer_idx;
    uint64_t timer_counter_value;
} timer_event_t;

xQueueHandle timer_queue;

/*
 * A simple helper function to print the raw timer counter value
 * and the counter value converted to seconds
 */
static void inline print_timer_counter(uint64_t counter_value)
{
    printf("Counter: 0x%08x%08x\n", (uint32_t) (counter_value >> 32),
           (uint32_t) (counter_value));
    printf("Time   : %.8f s\n", (double) counter_value / TIMER_SCALE);
}

/*
 * Timer group0 ISR handler
 *
 * Note:
 * We don't call the timer API here because they are not declared with IRAM_ATTR.
 * If we're okay with the timer irq not being serviced while SPI flash cache is disabled,
 * we can allocate this interrupt without the ESP_INTR_FLAG_IRAM flag and use the normal API.
 */
void IRAM_ATTR timer_group0_isr(void *para)
{
	interrupt_cntr++;
    timer_spinlock_take(TIMER_GROUP_0);
    int timer_idx = (int) para;

    /* Retrieve the interrupt status and the counter value
       from the timer that reported the interrupt */
    uint32_t timer_intr = timer_group_get_intr_status_in_isr(TIMER_GROUP_0);
    uint64_t timer_counter_value = timer_group_get_counter_value_in_isr(TIMER_GROUP_0, timer_idx);

    /* Prepare basic event data
       that will be then sent back to the main program task */
    timer_event_t evt;
    evt.timer_group = 0;
    evt.timer_idx = timer_idx;
    evt.timer_counter_value = timer_counter_value;

    /* Clear the interrupt
       and update the alarm time for the timer with without reload */
    if (timer_intr & TIMER_INTR_T0) {
        evt.type = TEST_WITHOUT_RELOAD;
        timer_group_clr_intr_status_in_isr(TIMER_GROUP_0, TIMER_0);
        timer_counter_value += (uint64_t) (TIMER_INTERVAL0_SEC * TIMER_SCALE);
        timer_group_set_alarm_value_in_isr(TIMER_GROUP_0, timer_idx, timer_counter_value);
    } else if (timer_intr & TIMER_INTR_T1) {
        evt.type = TEST_WITH_RELOAD;
        timer_group_clr_intr_status_in_isr(TIMER_GROUP_0, TIMER_1);
    } else {
        evt.type = -1; // not supported even type
    }

    /* After the alarm has been triggered
      we need enable it again, so it is triggered the next time */
    timer_group_enable_alarm_in_isr(TIMER_GROUP_0, timer_idx);

    /* Now just send the event data back to the main program task */
    xQueueSendFromISR(timer_queue, &evt, NULL);
    timer_spinlock_give(TIMER_GROUP_0);
}

/*
 * Initialize selected timer of the timer group 0
 *
 * timer_idx - the timer number to initialize
 * auto_reload - should the timer auto reload on alarm?
 * timer_interval_sec - the interval of alarm to set
 */
static void example_tg0_timer_init(int timer_idx,
                                   bool auto_reload, double timer_interval_sec)
{
    /* Select and initialize basic parameters of the timer */
    timer_config_t config = {
        .divider = TIMER_DIVIDER,
        .counter_dir = TIMER_COUNT_UP,
        .counter_en = TIMER_PAUSE,
        .alarm_en = TIMER_ALARM_EN,
        .auto_reload = auto_reload,
    }; // default clock source is APB
    timer_init(TIMER_GROUP_0, timer_idx, &config);

    /* Timer's counter will initially start from value below.
       Also, if auto_reload is set, this value will be automatically reload on alarm */
    timer_set_counter_value(TIMER_GROUP_0, timer_idx, 0x00000000ULL);

    /* Configure the alarm value and the interrupt on alarm. */
    timer_set_alarm_value(TIMER_GROUP_0, timer_idx, timer_interval_sec * TIMER_SCALE);
    timer_enable_intr(TIMER_GROUP_0, timer_idx);
    timer_isr_register(TIMER_GROUP_0, timer_idx, timer_group0_isr,
                       (void *) timer_idx, ESP_INTR_FLAG_IRAM, NULL);

    timer_start(TIMER_GROUP_0, timer_idx);
}

/*
 * The main task of this example program
 */
static void timer_example_evt_task(void *arg)
{
    while (1) {
        printf( "\n -------------------------------------------------------------------------------");
    	printf( "\n Interrupt disabled interrupt_cntr = %d **********************", interrupt_cntr );
//    	portENTER_CRITICAL( &spinlock );
    	portDISABLE_INTERRUPTS();
        timer_event_t evt;
        xQueueReceive(timer_queue, &evt, portMAX_DELAY);

        /* Print information that the timer reported an event */
        if (evt.type == TEST_WITHOUT_RELOAD) {
//            printf("\n    Example timer without reload\n");
        } else if (evt.type == TEST_WITH_RELOAD) {
//            printf("\n    Example timer with auto reload\n");
        } else {
//            printf("\n    UNKNOWN EVENT TYPE\n");
        }
        uint64_t task_counter_value;
        timer_get_counter_value(evt.timer_group, evt.timer_idx, &task_counter_value);
//        print_timer_counter(task_counter_value);
        portENABLE_INTERRUPTS();
//        portEXIT_CRITICAL( &spinlock );
        printf( "\n Interrupt  interrupt_cntr = %d **********************", interrupt_cntr );
        vTaskDelay( 10 );
    }
}

/*
 * In this example, we will test hardware timer0 and timer1 of timer group0.
 */
void app_main(void)
{
    timer_queue = xQueueCreate(10, sizeof(timer_event_t));
    example_tg0_timer_init(TIMER_0, TEST_WITHOUT_RELOAD, TIMER_INTERVAL0_SEC);
    example_tg0_timer_init(TIMER_1, TEST_WITH_RELOAD,    TIMER_INTERVAL1_SEC);
    xTaskCreate(timer_example_evt_task, "timer_evt_task", 2048, NULL, 5, NULL);
}
```

## Debug Logs
Output is same in both cases
1.Interrupts are disabled using portDISABLE_INTERRUPTS() and portENABLE_INTERRUPTS()
2. Interrupts are not disabled.
[image]

vntsn, 2021-08-05:
@igrr Thanks for your response!
any suggestion on how can we confirm that interrupt is not occurring in critical section, I checked that context switching is also happening if I use portENTER_CRITICAL( &spinlock ); or portDISABLE_INTERRUPTS(); (reason of this could be the same that you have mentioned). how can we validate these functions.

igrr, 2021-08-05:
It seems to be an XY problem — it look like you want to disable interrupts to achieve certain goal in your application, and the way you are applying this method doesn't yield the result you expect. So you suspect that there is something wrong with the method.

If you need to protect a shared variable accessed by tasks and ISRs, using portENTER/EXIT_CRITICAL is the correct approach. You need to use the same portMUX_TYPE spinlock in every place that shared variable is accessed.

Beyond that, any other use of portENTER_CRITICAL in an application is usually not the correct way on a mult-core system. For example, disabling interrupts does prevent the tick interrupt from occurring, but only on the CPU where the interrupts are disabled. The scheduler still runs on the other CPU, and can execute tasks which have either no affinity or those pinned to that other CPU.

As for testing that interrupts get disabled, you can try using the following logic:

- enter critical section
- assign interrupt_cntr_0 = interrupt_cntr
- busy-wait for a short period of time (`usleep(1000)`)
- assign interrupt_cntr_1 = interrupt_cntr
- exit the critical section
- display the difference between the two samples.

A few caveats:
- FreeRTOS doesn't support blocking inside a section of code where task switch cannot happen. For example, calling xQueueReceive or vTaskDelay inside a critical section is not supported.
- When use usleep to wait for an interval longer than the tick period, it will use vTaskDelay; for shorter periods it will busy-wait.

vntsn, 2021-08-06:
Thanks @igrr  I am getting expected output with the logic you have shared.

closed 2021-08-06 by vntsn