espressif/esp-idf #7368: Flash ops. panicking due to SPI ISR accessed code not in IRAM

KaeLL, 2021-08-04:
## Environment
- IDF version: v4.4-dev-2487-g8131d6f46

## Problem Description

Flash ops cause panics due to SPI's ISR being initialized with the `ESP_INTR_FLAG_IRAM` flag while the [SPI code that disables the ISR](https://github.com/espressif/esp-idf/blob/8131d6f46d690e9cb60cc5cd457863cc5479351f/components/driver/spi_master.c#L505) isn't in IRAM.

### Code to reproduce this issue

```cpp
spi_bus_initialize(VSPI_HOST, &(spi_bus_config_t){ .intr_flags = ESP_INTR_FLAG_IRAM, /* ... */ }, 1);
// Perform flash ops, like NVS, etc = crash
```

ginkgm, 2021-08-05:
Hi @KaeLL ,

Thanks for reporting. What's the difference with https://github.com/espressif/esp-idf/issues/6781?

KaeLL, 2021-08-05:
So the deja-vu wasn't coincidence.
Btw, the fact no one could be bothered to copy and paste a macro after 6 months speaks volumes.

closed 2021-08-06 by espressif-bot