espressif/esp-idf #7369: BOOTLOADER_OTA_DATA_ERASE is not working

pbz-hyunwoo, 2021-08-05:
## Environment

- Development Kit:        **none**
- Module or chip used: **ESP32**
- IDF version :                **v4.2.2**
- Build System:              **idf.py**
- Compiler version :      **8.4.0**
- Operating System:      **Windows**
- environment type:      **Plain Command Prompt**
- Using an IDE?:            **VS code** (editing source code only)
- Power Supply:             **USB**

## Problem Description
<BOOTLOADER_OTA_DATA_DRASE> is not working.

I tested switch partition between Factory partition & OTA partition using switch.

Firmware in Factory partition is for BLE & Update OTA.
And OTA partition firmware is for communication with other module via UART. (like, GPS)

**A. normal test flow to check OTA update is done and switching no problem. (always succeed)**
  1. At first, erase_flash in idf.py and flash factory firmware to ESP32.
  2. Connect ESP32 and my phone for OTA update. OTA update is perfoming by own app.
  3. After all of downloading and writing OTA firmware to OTA partition, restart ESP32.
  4. Check output of OTA firmware on ESP32.
  5. Push switch to change to Factory firmware.
  6. Push switch to change to OTA firmware.

**B. Factory Boot with clear OTA partition test flow. (after normal test)**
  1. If now firmware is not OTA firmware, push switch to change to OTA firmware.
  2. After booting, hard reset (Chip_PU) and push trigger factory reset pin at the same time.
  3. Check what firmware is loaded.
  4. Push switch to change to OTA firmware.
  5. Check what firmware is loaded.

### Expected Behavior
At 5. in **B test**, I excepted below message and reboot to firmware partition
`E (20808) esp_image: image at 0x110000 has invalid magic byte`

### Actual Behavior

But performed OTA firmware booting.

I think my test code is not necessary, because it just update ota partition and read gpio level for change partition like below codes.

#### To Factory firmware
`
            esp_partition_t const* factory_partition;
            factory_partition = esp_partition_find_first( ESP_PARTITION_TYPE_APP,
                                                          ESP_PARTITION_SUBTYPE_APP_FACTORY,
                                                          "factory" );
            esp_ota_set_boot_partition(factory_partition);
            esp_restart();
`
#### To OTA firmware
`            
            esp_partition_t const* factory_partition;
            factory_partition = esp_partition_find_first( ESP_PARTITION_TYPE_APP,
                                                          ESP_PARTITION_SUBTYPE_APP_OTA_0,
                                                          "ota_0" );
            esp_ota_set_boot_partition(factory_partition);
            esp_restart();
`

Is this problem is general issue? or just user error by me?

pbz-hyunwoo, 2021-08-05:
If there is no way to clear OTA partition via bootloader,
Can I detect Factory Reset on running? (like `ESP_ERR_OTA_ROLLBACK_INVALID_STATE` macro)

KonstantinKondrashov, 2021-08-05:
Hi @pbz-hyunwoo !
Seems you meant CONFIG_BOOTLOADER_OTA_DATA_ERASE. This option doesn't erase apps, it erases only otadata partition.
If the running app was ota_1 or ota_0 then after factory reset you should see the factory app running.

nvs,      data, nvs,     ,        0x4000,
otadata,  data, ota,     ,        0x2000,  <--------------- this partition is erased
phy_init, data, phy,     ,        0x1000,
factory,  app,  factory, ,        1M,
ota_0,    app,  ota_0,   ,        1M,
ota_1,    app,  ota_1,   ,        1M,

pbz-hyunwoo, 2021-08-05:
Ahha, I misunderstood.

Then, there is no way to erase app partition?
Can I detect otadata partition is erased?

KonstantinKondrashov, 2021-08-05:
@pbz-hyunwoo 
In factory app, you can call `esp_partition_erase_range(app_partition,....)` to erase ota_0&1 (not running app).

There is a way that you can implement your way of factory reset using `Custom bootloader` https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/bootloader.html#custom-bootloader
to do it here https://github.com/espressif/esp-idf/blob/master/components/bootloader/subproject/main/bootloader_start.c#L124-L127

pbz-hyunwoo, 2021-08-06:
Hi, Thank you for respond.

pbz-hyunwoo, 2021-08-06:
@KonstantinKondrashov 
Is it possible to erase app partition in custom bootloader?
<details>
<summary>bootloader compiled with only these headers, so cannot import `esp_partition.h`</summary>

```
-IConfig
-IC:/Users/<MyName>/Desktop/esp-idf/components/log/include
-IC:/Users/<MyName>/Desktop/esp-idf/components/esp_rom/include
-IC:/Users/<MyName>/Desktop/esp-idf/components/esp_common/include
-IC:/Users/<MyName>/Desktop/esp-idf/components/xtensa/include
-IC:/Users/<MyName>/Desktop/esp-idf/components/xtensa/esp32/include
-IC:/Users/<MyName>/Desktop/esp-idf/components/esp32/include
-IC:/Users/<MyName>/Desktop/esp-idf/components/bootloader_support/include
-IC:/Users/<MyName>/Desktop/esp-idf/components/bootloader_support/include_bootloader
-IC:/Users/<MyName>/Desktop/esp-idf/components/soc/src/esp32/.
-IC:/Users/<MyName>/Desktop/esp-idf/components/soc/src/esp32/include
-IC:/Users/<MyName>/Desktop/esp-idf/components/soc/include
-IC:/Users/<MyName>/Desktop/esp-idf/components/soc/soc/esp32/include
-IC:/Users/<MyName>/Desktop/esp-idf/components/soc/soc/esp32/../include
-IC:/Users/<MyName>/Desktop/esp-idf/components/soc/soc/esp32/private_include
```
</details>
But I found `bootloader_flash.h` in `component/bootloader_support/include_bootloader`.

Is it ok to erase ota data partition by this function in `bootloader_flash.h`?

 ```
esp_err_t bootloader_flash_write(size_t dest_addr, void *src, size_t size, bool write_encrypted);
```

KonstantinKondrashov, 2021-08-06:
Hi @pbz-hyunwoo !
Yes, it is possible but it will take time of course `bootloader_flash_erase_range(partition->pos.offset, partition->pos.size);`
You can take a look at this code as an example:
https://github.com/espressif/esp-idf/blob/master/components/bootloader_support/src/bootloader_common.c#L99-L151

pbz-hyunwoo, 2021-08-06:
When import `bootloader_flash.h` to `bootloader_start.c`, "**No such file or directory**" error is occurred because of `esp_spi_flash.h`.
Maybe include path problem.

pbz-hyunwoo, 2021-08-10:
@KonstantinKondrashov 
Would you help me?

I cannot add that edit partition code to custom bootloader.

KonstantinKondrashov, 2021-08-10:
Hi @pbz-hyunwoo!
To customize the bootloader is is enough to copy just `bootloader`  folder - the bootloader folder from `esp-idf/components/bootloader` to `path_project/components/bootloader`. See the image below.
[image]

pbz-hyunwoo, 2021-08-10:
Thank you for respond, @KonstantinKondrashov.
But I still have some questions.

1. It doesn't work copy & paste component/bootloader directory to my project, and I cannot find about copy & paste to my project in useful example. So I can't guess what I missed.
2. In this [comment](https://github.com/espressif/esp-idf/issues/7369#issuecomment-894048002), I override the bootloader, so I added a log that custom message in booting and checked succeed it. But cannot include `bootloader_flash.h` at `bootloader_start.c`. Is there the way to modify flash memory in `bootloader_start.c`?
3. When I override `bootloader_start.c`, I added 4 messages:
```
    if (bootloader_common_get_reset_reason(0) != DEEPSLEEP_RESET) {
        // Factory firmware.
#ifdef CONFIG_BOOTLOADER_FACTORY_RESET
        /* Custom LOG Messege 1 */
        if (bootloader_common_check_long_hold_gpio(CONFIG_BOOTLOADER_NUM_PIN_FACTORY_RESET, CONFIG_BOOTLOADER_HOLD_TIME_GPIO) == 1) {
            ESP_LOGI(TAG, "Detect a condition of the factory reset");
            bool ota_data_erase = false;
#ifdef CONFIG_BOOTLOADER_OTA_DATA_ERASE
            /* Custom LOG Messege 2 */
            ota_data_erase = true;
#endif
            const char *list_erase = CONFIG_BOOTLOADER_DATA_FACTORY_RESET;
            ESP_LOGI(TAG, "Data partitions to erase: %s", list_erase);
            /* Custom LOG Messege 3 */
            if (bootloader_common_erase_part_type_data(list_erase, ota_data_erase) == false) {
                ESP_LOGE(TAG, "Not all partitions were erased");
            }
            /* Custom LOG Messege 4 */
            return bootloader_utility_get_selected_boot_partition(bs);
        }
```
but ONLY /* Custom LOG Messege 1 */ was printed. Other messages are never printed. (Of course, I set CONFIG_BOOTLOADER_OTA_DATA_ERASE)

pbz-hyunwoo, 2021-08-13:
I still have a problem, so reopen this issue.

KonstantinKondrashov, 2021-08-13:
Hi @pbz-hyunwoo!

> Some useful examples about custom_bootloader - https://github.com/espressif/esp-idf/tree/83956eb/examples/custom_bootloader

They are related to master and no in v4.2.2 that you use.

You use 4.2.2, for this IDF version https://docs.espressif.com/projects/esp-idf/en/v4.2.2/esp32/api-guides/bootloader.html#customer-bootloader.

Please try this way: 
* copy bootloader  from esp-idf/components/bootloader to path_project/components/bootloader
* copy bootloader_support from esp-idf/components/bootloader_support to path_project/components/bootloader/subproject/components/bootloader_support.
After that, you can modify the bootloader_support  and bootloader.
[image]

For your goal, I might suggest to modify `bootloader_common_erase_part_type_data()` as shown below to erase app there.
In this case, you need to add to the CONFIG_BOOTLOADER_DATA_FACTORY_RESET the list of erasing partitions (like ota_1, ota_0, nvs).

```diff
--- a/components/bootloader_support/src/bootloader_common.c
+++ b/components/bootloader_support/src/bootloader_common.c
@@ -134,9 +134,9 @@ bool bootloader_common_erase_part_type_data(const char *list_erase, bool ota_dat
         for (int i = 0; i < num_partitions; i++) {
             const esp_partition_info_t *partition = &partitions[i];
             char label[sizeof(partition->label) + 1] = {0};
-            if (partition->type == PART_TYPE_DATA) {
+            if (1) {
                 bool fl_ota_data_erase = false;
-                if (ota_data_erase == true && partition->subtype == PART_SUBTYPE_DATA_OTA) {
+                if (ota_data_erase == true && partition->subtype == PART_SUBTYPE_DATA_OTA && partition->type == PART_TYPE_DATA) {
                     fl_ota_data_erase = true;
                 }
                 // partition->label is not null-terminated string.
@@ -153,7 +153,7 @@ bool bootloader_common_erase_part_type_data(const char *list_erase, bool ota_dat
                     marker = "no";
                 }
 
-                ESP_LOGI(TAG, "%2d %-16s data  %08x %08x [%s]", i, partition->label,
+                ESP_LOGI(TAG, "%2d %-16s %d  %08x %08x [%s]", i, partition->label, partition->type,
                          partition->pos.offset, partition->pos.size, marker);
             }
         }
```

closed 2021-08-24 by Alvin1Zhang