espressif/esp-idf #7370: does mbedTLS  actually use external RAM?

JB-DX, 2021-08-05:
Using IDF v4.4-dev-479-g1067b2870-dirty

On a ESP-CAM with external RAM calling esp_http_client_open() with https url fails because of insufficient internal heap space. Thus I configured mbedTLS to "Memory allocation strategy (External SPIRAM)"
However that doesn't help much. I now get this:
```
---- stack overflow 
0x4008e95e: panic_abort at P:/ESP32/esp-idf/components/esp_system/panic.c:354

0x4008f169: esp_system_abort at P:/ESP32/esp-idf/components/esp_system/system_api.c:112

0x40092cf9: vApplicationStackOverflowHook at P:/ESP32/esp-idf/components/freertos/port/xtensa/port.c:489

0x4009115d: vTaskSwitchContext at P:/ESP32/esp-idf/components/freertos/tasks.c:3270

0x4008f260: _frxt_dispatch at P:/ESP32/esp-idf/components/freertos/port/xtensa/portasm.S:432

0x4008f212: _frxt_int_exit at P:/ESP32/esp-idf/components/freertos/port/xtensa/portasm.S:231

0x400857ac: _xt_lowint1 at P:/ESP32/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1098

0x40080400: _init at ??:?

```
Before calling esp_http_client_open() the heapSize is: internal:38911, external: 2234111

I tried the same thing on a non-ESP-CAM module without external RAM, 
heapSize is 207528 and 36 kB less after esp_http_client_open() 

So it should be able to run even when heapsize is 38.9 kb, especially when external RAM is used.

Q: isn't mbedTLS able to use external RAM?
What else can I do?

I can't set log to verbose or debug because that ends in a stack overlow also. I tried increasing / decreasing buffer size in menuconfig, but nothing helped. 
So I am running out of ideas.

mahavirj, 2021-08-05:
@JB-DX 

mbedTLS should redirect buffers to external SPIRAM if relevant config is enabled, you may check using some debug prints at:

https://github.com/espressif/esp-idf/blob/8131d6f46d690e9cb60cc5cd457863cc5479351f/components/mbedtls/port/esp_mem.c#L26-L27

Additionally, I would request to characterize memory using example from https://github.com/espressif/esp-idf/tree/master/examples/protocols/https_request, which creates task with sufficiently larger stack size and hence you should not face problem with logging level etc.

JB-DX, 2021-08-05:
Thank you very much !
That helped me to find the problem.  mbedTLS indeed uses external RAM. However the stack size of main() was too small with its default value. Increasing it to 6 kB now allows even verbose log-messages without crashing. It turned out I was looking at the wrong spot.

Thanks you again, case is closed now .

closed 2021-08-05 by JB-DX