espressif/esp-idf #7373: Crash code when using esp_register_freertos_tick_hook() at 1000Hz with gatt_security_server in example(ESP_BLE_ONLY_ACCEPT_SPECIFIED_AUTH_ENABLE)

VardhamanJain, 2021-08-05:
We are facing crashing issue in [**gatt_security_server**](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_security_server) example code.
We are using 
```
esp_ble_io_cap_t iocap = ESP_IO_CAP_OUT;
uint8_t auth_option = ESP_BLE_ONLY_ACCEPT_SPECIFIED_AUTH_ENABLE;
```

We are also using **esp_register_freertos_tick_hook()** function with **Tick rate 1000Hz**.

- We tried with different Tick Rate (10Hz, 100Hz and 1000Hz).
- It is working with (Tick Rate : 10Hz, 100Hz)
- **It is crashing with (Tick Rate : 1000Hz)**

Please fix this issue and give some solution.

**Patch File:** 

- [Test_esp_register_freertos_tick_hook_feature_with_ example_ble_sec_gatts_demo.c__.patch.txt](https://github.com/espressif/esp-idf/files/6940436/Test_esp_register_freertos_tick_hook_feature_with_.example_ble_sec_gatts_demo.c__.patch.txt)

**Log Files:** 

- [BLE_EXAMPLE_WITH_TICK_HOOK_10HZ.txt](https://github.com/espressif/esp-idf/files/6940439/BLE_EXAMPLE_WITH_TICK_HOOK_10HZ.txt)
- [BLE_EXAMPLE_WITH_TICK_HOOK_100HZ.txt](https://github.com/espressif/esp-idf/files/6940440/BLE_EXAMPLE_WITH_TICK_HOOK_100HZ.txt)
- [BLE_EXAMPLE_WITH_TICK_HOOK_1000HZ_Crashing.txt](https://github.com/espressif/esp-idf/files/6940441/BLE_EXAMPLE_WITH_TICK_HOOK_1000HZ_Crashing.txt)

**Git Repo:** https://github.com/espressif/esp-idf.git
**Git Tag:** v4.2.1
**Commit ID:** 21ecef5c56e01d3cccf0648c869e3fa9248bf3d4

**Note:** We are using "[nRF Connect for Mobile](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en_IN&gl=US)"  Android Application as GATT Client on Mobile [Moto G4 Plus(Android Version 7.0)].

Code Changes
===================================================================
```
diff --git a/examples/bluetooth/bluedroid/ble/gatt_security_server/main/example_ble_sec_gatts_demo.c b/examples/bluetooth/bluedroid/ble/gatt_security_server/main/example_ble_sec_gatts_demo.c
index 6cbfb7182..0dec204bb 100644
--- a/examples/bluetooth/bluedroid/ble/gatt_security_server/main/example_ble_sec_gatts_demo.c
+++ b/examples/bluetooth/bluedroid/ble/gatt_security_server/main/example_ble_sec_gatts_demo.c
@@ -19,6 +19,7 @@
 #include "esp_bt_defs.h"
 #include "esp_bt_main.h"
 #include "example_ble_sec_gatts_demo.h"
+#include "esp_freertos_hooks.h"
 
 #define GATTS_TABLE_TAG "SEC_GATTS_DEMO"
 
@@ -497,10 +498,24 @@ static void gatts_event_handler(esp_gatts_cb_event_t event, esp_gatt_if_t gatts_
     } while (0);
 }
 
+static uint32_t tick_count = 0;
+static void Tick_Event ()
+{
+	tick_count++;
+}
+
 void app_main(void)
 {
     esp_err_t ret;
 
+	// Register Hook callback mechanism using ESP32 API
+	// This mechanism works and we are able to create and run multiple CR Tasker
+	esp_err_t register_tick_hook = esp_register_freertos_tick_hook( Tick_Event );
+	if ( register_tick_hook != ESP_OK )
+	{
+		printf( "\n Error in registering tick hook: %d", register_tick_hook );
+	}
+
     // Initialize NVS.
     ret = nvs_flash_init();
[... 134 lines cut ...]
0x4008ab59: spi_flash_read at D:/esp-idf/components/spi_flash/esp_flash_api.c:762

0x40125d82: nvs::Page::load(unsigned int) at D:/esp-idf/components/nvs_flash/src/nvs_page.cpp:57

0x40126188: nvs::PageManager::load(unsigned int, unsigned int) at D:/esp-idf/components/nvs_flash/src/nvs_pagemanager.cpp:29

0x40123978: nvs::Storage::init(unsigned int, unsigned int) at D:/esp-idf/components/nvs_flash/src/nvs_storage.cpp:93

0x401248f9: nvs::NVSPartitionManager::init_custom(char const*, unsigned int, unsigned int) at D:/esp-idf/components/nvs_flash/src/nvs_partition_manager.cpp:71

0x40124971: nvs::NVSPartitionManager::init_partition(char const*) at D:/esp-idf/components/nvs_flash/src/nvs_partition_manager.cpp:52

0x401233df: nvs_flash_init_partition at D:/esp-idf/components/nvs_flash/src/nvs_api.cpp:163

0x40123402: nvs_flash_init at D:/esp-idf/components/nvs_flash/src/nvs_api.cpp:168

0x400d4fe5: app_main at d:\esp-idf\examples\bluetooth\bluedroid\ble\gatt_security_server\build/../main/example_ble_sec_gatts_demo.c:520

0x400d364e: main_task at D:/esp-idf/components/esp32/cpu_start.c:609 (discriminator 2)

0x4008ee21: vPortTaskWrapper at D:/esp-idf/components/freertos/xtensa/port.c:143

ELF file SHA256: 8a7a5f4fcfb41e1d

```

negativekelvin, 2021-08-05:
Try placing in iram

https://github.com/espressif/esp-idf/blob/1cb31e50943bb757966ca91ed7f4852692a5b0ed/components/esp_timer/test/test_esp_timer.c#L682

VardhamanJain, 2021-08-09:
> Try placing in iram
> 
> https://github.com/espressif/esp-idf/blob/1cb31e50943bb757966ca91ed7f4852692a5b0ed/components/esp_timer/test/test_esp_timer.c#L682

Thank you @negativekelvin , It fixed.

Alvin1Zhang, 2021-09-13:
Thanks for reporting and sharing updates, feel free to reopen.

closed 2021-09-13 by Alvin1Zhang