espressif/esp-idf #7374: TWAI alert for RX (like UART_DATA)

crackwitz, 2021-08-05:
Use case: I need to send and receive CAN frames. I use TX_SUCCESS to confirm that a frame was sent. My task tries `twai_receive()` and `twai_read_alerts()` in a loop (`twai_transmit` asynchronously from elsewhere). Currently I have to either call both with 0 timeout, which makes the idle task's watchdog angry, or I give one of them a timeout... which delays the other call: either I get TX_SUCCESS 10 ms later than necessary or I may *lose* CAN frames because I'm spending 10 ms waiting for alerts.

I'd rather not have to use separate tasks for this.

I think a TWAI alert would be fine that indicates messages waiting in the RX queue and to try twai_receive (with 0 timeout repeatedly until the queue is empty).

The UART peripheral also has "alerts", but calls them "events", and it has `UART_DATA`, which indicates data waiting in the input queue. The idea applies to multiple peripherals but it has different names and implementations.

I've considered adding such a TWAI alert into esp-idf myself but it would have to be upstreamed because I can't justify the effort of maintaining a patch beside esp-idf.

If this idea or an alternative is well received, I might look into making a pull request.

joncmaloney, 2021-08-14:
We also have this issue with TWAI interface. We have ended up using 2 threads one for TX and another for RX. It would be much better if there was an implementation that allowed for operations with only one thread. 

One solution could be that a tx_queue and rx_queue are added to twai_general_config_t structure and passed into the twai_driver_install function. This way the user could add both these queues into a queue set and then wait for either a tx frame or an rx frame by blocking on the queue set.

closed 2021-08-29 by espressif-bot

dev-techshlok, 2022-06-14:
Hello @crackwitz 

Amazing contribution. 

How can we use this in the main code, do we need to make a function and pass that to interrupt, can you give a small example, please.

Thank you.

crackwitz, 2022-06-14:
- Documentation of alerts: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/twai.html#alerts

- Docs for `twai_read_alerts`: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/twai.html#_CPPv416twai_read_alertsP8uint32_t10TickType_t

- Example for alerts: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/twai/twai_alert_and_recovery