espressif/esp-idf #7377: Custom Header in .cpp file is not reflecting in bin file

vim-git, 2021-08-06:
## Environment

- Development Kit: ESP32-Mini-1
-Kit version :
-Module or chip used: ESP32
-IDF version:v4.2.1
-Build System: Make|CMake|idf.py
-Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
-Operating System: Windows
-(Windows only) environment type: ESP Command Prompt
-Using an IDE?: Eclipse
-Power Supply: USB

## Problem Description

We are defining the custom header in .cpp file. But same has been working in .c file ( We can't use .c file due to some dependency )
const __attribute__((section(".rodata_custom_desc"))) foo_app_desc_t foo_app_desc = {..}

added target_link_libraries(${COMPOENENT_TARGET} "-u foo_app_desc") in CMakeLists.txt file.

But it is not reflecting in .bin file or in .map file under ".rodata_custom_desc" section.
[image]

Documentation referred:- https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/app_image_format.html#adding-a-custom-structure-to-an-application 
### Expected Behavior

After defining the custom header and added into the CMakeLists.txt file, It should reflect in .bin and .map file under ".rodata_custom_desc" section.

### Actual Behavior
After defining custom header and adding in CMakeLists.txt, Build is getting successful but defined custom header is not present in .bin and in .map file under ".rodata_custom_desc" section.

When we are defining it in .c file, It is reflecting in .bin and in .map under ".rodata_custom_desc" section as well.

### Steps to reproduce

1. Defined a structure foo_app_desc_t.
2. Defined custom header
3. add target_link_libraries(${COMPOENENT_TARGET} "-u foo_app_desc") in CMakeLists.txt file Where custom header defined file is present.

[image]

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
typedef struct foo_app_desc_t
{
	unsigned int version;
	unsigned int serial_number;
	unsigned int os_version;
	unsigned int header_version;
	unsigned int header_size;
	unsigned int app_size;
	unsigned int reserved[10];
} foo_app_desc_t;

const __attribute__((section(".rodata_custom_desc"))) foo_app_desc_t foo_app_desc = {
	.version  = 1,
	.serial_number = 2,
	.os_version = 10,
	.header_version = 0,
	.header_size = 1024,
	.app_size = 10240,
	.reserved = { 0 }
}
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

### Debug Logs

[image]

```
### Other items if possible
If we are following same method in .c file, It is working 
![image](https://user-images.githubusercontent.com/49974143/128466645-0daffc9b-a075-4c1e-9750-1787e8a6e45b.png)

boarchuz, 2021-08-06:
https://www.esp32.com/viewtopic.php?f=13&t=18612#p69279

vim-git, 2021-08-08:
Thanks for your reference, But I am still facing the issue. As per my understanding you are defining 
"const __attribute__((section(".rodata_custom_desc"))) foo_app_desc_t foo_app_desc = {..}" in foo.c file and accessing this in .h file using extern "C" in any.cpp file.  This scenario is working for me as well. 

Since I wanted to define "const __attribute__((section(".rodata_custom_desc"))) foo_app_desc_t foo_app_desc = {..}" in foo.cpp file because I have CPP dependency while defining this. In this scenario It is not working for me. Please guide me to fix in this scenario.

igrr, 2021-08-09:
@vim-git You can declare and define foo_app_desc with `extern "C"` keyword:

in .h file:
```cpp
#ifdef __cplusplus
extern "C" {
#endif

typedef struct foo_app_desc_t
{
	unsigned int version;
	unsigned int serial_number;
	unsigned int os_version;
	unsigned int header_version;
	unsigned int header_size;
	unsigned int app_size;
	unsigned int reserved[10];
} foo_app_desc_t;

extern const foo_app_desc_t foo_app_desc;

#ifdef __cplusplus
}
#endif
```

in cpp file:
```cpp
extern "C" const __attribute__((section(".rodata_custom_desc"))) foo_app_desc_t foo_app_desc = {
   /* initializer */
};
```

vim-git, 2021-08-10:
Thanks @igrr and @boarchuz for your quick response and suggested solutions. 

Now it is working for me, I was trying the extern "C" {...}, In this case it was not working but using extern "C" without Curled Brackets it is working.
Thanks @igrr

closed 2021-08-10 by vim-git