espressif/esp-idf #7381: Panic abort when using init_fs()

GeorgeFlorian, 2021-08-06:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version: v4
- Module or chip used:  ESP32-WROVER-B
- IDF version: ESP-IDF v4.4-dev-1849-g8e3e65a47
- Build System:         [Make|CMake|idf.py]
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System:     Linux Mint
- Power Supply:         USB

## Problem Description

I have compiled, built and uploaded the source code of [this library](https://github.com/DroneBridge/ESP32) with ESP-IDF 4.4, and it doesn't work:

```
I (826) DB_ESP32: AP started!
E (826) esp_semihost: OpenOCD is not connected!
E (826) DB_ESP32: Failed to register semihost driver (ESP_ERR_NOT_SUPPORTED)!
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40088894
0x40088894: _esp_error_check_failed at /home/georgeflorian/esp/esp-idf/components/esp_system/esp_err.c:42

file: "../main/main.c" line 260
func: app_main
expression: init_fs()

abort() was called at PC 0x40088897 on core 0
0x40088897: _esp_error_check_failed at /home/georgeflorian/esp/esp-idf/components/esp_system/esp_err.c:43

Backtrace:0x400d344e:0x3ffbc6f00x400888a5:0x3ffbc710 0x4008edf2:0x3ffbc730 0x40088897:0x3ffbc7a0 0x400d9ffb:0x3ffbc7c0 0x40152d3f:0x3ffbc7e0 
0x400d344e: panic_abort at /home/georgeflorian/esp/esp-idf/components/esp_system/panic.c:390

0x400888a5: esp_system_abort at /home/georgeflorian/esp/esp-idf/components/esp_system/esp_system.c:126

0x4008edf2: abort at /home/georgeflorian/esp/esp-idf/components/newlib/abort.c:46

0x40088897: _esp_error_check_failed at /home/georgeflorian/esp/esp-idf/components/esp_system/esp_err.c:43

0x400d9ffb: app_main at /home/georgeflorian/NewVolume/ESP-IDF/DroneBridgeMaster/build/../main/main.c:260 (discriminator 1)

0x40152d3f: main_task at /home/georgeflorian/esp/esp-idf/components/freertos/port/port_common.c:142
```

I have attached the sdkconfig and the .elf file:

[elf.zip](https://github.com/espressif/esp-idf/files/6946208/elf.zip)

atanisoft, 2021-08-06:
@GeorgeFlorian This looks to be a config issue in the project you are building. From the output you have provided it looks like it is entering [this](https://github.com/DroneBridge/ESP32/blob/master/main/main.c#L87-L96) method which is returning an error for your setup regarding unsupported "semihost" FS. You may need to enable `CONFIG_WEB_DEPLOY_SF` instead of `CONFIG_WEB_DEPLOY_SEMIHOST`

GeorgeFlorian, 2021-08-09:
> @GeorgeFlorian This looks to be a config issue in the project you are building. From the output you have provided it looks like it is entering [...]

It builds now.
Thank you.

closed 2021-08-09 by GeorgeFlorian