espressif/esp-idf #7382: idf.py has no 'bootloader-encrypted-flash' target

sU8U7SfkcwTJVH7PjaVmej7D, 2021-08-06:
## Environment

- Development Kit: ESP32-DevKitC
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.4-dev-2487-g8131d6f46d
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System:  Linux

## Problem Description

When using both **secure boot v2** and **flash encryption**, idf.py has no easy way to reflash the bootloader.

### Expected Behavior

`idf.py` has a 'bootloader-encrypted-flash' target which calls esptool with the `--encrypt` flag.

( Note: for other partitions, one can use 'encrypted-flash', but due to secure boot v2 being enabled, the bootloader partition is expectedly not flashed. )

### Actual Behavior

`idf.py` has no such target.

### Fix

Could you add a 'bootloader-encrypted-flash' target to `idf.py, is the same vein there is already 'flash' and 'encrypted-flash' ?

AshUK, 2021-08-10:
+1 would like to see this, very frequently I am testing boot loader configuration changes on a device that has flash encryption enabled. This would save me time having over having to copy/type out the esptool command every time.

mahavirj, 2022-11-25:
Sorry that no one replied here.

We already have target for this, `idf.py encrypted-bootloader-flash` . It works only in `CONFIG_SECURE_FLASH_ENCRYPTION_MODE_DEVELOPMENT` mode.

closed 2022-11-25 by mahavirj