espressif/esp-idf #7383: CRC calculation for whole Image(.bin) and append in .bin file at some memory location on eclipse for ESP32

vim-git, 2021-08-06:
**Feature Request:**

I wish I could calculate CRC for whole application from Header to app end address and append the CRC value to some memory location. Remaining location should get filled with some 0xFF or 0x00 value( Depends upon choice ). There we can calculate the CRC 32, CRC16 and CRC8. CRC16 should have different methods to calculate CRC.

**What I'd like to have**

After configuring the CRC, It should calculate the CRC value at compile time only and append that CRC value in some memory location. it should reflect in .bin file as well. So we can use that later. 

**My alternate solution**

I am trying to calculate the CRC using some tools and appending to that some particular memory location. But I have not tested this yet. I am not sure what will be the behavior after adding the CRC value to that location. 

** **
If you are aware about any other method or tool which could be used along with eclipse IDE for same. Please provide your suggestions.

igrr, 2021-08-09:
Hi @vim-git. There is no feature to calculate CRC, specifically, of the binary image. However if your goal is to validate the binary image, please be aware of the following. ESP-IDF automatically appends a SHA-256 hash of the binary image at the end of it, see https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/app_image_format.html#app-image-format. At startup, the 2nd stage bootloader validates the hash of the binary image before loading it. So if the application is running (`app_main` is called), you can be sure that the hash has already been validated. If you'd like to do additional validation of the binary image during startup, you can call `bootloader_common_get_sha256_of_partition` (declared in "bootloader_common.h") function to check the partition, passing in the offset of the current app partition and its size. You can use `esp_ota_get_running_partition` function to determine which partition the application is currently running from.
While SHA-256 is not the same as CRC, SHA256 is for all purposes a stronger check for partition integrity than a CRC.

vim-git, 2021-10-04:
Thanks @igrr

closed 2021-10-04 by vim-git