espressif/esp-idf #7385: ESP32 Firmware image size too high

ashigupta, 2021-08-06:
## Environment

- Development Kit:      ESP32-DevKitM-1
- Kit version : -
- Module or chip used:  ESP32-SOLO-1 (ESP32-Mini-1)
- IDF version : v4.2.1
- Build System:  Make|CMake|idf.py
- Compiler version :  xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:  Windows
- (Windows only) environment type: ESP Command Prompt|
- Using an IDE?: Yes (Eclipse)
- Power Supply: USB

## Problem Description
Firmware image size is too high after build even for 'hello world' example code 

//Detailed problem description goes here.
I am developing my application on ESP32 and using Wifi Station mode, IOT, BLE(Bluedroid). I realized that my firmware size is going beyond **1.6MB**. 

I compiled 'Hello World' example code and found that firmware image is **
[sdkconfig - Copy.txt](https://github.com/espressif/esp-idf/files/6947562/sdkconfig.-.Copy.txt)
** which is too huge just for hello world program. Below is the output of _'idf.py size'_ command 
Total sizes:
    DRAM .data size:    8376 bytes
    DRAM .bss  size:    2104 bytes
    Used static DRAM:   10480 bytes ( 170256 available, 5.8% used)
    Used static IRAM:   40689 bytes (  90383 available, 31.0% used)
    Flash code:   77795 bytes
    Flash rodata:   24448 bytes
    Total image size:~ 153412 bytes (.bin may be padded larger

I read espressif document on below link that explains how we can reduce image size.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/size.html 

Following this i am able to reduce the firmware image size to **120KB** which is still huge for just hello world program.  Below is the output of _idf.py size_ command 
Total sizes:
    DRAM .data size:    5528 bytes
    DRAM .bss  size:    1856 bytes
    Used static DRAM:    7384 bytes ( 173352 available, 4.1% used)
    Used static IRAM:   32363 bytes (  98709 available, 24.7% used)
    Flash code:   72207 bytes
    Flash rodata:   11044 bytes
    Total image size:~ 122998 bytes (.bin may be padded larger)

I have attached my modified sdkconfig file for hello world with which i am able to reduce size to **120KB**.

I want to understand how i can reduce the firmware image size more. 

I know Bluedroid consume more memory as compare to nimble but is there any way to reduce size when i am using Wifi, BLE both. I will be using OTA firmware update and with this huge image size upgrade size will be too high and i also have to alot high memory size OTA partition. 
I have gone through many blogs, videos but still there is no great help.

### Expected Behavior
I expect bin image size should be small for just a hello world program. 

### Actual Behavior
Bin image size is too high.

### Steps to reproduce

1. Build 'Hello world' example code using _idf.py_ build command
2. Check image size using _idf.py size_

negativekelvin, 2021-08-07:
https://github.com/espressif/esp-idf/issues/7007

ashigupta, 2021-08-09:
> #7007

Hi,

I have already followed [](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-bootloader-compiler-optimization)  and able to reduce around 300KB in my application that uses Wifi station mode + BLE ( Bluedroid). But still the memory footprint is above 1.3MB.

Below is memory consumption component wise in my application: 
#####################################################################
Per-archive contributions to ELF file:
            Archive File DRAM .data & .bss & other   IRAM   D/IRAM Flash code & rodata   Total
                 libbt.a        163  26171       0    316        0     214659    61218  302527
           libnet80211.a       1038   6181       0  11732        0      98433    21536  138920
          libAzure_IOT.a        408      0       0      0        0      66157    39998  106563
           libbtdm_app.a        763   2405       0  27634        0      67198     4551  102551
    libcrypto_matixssl.a         96   1294       0      0        0      84464    13393   99247
                  libc.a          4      4       0      0        0      80915     4638   85561
               liblwip.a         29   3823       0      0        0      70243     4238   78333
                 libpp.a       1262   3785       0  20829        0      34943     4485   65304
     libwpa_supplicant.a         12    844       0      0        0      42607     3364   46827
         libmbedcrypto.a         64     29       0     30        0      40598     3853   44574
                libphy.a       1621    675       0   7837        0      32443        0   42576
               libCode.a         49   7332       0      0        0      20547    10962   38890
                libLib.a       2444   4320       0      0        0       2890    28023   37677
           libfreertos.a       2084    736       0  12243        0          0     1406   16469
          libESP32_BLE.a         48     92       0      0        0       2361    13679   16180
                libIOT.a          0      4       0      0        0       7333     3935   11272
 libAzure_ltk_adapters.a         36     26       0      0        0       6605     4600   11267
            libcoexist.a       2030     28       0   3704        0       4632      325   10719
          libnvs_flash.a          0     24       0      0        0      10487       80   10591
                libDCI.a          4     57       0      0        0       9532      283    9876
                libsoc.a        155      4       0   6574        0       1639     1273    9645
          libspi_flash.a        623    288       0   4826        0        968        0    6705
                libgcc.a          8     12       0    193        0       5555      872    6640
                  libm.a          4      0       0      0        0       5111       69    5184
                libvfs.a        308     48       0      0        0       4492      159    5007
           libesp_wifi.a        492     76       0    496        0       3477      205    4746
              libesp32.a         16     21       0   1938        0       2023      544    4542
         libesp_system.a       1682     17       0   2188        0        250       68    4205
             libstdc++.a          8     20       0      0        0       2577     1084    3689
          libesp_netif.a         12     25       0      0        0       3366      129    3532
       libDebugManager.a         48     50       0      0        0       1954     1128    3180
             libdriver.a         80     55       0    156        0       2248      284    2823
               libheap.a         12      8       0   1811        0        933       35    2799
             libnewlib.a        190    272       0    953        0       1117      119    2651
          libesp_event.a          4     13       0      0        0       2411       68    2496
                librtc.a          0      4       0   2235        0          0        0    2239
                libBLE.a          0    320       0      0        0       1589       74    1983
          libesp_timer.a         16     20       0    954        0        795       19    1804
               libWiFi.a          0     18       0      0        0       1297      428    1743
        libOS_FreeRTOS.a          0     73       0      0        0       1485       44    1602
 libbootloader_support.a          0      0       0    886        0        544       36    1466
                liblog.a          8    268       0    250        0        526       25    1077
        libesp_ringbuf.a          0      0       0    709        0          0      183     892
               libcore.a          0     29       0      0        0        617      228     874
            libpthread.a         16     12       0    186        0        586        0     800
              libefuse.a         36      4       0      0        0        676       60     776
         libesp_common.a          8     70       0     41        0        582       43     744
         libapp_update.a          1     12       0    154        0         97      260     524
            libGeneric.a          0      4       0      0        0        434       70     508
                libhal.a          0      0       0    443        0          0       32     475
             libxtensa.a          0      0       0    416        0          0       58     474
[... cut ...]
I have tried the same sdkconfig changes in 'Hello_World' program and memory footprint for the same reduced to 120KB but still 120KB is too high for just a simple 'Hello_World' program. 

Also the link you have attached below is mentioned:

_"The default ESP-IDF configuration is also targeted for maximum functionality and easy debug-ability, rather than code size. One of the simplest size improvements is to change the compiler optimization level to Size. There are a number of other possible actions listed in the upcoming Performance Guide, will link to this as soon as it's available."_

Is the performance guide available?

negativekelvin, 2021-08-09:
It is the same guide you linked to.  If you switch to newlib nano printf you can save some more space.  Idf.py size-files will show you exactly what is using the space.  Remember it is hello world running on top of a full rtos.  Optimizing the size of hello world is not going to be that useful in helping you optimize a large app.  They sell 16mb modules for a reason.

ashigupta, 2021-08-09:
> It is the same guide you linked to. If you switch to newlib nano printf you can save some more space. Idf.py size-files will show you exactly what is using the space. Remember it is hello world [...]

Thanks. I am able to further reduce memory footprint by enabling nano printf in hello world. Its now ~80KB. 
Yes hello world is running on top of full rtos and this will also contribute in in size. I have attached idf.py file-size details.

What are the options to reduce memory footprint if BLE(Bluedroid) is enabled? I have already disabled debug logs but is there any other options that can help ?
[Hello World - File size.txt](https://github.com/espressif/esp-idf/files/6954801/Hello.World.-.File.size.txt)

igrr, 2021-08-09:
Can you consider using NimBLE stack instead of Bluedroid? It has lower code size.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/nimble/index.html

ashigupta, 2021-08-10:
> Can you consider using NimBLE stack instead of Bluedroid? It has lower code size.
> https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/nimble/index.html

Hi,

Is there any document available that shows memory consumption / footprint differences when using Bluedroid and nimble. 
It will be very helpful for understanding before starting any work using nimble.

rahult-github, 2023-08-10:
Hi @ashigupta  , did the above nimble stack suggestion help ?

rahult-github, 2023-09-25:
Closing this issue. Please feel free to reopen in case of any queries.

closed 2023-09-25 by rahult-github