espressif/esp-idf #7387: How much minimum heap is required ? Because my free heap size is Free "48428" and still I am getting below error.

MarkEvens, 2021-08-07:
IDF Version 4.2.1
OS Ubuntu
IDE VS-code

I am using BLE + WiFi together and want to do OTA update. But got below error.

How much minimum heap is required ? Because my free heap size is Free "48428" and still I am getting below error.
```
E (61836) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x10
I (61846) esp-tls-mbedtls: Certificate verified.
E (61846) esp-tls: Failed to open new connection
E (61846) TRANS_SSL: Failed to open a new connection
E (61856) HTTP_CLIENT: Connection failed, sock < 0
E (61856) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (61866) esp_https_ota: Failed to establish HTTP connection
```

_Originally posted by @MarkEvens in https://github.com/espressif/esp-idf/issues/4357#issuecomment-894220622_

chegewara, 2021-08-07:
70kB+

negativekelvin, 2021-08-08:
40k can be enough but it depends on how fragmented it is

kayzen9, 2021-08-08:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_tls.html#comparison-between-mbedtls-and-wolfssl
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/ram-usage.html#reducing-heap-usage

mahavirj, 2021-08-11:
@MarkEvens 

Pointers mentioned above have some rough ballpark for memory numbers for TLS connection. One feature that may help to reduce memory further is:

https://github.com/espressif/esp-idf/blob/83956ebbae5b335fbe7537fee737c0784d1cf471/components/mbedtls/Kconfig#L100

You may try enabling this once in your application. Additionally [blog post](https://blog.espressif.com/esp32-memory-analysis-case-study-eacc75fe5431) can be further useful in understanding/optimizing memory for typical TLS enabled cloud application on ESP32.

Hope this helps. If you still face any issues, please feel free to create new issue providing more details along with your sample application.

closed 2021-08-11 by mahavirj