espressif/esp-idf #7389: With ESP32-C3 examples/storage/sd_card/sdspi does not work

nopnop2002, 2021-08-07:
# Environment

```
freeRTOS version:V10.2.1
NEWLIB version:3.0.0
lwIP version:2-1-3-0
ESP-IDF version:v4.4-dev-2184-g166c30e7b-dirty

```

# Problem Description
With ESP32-C3 examples/storage/sd_card/sdspi does not work

```
I (30) boot: ESP-IDF v4.4-dev-2184-g166c30e7b-dirty 2nd stage bootloader
I (30) boot: compile time 17:50:27
I (30) boot: chip revision: 3
I (34) boot.esp32c3: SPI Speed      : 80MHz
I (39) boot.esp32c3: SPI Mode       : DIO
I (43) boot.esp32c3: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=09e68h ( 40552) map
I (106) esp_image: segment 1: paddr=00019e90 vaddr=3fc8b600 size=014f4h (  5364) load
I (109) esp_image: segment 2: paddr=0001b38c vaddr=40380000 size=04c8ch ( 19596) load
I (120) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=2a8bch (174268) map
I (152) esp_image: segment 4: paddr=0004a8e4 vaddr=40384c8c size=068a4h ( 26788) load
I (157) esp_image: segment 5: paddr=00051190 vaddr=50000000 size=00010h (    16) load
I (161) boot: Loaded app from partition at offset 0x10000
I (164) boot: Disabling RNG early entropy source...
I (181) cpu_start: Pro cpu up.
I (190) cpu_start: Pro cpu start user code
I (190) cpu_start: cpu freq: 160000000
I (190) cpu_start: Application information:
I (193) cpu_start: Project name:     sd_card
I (197) cpu_start: App version:      v4.4-dev-2184-g166c30e7b-dirty
I (204) cpu_start: Compile time:     Aug  7 2021 17:50:46
I (211) cpu_start: ELF file SHA256:  7f7819852a70c36a...
I (216) cpu_start: ESP-IDF:          v4.4-dev-2184-g166c30e7b-dirty
I (224) heap_init: Initializing. RAM available for dynamic allocation:
I (231) heap_init: At 3FC8D9C0 len 00032640 (201 KiB): DRAM
I (237) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (244) heap_init: At 50000010 len 00001FF0 (7 KiB): RTCRAM
I (250) spi_flash: detected chip: generic
I (255) spi_flash: flash io: dio
I (259) sleep: Configure to isolate all GPIO pins in sleep state
I (265) sleep: Enable automatic switching of GPIO sleep configuration
I (273) cpu_start: Starting scheduler.
I (277) example: Initializing SD card
I (277) example: Using SPI peripheral
E (277) spi: spi_bus_initialize(626): invalid dma channel, chip only support spi dma channel auto-alloc
E (287) example: Failed to initialize bus.
```

# Solutions
I changed some code:

https://github.com/espressif/esp-idf/blob/master/examples/storage/sd_card/sdspi/main/sd_card_example_main.c#L37
```
#if 0
#if CONFIG_IDF_TARGET_ESP32S2 ||CONFIG_IDF_TARGET_ESP32C3
#define SPI_DMA_CHAN    host.slot
#else
#define SPI_DMA_CHAN    1
#endif
#endif

#ifdef CONFIG_IDF_TARGET_ESP32
#define HOST    HSPI_HOST
#elif defined CONFIG_IDF_TARGET_ESP32S2
#define HOST    SPI2_HOST
#elif defined CONFIG_IDF_TARGET_ESP32C3
#define HOST    SPI2_HOST
#endif
```

https://github.com/espressif/esp-idf/blob/master/examples/storage/sd_card/sdspi/main/sd_card_example_main.c#L79
```
    //ret = spi_bus_initialize(host.slot, &bus_cfg, SPI_DMA_CHAN);
    ret = spi_bus_initialize(HOST, &bus_cfg, SPI_DMA_CH_AUTO);

```

It's work fine.
```
I (30) boot: ESP-IDF v4.4-dev-2184-g166c30e7b-dirty 2nd stage bootloader
I (30) boot: compile time 17:50:27
I (30) boot: chip revision: 3
I (34) boot.esp32c3: SPI Speed      : 80MHz
I (39) boot.esp32c3: SPI Mode       : DIO
I (43) boot.esp32c3: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=09e68h ( 40552) map
I (106) esp_image: segment 1: paddr=00019e90 vaddr=3fc8b600 size=014f4h (  5364) load
I (109) esp_image: segment 2: paddr=0001b38c vaddr=40380000 size=04c8ch ( 19596) load
I (120) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=2a8bch (174268) map
I (152) esp_image: segment 4: paddr=0004a8e4 vaddr=40384c8c size=068a4h ( 26788) load
I (157) esp_image: segment 5: paddr=00051190 vaddr=50000000 size=00010h (    16) load
I (161) boot: Loaded app from partition at offset 0x10000
I (164) boot: Disabling RNG early entropy source...
I (181) cpu_start: Pro cpu up.
I (190) cpu_start: Pro cpu start user code
I (190) cpu_start: cpu freq: 160000000
I (190) cpu_start: Application information:
I (193) cpu_start: Project name:     sd_card
I (197) cpu_start: App version:      v4.4-dev-2184-g166c30e7b-dirty
I (204) cpu_start: Compile time:     Aug  7 2021 17:50:46
I (211) cpu_start: ELF file SHA256:  69bb378dacfae6fc...
I (216) cpu_start: ESP-IDF:          v4.4-dev-2184-g166c30e7b-dirty
I (224) heap_init: Initializing. RAM available for dynamic allocation:
I (231) heap_init: At 3FC8D9C0 len 00032640 (201 KiB): DRAM
I (237) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (244) heap_init: At 50000010 len 00001FF0 (7 KiB): RTCRAM
I (250) spi_flash: detected chip: generic
I (255) spi_flash: flash io: dio
I (259) sleep: Configure to isolate all GPIO pins in sleep state
I (265) sleep: Enable automatic switching of GPIO sleep configuration
I (273) cpu_start: Starting scheduler.
I (277) example: Initializing SD card
I (277) example: Using SPI peripheral
I (277) gpio: GPIO[19]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (287) sdspi_transaction: cmd=52, R1 response: command not supported
I (337) sdspi_transaction: cmd=5, R1 response: command not supported
Name: SC16G
Type: SDHC/SDXC
Speed: 20 MHz
Size: 15193MB
I (367) example: Opening file /sdcard/hello.txt
I (387) example: File written
I (387) example: Renaming file /sdcard/hello.txt to /sdcard/foo.txt
I (397) example: Reading file /sdcard/foo.txt
I (397) example: Read from file: 'Hello SC16G!'
I (397) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (407) example: Card unmounted
I (407) gpio: GPIO[9]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (417) gpio: GPIO[18]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (427) gpio: GPIO[8]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
```

# Reference
https://github.com/espressif/esp-idf/issues/7128

yshestakov, 2021-08-23:
I have a similar issue with my ESP32-C3 board (LILYGO T-IO Plus).
However, changing the DMA channel to AUTO doesn't help. I got reset by watchdog:

```
I (296) example: Initializing SD card
I (296) example: Using SPI peripheral
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380080
0x40380080: _panic_handler at vectors.S.obj:?
```

nopnop2002, 2021-08-23:
@yshestakov 

Is this part the same as me?

```
I (34) boot.esp32c3: SPI Speed      : 80MHz
I (39) boot.esp32c3: SPI Mode       : DIO
I (43) boot.esp32c3: SPI Flash Size : 2MB
```

yshestakov, 2021-08-23:
@nopnop2002  yes, the same

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x15d4
load:0x403ce000,len:0x92c
load:0x403d0000,len:0x2a98
entry 0x403ce000
I (30) boot: ESP-IDF v4.4-dev-2595-g38bbf87a2a-dirty 2nd stage bootloader
I (30) boot: compile time 11:34:38
I (30) boot: chip revision: 3
I (34) boot.esp32c3: SPI Speed      : 80MHz
I (39) boot.esp32c3: SPI Mode       : DIO
I (43) boot.esp32c3: SPI Flash Size : 2MB
```
Meanwhile, I put `ESP_LOGI(...)` calls into the body of `spi_bus_initialize()` and found that the `spicommon_bus_initialize_io` call caused the WDT reset.

nopnop2002, 2021-08-23:
@yshestakov 

When using SPI with the ESP32-C3, there is a problem that SPI CLOCK cannot be used on certain pins.

There may be some issues with the ESP32-C3 SPI driver.

yshestakov, 2021-08-23:
@nopnop2002  In my case it seems MISO pin configuration caused the reset
```
        if (bus_config->miso_io_num >= 0) {
            if (miso_need_output || (temp_flag&SPICOMMON_BUSFLAG_DUAL)) {
                gpio_set_direction(bus_config->miso_io_num, GPIO_MODE_INPUT_OUTPUT);
                esp_rom_gpio_connect_out_signal(bus_config->miso_io_num, spi_periph_signal[host].spiq_out, false, false);
            }

yshestakov, 2021-08-23:
Ha! I made it working. I missed `Dx` pin names (WEMOS.cc) with actual GPIO numbers. So, it works for me now :)

closed 2021-10-12 by igrr