espressif/esp-idf #7390: ADC 1 Time drift

francis2tm, 2021-08-08:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (v3]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
    v4.3-dev-1720-g494a124d9
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description

I'm exploring the ESP's ADC capabilities. To characterize a potential time drift of a ADC acquisition, I acquired the same signal,  a 100Hz sine wave, using an oscilloscope and esp wroom 32. The ESP was programmed to have a sample rate of 16kHz (using timer 0,0). The acquisition firmware for the esp involved: TIMER interruption wakes up a task, like so:

`void IRAM_ATTR timerGrp0Isr(){
    timer_spinlock_take(TIMER_GROUP_USED);

    //Clear the interrupt
    timer_group_clr_intr_status_in_isr(TIMER_GROUP_USED, TIMER_IDX_USED);

    //After the alarm has been triggered we need enable it again, so it is triggered the next time
    timer_group_enable_alarm_in_isr(TIMER_GROUP_USED, TIMER_IDX_USED);

    timer_spinlock_give(TIMER_GROUP_USED);

    //Wake acqAdc2 in order to start ADC readings from adc2. CPU0 will start immediatly acquiring
    //vTaskNotifyGiveFromISR(acquiring_2_task, (BaseType_t*)NULL);

    //Wake acqAdc1 in order to start ADC readings form adc1. This will only start when this handler is terminated.
    vTaskNotifyGiveFromISR(acquiring_1_task, NULL);

    /* If xHigherPriorityTaskWoken is now set to pdTRUE then a context switch
    should be performed to ensure the interrupt returns directly to the highest
    priority task.  The macro used for this purpose is dependent on the port in
    use and may be called portEND_SWITCHING_ISR(). */
    portYIELD_FROM_ISR();
}`

The task, after waking up (with ulTaskNotifyTake()), proceeds with the adc reading.

The results I got show a big time drift after a bit some seconds. I

[image]

[image]

What is the cause of this drift?

summivox, 2021-08-08:
Your ESP32 is clocked by its onboard crystal oscillator and PLL. Their frequency accuracy is likely to be lower than the oscilloscope.

According to your plot, scope wave at t=26.490 is equal to esp wave at t=26.492 (roughly).
This is a relative frequency error of 0.002/26.490 = 7.55e-5 (76 ppm) which kinda high for a crystal but not off by that much.

One easy way to show this is to have ESP32 toggle a GPIO pin on a timer event (you don't even need to wake up a task). You can then measure the frequency of this pin with your oscilloscope.

Another thing you might want to look into: How did you set the frequency of the timer? Was there an off-by-one-tick error?

francis2tm, 2021-08-10:
I already did the esp32 toggle gpio pin oscilastion to measure frequency, I got these results:
[image]

francis2tm, 2021-08-13:
Thank you for the reply, I'm going to try that feature.

francis2tm, 2021-08-24:
Hey,
Thanks @ginkgm , it improved a lot.

closed 2021-08-24 by francis2tm