espressif/esp-idf #7391: ESP-install toolchain

UnoCube, 2021-08-08:
While installing using install.bat esp32s2, progress is stuck at downloading ninja.

Same occurs when using ESP-IDF installer. 

Probably a bad entry in tools.json. Solved for me by deleting the entry for ninja, running the install.bat, manually downloading the ninja, putting it into the dist file under .espressif and renaming it to match the expectation in tools.json, after again running install.bat it works like charm and the export.bat ends in success.

The tools.json should be edited to reflect the names of files in source from which they are to be downloaded.

dobairoland, 2021-08-09:
Hi @UnoCube. Thanks for your report.

Do you remember what was the file name after download? You can still get a wrong filename when you download https://dl.espressif.com/dl/ninja-1.10.2-win64.zip?

You don't have some antivirus program which would rename it (quarantine it)?

UnoCube, 2021-08-09:
Hi @dobairoland .
I do have Avast, but I don't think it caused it as It was inactive during whole install.

At first I had problems installing it through the VS CODE extension, where it got stuck at the third elf file for the ESP32-S3.
I thought I had Internet problems, but then github was working fine. Then I thought that the file ​was simply missing from
server and then I tried the manual toolchain install and with the workaround and it worked. To test it, I later deleted both .espressif and esp folders and again tried the VS CODE and to my surprise it worked. I did not want to try my luck anymore so I stopped.

Looking back it may have been some weird internet issues, while my connection to usual servers was fine I struggled to load espressif docks and forums. I you think this is the case go ahead and close the issue. Sorry for wasting your time :)

BTW yes that is the file in question however as I wrote higher I encountered problems with: xtensa-esp32s3-elf-gcc8_4_0-esp-2020r3-win64 too in VS CODE but not in the ESP-IDF tools 2.4.

Prajem pekný deň :)

dobairoland, 2021-08-10:
Hi @UnoCube. This could be a new issue with Vscode. I've found https://github.com/espressif/vscode-esp-idf-extension/issues/490 which has been recently repored. Maybe there will be a follow-up discussion there.

Both the files are downloaded and used for more than 6 months and this is the first report I see.

I thank you for your report. I'm closing this now because I believe there is nothing we can fix in ESP-IDF to avoid such issues.

closed 2021-08-10 by dobairoland