espressif/esp-idf #7392: esp_ble_resolve_adv_data

FaeSorceress, 2021-08-09:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it):
    v4.3-302-g47b96db12d
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System:     [Linux]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

Can't resolve device name.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. run demo gatt_client
2. it can print device address but cant't resolve name.
3. in my ble device, the device name type data place into scan rst dat ,and it has 0 between adv data and scan rst data.
4. I think this question is because to the esp_ble_resolve_adv_data return on 0.
[image]

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
//modification the code in btm_ble_gap.c 

UINT8 *BTM_CheckAdvData(UINT8 *p_adv, UINT8 type, UINT8 *p_length)
{
        /*code*/
        while (length && (p - p_adv <= BTM_BLE_CACHE_ADV_DATA_MAX))
       {
              /*code*/
               p += length - 1; /* skip the length of data */

               while (*p == 0) //add
              {                         //add
                  p++;              //add
              }                        //add
              STREAM_TO_UINT8(length, p);
       }
       /*code*/
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
According to above modification, it can  resolve adv device name properly.
I (10657) 8MS_GATT: c0 00 00 00 00 01 
I (10657) 8MS_GATT: searched Adv Data Len 31, Scan Response Len 28
I (10667) 8MS_GATT: searched Device Name Len 9
I (10667) 8MS_GATT: WT5105-M1
I (10667) 8MS_GATT: adv data:
I (10677) 8MS_GATT: 02 01 06 11 07 1b d7 90 ec e8 b9 75 80 0a 46 44 
I (10687) 8MS_GATT: d3 01 02 ed a6 03 ff ff ff 00 00 00 00 00 00 
I (10687) 8MS_GATT: scan resp:
I (10697) 8MS_GATT: 0a 09 57 54 35 31 30 35 2d 4d 31 00 00 00 00 00 
I (10697) 8MS_GATT: 00 00 00 00 00 00 00 00 00 00 00 00 
I (10707) 8MS_GATT: 

I (10707) 8MS_GATT: searched device WT5105-M1
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

xulongzhuang, 2021-08-10:
@FaeSorceress 
Hi, I have reproduced the problem you encountered. When adv_data has invalid data 0 and the device name is placed in scan_rsp_data, BTM_CheckAdvData will return early because length =0 is encountered. Thank you very much for reporting this problem.  Can you submit a pull request to try to incorporate your solution?

FaeSorceress, 2021-08-10:
It's my pleasure. I have don so.

---------------------------------------------------
深圳市启明云端科技有限公司
研发部 软件组
彭飞 18035864063
***@***.*** 
 
From: xulongzhuang
Date: 2021-08-10 09:52
To: espressif/esp-idf
CC: stephen; Mention
Subject: Re: [espressif/esp-idf] esp_ble_resolve_adv_data (IDFGH-5671) (#7392)
@FaeSorceress
Hi, I have reproduced the problem you encountered. When adv_data has invalid data 0 and the device name is placed in scan_rsp_data, BTM_CheckAdvData will return early because length =0 is encountered. Thank you very much for reporting this problem. Can you submit a pull request to try to incorporate your solution?

closed 2021-08-10 by FaeSorceress