espressif/esp-idf #7393: TWAI: Baud rate prescaler in interrupt_enable_reg not working

Th3Link, 2021-08-09:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32E (should be a revision 3 chip according to docs and logs)
- IDF version: v4.4-dev-960-gcf457d412
- Build System:  idf.py
- Compiler version: 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: external 5V

## Problem Description

When we want to use a TWAI baud rate of 20kbits, the additional clock halving bit in interrupt_enable_reg (bit 5) is set in twai_ll.h : twai_ll_set_bus_timing. However, the bus baud rate does not change; When TWAI_TIMING_CONFIG_12_5KBITS we get 25KBITS (used oscilloskop to double check), when setting TWAI_TIMING_CONFIG_20KBITS we get 40KBITS and when setting TWAI_TIMING_CONFIG_25KBITS we get 25KBITS because then the halving is not needed.
We have checked if the bit in the register changed and discovered, that it is set to 1 in reset state and will again be set to 1 when configuring a lower baud rate and is cleared to 0 on >= 25KBITS.

We ask this as a question here: https://esp32.com/viewtopic.php?f=13&t=22610

### Expected Behavior
When setting the TWAI speed to TWAI_TIMING_CONFIG_20KBITS we get a baud rate of 20KBITS

### Actual Behavior
When setting the TWAI speed to TWAI_TIMING_CONFIG_20KBITS we get a baud rate of 40KBITS

### Steps to reproduce
Start the program, happens always

### Code to reproduce this issue

- [main.c](https://gist.github.com/Th3Link/ae172b4eacfcd7d51ae615492e46a87d)

## Other items if possible

- [sdkconfig.txt](https://github.com/espressif/esp-idf/files/6957099/sdkconfig.txt)
- [startup_log](https://gist.github.com/Th3Link/00f8f5adc0b9418972d619f224efc457)

Dazza0, 2021-09-07:
@Th3Link I've been able to reproduce the issue on a ESP32-WROOM-32E. The cause is the issue is the `fstrict-volatile-bitfields` compiler bug (described in https://github.com/espressif/esp-idf/issues/5604). The fix to that issue has been addressed in 2e0ffbd543abc0226f0dbd2f725eb891235abe02. Could you try checking out to 2e0ffbd543abc0226f0dbd2f725eb891235abe02 or later and see if the issue is resolved?

Th3Link, 2021-09-08:
@Dazza0 We checked out commit 2e0ffbd and can confirm, that the issue is fixed with it and the baud rate prescaler works as expected. Thanks a lot.

closed 2021-09-08 by Th3Link