espressif/esp-idf #7396: MD5 of current firmware

PratikMistry-cbi, 2021-08-10:
**Is your feature request related to a problem? Please describe.**

A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**

A clear and concise description of what you want to happen.

**Describe alternatives you've considered**

A clear and concise description of any alternative solutions or features you've considered.

Please give as many details as you can. Include suggestions for useful APIs or interfaces if relevant.

**Additional context**

Add any other context or screenshots about the feature request here.

mahavirj, 2021-08-12:
@PratikMistry-cbi 

For MD5 sum, API reference is available at:

https://github.com/espressif/esp-idf/blob/master/components/esp_rom/include/esp_rom_md5.h

There is no built-in API to retrieve MD5 of current firmware, but you may create one along the lines of SHA256 from:

https://github.com/espressif/esp-idf/blob/166c30e7b2ed1dcaae56179329540a862915208a/components/bootloader_support/src/bootloader_utility.c#L880

mahavirj, 2021-08-18:
Closing the issue, if there are any additional questions please feel free to start thread at esp32.com

closed 2021-08-18 by mahavirj