espressif/esp-idf #7398: pll_vol_cal abort

xno-ai, 2021-08-10:
----------------------------- Delete below -----------------------------
## Debug Logs

I (21) boot: ESP-IDF v4.3-dirty 2nd stage bootloader
I (21) boot: compile time 11:55:44
I (21) boot: chip revision: 0

I (6117) wifi:wifi driver task: 3ffddec0, prio:23, stack:3584, core=0
I (6117) system_api: Base MAC address is not set
I (6117) system_api: read default base MAC address from EFUSE
I (6117) wifi:wifi firmware version: c7d0450
I (6127) wifi:wifi certification version: v7.0
I (6127) wifi:config NVS flash: enabled
I (6127) wifi:config nano formating: disabled
I (6137) wifi:Init data frame dynamic rx buffer num: 32
I (6137) wifi:Init management frame dynamic rx buffer num: 32
I (6147) wifi:Init management short buffer num: 32
I (6147) wifi:Init static tx buffer num: 16
I (6157) wifi:Init tx cache buffer num: 32
I (6157) wifi:Init static rx buffer size: 1600
I (6157) wifi:Init static rx buffer num: 10
I (6167) wifi:Init dynamic rx buffer num: 32
I (6167) wifi_init: rx ba win: 6
I (6177) wifi_init: tcpip mbox: 32
I (6177) wifi_init: udp mbox: 6
I (6177) wifi_init: tcp mbox: 6
I (6187) wifi_init: tcp tx win: 5744
I (6187) wifi_init: tcp rx win: 5744
I (6197) wifi_init: tcp mss: 1440
I (6197) wifi_init: WiFi IRAM OP enabled
I (6207) wifi_init: WiFi RX IRAM OP enabled
I (6207) main: start the WIFI SSID:[FreeWiFi]
I (6217) phy_init: phy_version 1800,e7ef680,Apr 13 2021,11:45:08
I (6227) phy: pll_cap_ext 10
I (6237) phy: pll_cap_ext 10
I (6237) phy: pll_cap_ext 10
I (6237) phy: pll_cap_ext 10
I (6237) phy: pll_cap_ext 10
I (6237) phy: pll_cap_ext 10
I (6247) phy: pll_cap_ext 10
I (6247) phy: pll_cap_ext 10
I (6247) phy: pll_cap_ext 10
I (6257) phy: pll_cap_ext 10
I (6257) phy: pll_cap_ext 10
I (6257) phy: pll_cap_ext 10
I (6277) phy: pll_cap_ext 10
I (6287) phy: pll_cap_ext 10
I (6297) phy: pll_cap_ext 10
I (6307) phy: pll_cap_ext 10
I (6307) phy: pll_cap_ext 10
I (6337) phy: pll_cap_ext 10
I (6347) phy: pll_cap_ext 10
I (6347) phy: pll_cap_ext 10
I (6347) phy: pll_cap_ext 10
I (6347) phy: pll_cap_ext 10
I (6347) phy: pll_cap_ext 10
I (6347) phy: pll_cap_ext 10
I (6357) phy: pll_cap_ext 10
I (6357) phy: pll_cap_ext 10
I (6357) phy: pll_cap_ext 10
I (6367) phy: pll_cap_ext 10
I (6367) phy: pll_cap_ext 10
I (6377) phy: pll_cap_ext 10
I (6377) phy: pll_cap_ext 10
I (6377) phy: pll_cap_ext 10

abort() was called at PC 0x40026caf on core 0
0x40026caf: lock_acquire_generic at E:/project/espressif/esp-idf-v4.3/components/newlib/locks.c:138

Backtrace:0x4002d73f:0x3ffe5a90 0x4002dfe9:0x3ffe5ab0 0x40035162:0x3ffe5ad0 0x40026caf:0x3ffe5b40 0x40026df9:0x3ffe5b70 0x400f7a8e:0x3ffe5b90 0x400faaed:0x3ffe5ea0
 0x401043c9:0x3ffe5ed0 0x40033c01:0x3ffe5f00 0x400efbad:0x3ffe5f50 0x400efbd7:0x3ffe5fd0 0x40029bbb:0x3ffe6020 0x400e5bc3:0x3ffe6080 0x400e5c5b:0x3ffe60a0 0x400e5c
cc:0x3ffe60c0 0x400e6c99:0x3ffe60e0 0x400e777a:0x3ffe6110 0x400e782b:0x3ffe6130 0x400ab984:0x3ffe6150 0x400ab9b8:0x3ffe6170 0x400cde26:0x3ffe6190 0x400cdef2:0x3ffe
61b0 0x400cb7b4:0x3ffe61d0 0x40036ae8:0x3ffe61f0 0x4003104d:0x3ffe6220
0x4002d73f: panic_abort at E:/project/espressif/esp-idf-v4.3/components/esp_system/panic.c:356

0x4002dfe9: esp_system_abort at E:/project/espressif/esp-idf-v4.3/components/esp_system/system_api.c:112

0x40035162: abort at E:/project/espressif/esp-idf-v4.3/components/newlib/abort.c:46

0x40026caf: lock_acquire_generic at E:/project/espressif/esp-idf-v4.3/components/newlib/locks.c:138

0x40026df9: _lock_acquire_recursive at E:/project/espressif/esp-idf-v4.3/components/newlib/locks.c:166

0x400f7a8e: _vfprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s2-elf/src/newlib/newlib/libc/stdio/vfprintf.c:853 (discriminator 2)

0x400faaed: vprintf at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s2-elf/src/newlib/newlib/libc/stdio/vprintf.c:34 (discriminator 5)

0x401043c9: esp_log_writev at E:/project/espressif/esp-idf-v4.3/components/log/log.c:189

0x40033c01: esp_log_write at E:/project/espressif/esp-idf-v4.3/components/log/log.c:199

0x400efbad: lib_printf at E:/project/espressif/esp-idf-v4.3/components/esp_wifi/src/lib_printf.c:47 (discriminator 13)

0x400efbd7: phy_printf at E:/project/espressif/esp-idf-v4.3/components/esp_wifi/src/lib_printf.c:57

0x40029bbb: pll_vol_cal at ??:?

TianaESP, 2021-08-16:
@xno-ai This was a known issue for C3 chip version0 and was fixed by the later chip version, please update your hardware to C3 ECO3 and try. Thanks.

xno-ai, 2021-08-19:
> @xno-ai This was a known issue for C3 chip version0 and was fixed by the later chip version, please update your hardware to C3 ECO3 and try. Thanks.

this bug chip:esp32s2.

AxelLin, 2022-03-01:
> @xno-ai This was a known issue for C3 chip version0 and was fixed by the later chip version, please update your hardware to C3 ECO3 and try. Thanks.

@TianaESP 
This bug report is against **ESP32S2**, is this a known issue for ESP32S2? Or do you have any update for this issue?

jack0c, 2022-05-20:
@xno-ai  Sorry for the delay, I will sync with team.

zhangyanjiaoesp, 2022-05-23:
@xno-ai @AxelLin @lonelyion Please update your IDF version to the latest v4.3, the phy version has update to `phy_init: phy_version 2300,d67cf06,Feb 10 2022,10:03:07`

jack0c, 2024-02-19:
Close it since no update for long. Thanks for reporting, feel free to reopen.

closed 2024-02-19 by jack0c