espressif/esp-idf #7399: Uart error after Ethernet link up in  idf 4.3

sowmimalu, 2021-08-10:
Hai ,
Module or chip used: ESP32-WROVER
IDF version: v4.3
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0

In my application , esp32 is connected with ethernet lan8720 and also with UART [ used pins (GPIO_NUM_5) as TXD and  (GPIO_NUM_4)  as RXD ]esp32 to other controller [nrf] to transmit and also to receive a data ,  from nrf to esp32 i am receiving packets which i transferred   but ,here i am facing that from esp to nrf trying to transmit data  till ethernet gets links up.... when ethernet connected  uart transmission gets stopped from esp to nrf .

this same was working  in idf 4.0 , ethernet and uart both TX and RX end has receiving and transmitting data 

was moved with idf 4.3 for new feature update ETH2AP .
I'm not sure what caused this problem ,I would like to hear your comments on this.

Thanks,
Sowmiya

kostaond, 2021-08-12:
Hi @sowmimalu, could you please check that "PHY Reset GPIO number" is not set to GPIO 5 (default value) in SDK Configuration Editor (or by idf.py menuconfig)?

sowmimalu, 2021-08-13:
hai ,
   In application i had removed phy_config.reset_gpio_num = CONFIG_EXAMPLE_ETH_PHY_RST_GPIO; 
not defined  before.

i had called etherent initialize first then uart added  both uart TX and RX worked.

kostaond, 2021-08-13:
Hi @sowmimalu, do I understand it correctly that everything works now?
However, just one note. If you initialize `phy_config` using `ETH_PHY_DEFAULT_CONFIG` as shown bellow
```c
eth_phy_config_t phy_config = ETH_PHY_DEFAULT_CONFIG();
```
the `reset_gpio_num` is initialized to 5 by default (see definition of `ETH_PHY_DEFAULT_CONFIG`).

sowmimalu, 2021-08-13:
eth_mac_config_t mac_config = ETH_MAC_DEFAULT_CONFIG();
	eth_phy_config_t phy_config = ETH_PHY_DEFAULT_CONFIG();
	phy_config.phy_addr = 0;

	mac_config.smi_mdc_gpio_num = PIN_SMI_MDC;
	mac_config.smi_mdio_gpio_num = PIN_SMI_MDIO;
	esp_eth_mac_t *mac = esp_eth_mac_new_esp32(&mac_config);
	esp_eth_phy_t *phy = esp_eth_phy_new_lan8720(&phy_config);

	esp_eth_config_t config = ETH_DEFAULT_CONFIG(mac, phy);

	gpio_pad_select_gpio(PIN_PHY_POWER);
	gpio_set_direction(PIN_PHY_POWER, GPIO_MODE_OUTPUT);
	gpio_set_level(PIN_PHY_POWER, 1);

yes working , but here i have make   phy_config.reset_gpio_num = CONFIG_EXAMPLE_ETH_PHY_RST_GPIO; as -1  ?

kostaond, 2021-08-13:
yes, if you don't want to use Eth driver's hardware reset to reset your PHY, then set `phy_config.reset_gpio_num` to -1.

sowmimalu, 2021-08-13:
yaah , thank you  @kostaond  closing this now

closed 2021-08-13 by sowmimalu